espressif/esp-idf #6014: unable to send 802.11 wifi probe request with esp_wifi_80211_tx()

arunrb, 2020-10-22:
## Environment

- Development Kit:      ESP32-PICO based  (ATOM Matrix ESP32 Development Kit) [https://docs.m5stack.com/#/en/core/atom_matrix](url)
- Kit version (for WroverKit/PicoKit/DevKitC):    Not sure
- Module or chip used:  ESP32-PICO-D4
- IDF version :   v4.0.1
- Build System:   idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     macOS
- Using an IDE?: Yes. **Note:** using Eclipse just for code/project browsing. Compilation/flasing is done via terminal using idf.py
- Power Supply:         USB

## Problem Description
Unable to send wifi probe requests with the api _**esp_wifi_80211_tx()**_ .The function call seems to be successful (returns ESP_OK) however I do not see any wifi probe requests form the teh ESP device.

//Detailed problem description goes here.
Hello,
I am noob to ESP and I am looking for help with the _**esp_wifi_80211_tx()**_  api to broadcast wifi probe requests. 
The ESP device (ATOM) is in STA mode and is associated to an AP (Channel 3). I could send wifi Beacons frames with this API, however I am unable to send wifi probe requests. Also, I am just interested in broadcasting wifi probe request, don't really need to wait for response form any AP. 

Gist:
Once I flash the firmware, the ATOM device boots, loads the application where it:  initializes bluetooth(BLE scanning is disabled), initializes wifi and connects to a pre-configured 2.4 GHz network, initializes MQTT client & connects to a broker etc and creates a software timed task that periodically sends the probe request by calling **_esp_wifi_80211_tx()_** . 

**Probe request frame:**

- I used the **_esp_wifi_scan_start()_** api to scan a specific channel (Channel 3 for example). Probe requests are broadcast during scanning, so I configured a network on my MacBook (on channel 3), and used Wireshark captured the probe requests that are being broadcast. 

- Reference probe packet captured by wireshark during **_esp_wifi_scan_start()_** is uploaded here [https://gist.github.com/arunrb/f0ec521855252bf4490c7c9cd55ed659#file-probe-request-reference-captured-with-wireshark](url) (Hex dump at the end). Wireshark capture (Pcap) is also attached just in case: [probe_requests_from_esp_wifi_scan_start_api.pcapng.zip](https://github.com/espressif/esp-idf/files/5424722/probe_requests_from_esp_wifi_scan_start_api.pcapng.zip)

- I then used the captured probe request as reference to create an input frame buffer which was fed to  **_esp_wifi_80211_tx()_**. The input frame buffer can bee found here: [https://gist.github.com/arunrb/f0ec521855252bf4490c7c9cd55ed659#file-input_probe_request_buffer](url)

- The **_esp_wifi_80211_tx()_** is then invoked as below:
` err_code = esp_wifi_80211_tx(ESP_IF_WIFI_STA, PKT2SEND, sizeof(PKT2SEND), true);`. 
code_snippet is uploaded here: [https://gist.github.com/arunrb/f0ec521855252bf4490c7c9cd55ed659#file-snippet_invoking_esp_wifi_80211_tx](url)

- No errors are show and looks like **_esp_wifi_80211_tx()_** was invoked succesfully, however I don't see any probe requests broadcasted. Console logs are a attached here: [https://gist.github.com/arunrb/f0ec521855252bf4490c7c9cd55ed659#file-application_console_output](url)

### Expected Behavior
Capture Probe request frame sent or error message from fn **_esp_wifi_80211_tx()_**.

### Actual Behavior
Looks like **_esp_wifi_80211_tx()_** was invoked succesfully, however I don't see any probe requests broadcasted. 

### Code to reproduce this issue

I don't have the libery of sharing the code, but here is the snippet [https://gist.github.com/arunrb/f0ec521855252bf4490c7c9cd55ed659#file-snippet_invoking_esp_wifi_80211_tx](url)

## Other items
I browsed and could not find any related discussion, so I create a new forum post in https://esp32.com . 
Forum post - [https://www.esp32.com/viewtopic.php?f=13&t=17819](url) 

Any help or pointers is appericiated!

YouDONG-ESP, 2020-10-23:
Hi @arunrb , could u show me your code? We will take a look at it.

arunrb, 2020-10-23:
Hello @YouDONG-ESP ,
Thanks for getting back. 

I am not sure if I am allowed to post our source code, I am checking  with my supervisor if I could post the code. 
Meantime, I could share snippets if you are looking for some specific parts. 

Please let me know.

YouDONG-ESP, 2020-10-23:
Hi @arunrb, I will take a look at the snippets at first. Meanwhile, The links you shared seem cant be opened, you could post your code directly here or send me a file.

arunrb, 2020-10-23:
@YouDONG-ESP, 
not sure why the links don't redirect, but I think it works when you copy paste the links in a browser tab.

Anyways, I am attaching the logs/code snippets here, please check.

[IDFGH-4152_issue_6014-code_snippets_logs.zip](https://github.com/espressif/esp-idf/files/5427293/IDFGH-4152_issue_6014-code_snippets_logs.zip)

YouDONG-ESP, 2020-10-27:
Hi @arunrb , I added your code in station example https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station, I connect ESP32 to an AP on channel 6, It totally works. Here is my code 
[station_example_main.zip](https://github.com/espressif/esp-idf/files/5443648/station_example_main.zip).
Make sure your packet capture software is on the right channel, more detaills for esp_wifi_80211_tx is here https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html?highlight=80211_tx

[image]

arunrb, 2020-11-04:
Thanks @YouDONG-ESP for checking this.
I haven't been able to run your station example code in my exising IDF (IDF version  v4.0.1). Are you running a different IDF version? Does the IDF version matter?

YouDONG-ESP, 2020-11-04:
Just checked v4.0.1, it also works. Replace station_example_main.c by this code then run station example. 
/* WiFi station Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sys.h"

/* The examples use WiFi configuration that you can set via project configuration menu

   If you'd rather not, just change the below entries to strings with
   the config you want - ie #define EXAMPLE_WIFI_SSID "mywifissid"
*/
#define EXAMPLE_ESP_WIFI_SSID      CONFIG_ESP_WIFI_SSID
#define EXAMPLE_ESP_WIFI_PASS      CONFIG_ESP_WIFI_PASSWORD
#define EXAMPLE_ESP_MAXIMUM_RETRY  CONFIG_ESP_MAXIMUM_RETRY
#define PKT2SEND probe_req

/* FreeRTOS event group to signal when we are connected*/
static EventGroupHandle_t s_wifi_event_group;

/* The event group allows multiple bits for each event, but we only care about two events:
 * - we are connected to the AP with an IP
 * - we failed to connect after the maximum amount of retries */
#define WIFI_CONNECTED_BIT BIT0
#define WIFI_FAIL_BIT      BIT1

static const char *TAG = "wifi station";

static int s_retry_num = 0;

static void event_handler(void* arg, esp_event_base_t event_base,
                                int32_t event_id, void* event_data)
{
    if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) {
        esp_wifi_connect();
    } else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED) {
        if (s_retry_num < EXAMPLE_ESP_MAXIMUM_RETRY) {
            esp_wifi_connect();
            s_retry_num++;
            ESP_LOGI(TAG, "retry to connect to the AP");
        } else {
            xEventGroupSetBits(s_wifi_event_group, WIFI_FAIL_BIT);
        }
        ESP_LOGI(TAG,"connect to the AP fail");
    } else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) {
        ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
        ESP_LOGI(TAG, "got ip:%s",
                 ip4addr_ntoa(&event->ip_info.ip));
        s_retry_num = 0;
        xEventGroupSetBits(s_wifi_event_group, WIFI_CONNECTED_BIT);
    }
}

void wifi_init_sta()
{
[... 96 lines cut ...]

   err_code = esp_wifi_80211_tx(ESP_IF_WIFI_STA, PKT2SEND, sizeof(PKT2SEND), true);

   vTaskDelay(200 / portTICK_PERIOD_MS);

   // check if the ERR is memory related
   if(err_code == ESP_OK)
	{	 printf("done\n");}

    vEventGroupDelete(s_wifi_event_group);
}

void app_main(void)
{
    //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    wifi_init_sta();
}

arunrb, 2020-11-04:
Hi @YouDONG-ESP ,
I tried the station example and I could see that one probe request being sent. However when I try sending probe requests repeatedly in a loop, I do not see probe requests coming in. 

I am suspecting that even the one probe request that we receive was actually sentout before the STA associates with the AP and after association, no probe requests is being sent.

For example, in the test below, note that the STA tries to associate with the AP 6 times(bold) and those are the only probe requets we receive. After associating with the AP, I don't receive any probe request.

**Example console log:**

I (759) phy: phy_version: 4370, 4e803b3, Aug 11 2020, 14:18:07, 0, 0
I (759) wifi:mode : sta (50:02:91:90:0c:70)
I (759) wifi station: wifi_init_sta finished.
I (879) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (879) wifi:state: init -> auth (b0)
I (889) wifi:state: auth -> assoc (0)
I (899) wifi:state: assoc -> run (10)
I (3999) wifi:state: run -> init (fc0)
I (3999) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (3999) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
**_I (4009) wifi station: retry to connect to the AP
I (4009) wifi station: connect to the AP fail
I (6059) wifi station: retry to connect to the AP
I (6059) wifi station: connect to the AP fail_**
I (6179) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (6179) wifi:state: init -> auth (b0)
I (6179) wifi:state: auth -> assoc (0)
I (6189) wifi:state: assoc -> run (10)
I (9289) wifi:state: run -> init (fc0)
I (9289) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (9289) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
**_I (9299) wifi station: retry to connect to the AP
I (9299) wifi station: connect to the AP fail
I (11349) wifi station: retry to connect to the AP
I (11349) wifi station: connect to the AP fail_**
I (11469) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (11469) wifi:state: init -> auth (b0)
I (11469) wifi:state: auth -> assoc (0)
I (11479) wifi:state: assoc -> run (10)
I (14589) wifi:state: run -> init (fc0)
I (14589) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (14589) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
**_I (14589) wifi station: retry to connect to the AP
I (14599) wifi station: connect to the AP fail
I (16639) wifi station: connect to the AP fail_**
I (16639) wifi station: Failed to connect to SSID:myssid, password:mypassword
I (17149) wifi station: sent probe..
ctr = 1
I (17649) wifi station: sent probe..
ctr = 2
I (18149) wifi station: sent probe..
ctr = 3
I (18649) wifi station: sent probe..
ctr = 4
I (19149) wifi station: sent probe..

**Wireshark capture**
[image]

**Below is my code that sends probe req in loop, please check:**

`/* WiFi station Example

This example code is in the Public Domain (or CC0 licensed, at your option.)

Unless required by applicable law or agreed to in writing, this
software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
[... 177 lines cut ...]
while(ctr < 100) {
	send_probe();
	ctr+= 1;
	printf("ctr = %d\n", ctr);
}

vEventGroupDelete(s_wifi_event_group);

}

void app_main(void)
{
//Initialize NVS
esp_err_t ret = nvs_flash_init();
if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
ESP_ERROR_CHECK(nvs_flash_erase());
ret = nvs_flash_init();
}
ESP_ERROR_CHECK(ret);

ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
wifi_init_sta();

}
`

arunrb, 2020-11-16:
Hi @YouDONG-ESP ,
checking to see if you got a chance to look at this or reproduce the issue. Please let me know.

YouDONG-ESP, 2020-11-18:
aa

YouDONG-ESP, 2020-11-18:
[image]

Alvin1Zhang, 2020-12-08:
@arunrb Thanks for reporting, would you please help share if any updates for this issue? Seems the issue still can't reproduce at our side.

closed 2021-08-12 by Alvin1Zhang