espressif/esp-idf #6016: IDF claims IO20 is invalid, but ESP-PICO-V3 exposes IO20

corecode, 2020-10-22:
## Environment

- Development Kit:      none
- Module or chip used:  [ESP32-PICO-V3]
- IDF version (run ``git describe --tags`` to find it):
    // v4.1-352-g5d258af93
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         external 5V

## Problem Description

`ledc_channel_config` does not accept IO20 and reports
> ledc: ledc_channel_config(380): gpio_num argument is invalid

### Expected Behavior

IO20 can be used

### Actual Behavior

IO20 cannot be used

### Steps to reproduce

1. call `ledc_channel_config` with `gpio_num = 20`

## Fix

```diff
diff --git a/components/soc/esp32/gpio_periph.c b/components/soc/esp32/gpio_periph.c
index d8a30b0af..0a6eee285 100644
--- a/components/soc/esp32/gpio_periph.c
+++ b/components/soc/esp32/gpio_periph.c
@@ -35,7 +35,7 @@ const uint32_t GPIO_PIN_MUX_REG[GPIO_PIN_COUNT] = {
     IO_MUX_GPIO17_REG,
     IO_MUX_GPIO18_REG,
     IO_MUX_GPIO19_REG,
-    0,
+    IO_MUX_GPIO20_REG,
     IO_MUX_GPIO21_REG,
     IO_MUX_GPIO22_REG,
     IO_MUX_GPIO23_REG,
```

ginkgm, 2020-10-23:
Hello @corecode ,
according to the [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf), there is no such GPIO20 on the chip. So it's not available.

The dev board may be desinged in this way to support more modules, but for ESP32, the pin is not available even if you remove the restriction.

boarchuz, 2020-10-23:
@ginkgm 
PICO-V3 datasheet notes:
`Comparing to other ESP32 series chips, ESP32-PICO-V3 has an additional pin GPIO20`

alessandro-montanari, 2020-10-24:
I'm using the pico v3 and having the same problem. Using the same fix suggested by @corecode I can use the pin 20 just fine.

See also #387

commit 6deaefde (2021-08-29): Enable IO20 on ESP32

Some newer ESP32 variants (like ESP32-PICO-V3 and ESP32-PICO-MINI-02)
do implement this pin and it can be used as a normal GPIO.

Fixes #6016
Fixes #6837

Closes https://github.com/espressif/esp-idf/pull/6918
commit f1a024e3 (2021-12-04): Enable IO20 on ESP32

Some newer ESP32 variants (like ESP32-PICO-V3 and ESP32-PICO-MINI-02)
do implement this pin and it can be used as a normal GPIO.

Fixes #6016
Fixes #6837

Closes https://github.com/espressif/esp-idf/pull/6918

(cherry picked from commit 6deaefde69c3f068d39b76d628b05b21fd142b79)
commit e2c35ca3 (2021-12-04): Enable IO20 on ESP32

Some newer ESP32 variants (like ESP32-PICO-V3 and ESP32-PICO-MINI-02)
do implement this pin and it can be used as a normal GPIO.

Fixes #6016
Fixes #6837

Closes https://github.com/espressif/esp-idf/pull/6918

(cherry picked from commit 6deaefde69c3f068d39b76d628b05b21fd142b79)
commit aafbb8f5 (2021-12-10): Enable IO20 on ESP32

Some newer ESP32 variants (like ESP32-PICO-V3 and ESP32-PICO-MINI-02)
do implement this pin and it can be used as a normal GPIO.

Fixes #6016
Fixes #6837

Closes https://github.com/espressif/esp-idf/pull/6918

(cherry picked from commit 6deaefde69c3f068d39b76d628b05b21fd142b79)
closed 2021-08-29 by espressif-bot