espressif/esp-idf #6017: esp32s2 phy: pll_cap_ext 10

datascraps, 2020-10-23:
## Environment

- Development Kit:     ESP32-S2-SAOLA
- Module or chip used:  ESP32S2 WROOM
- IDF version (run ``git describe --tags`` to find it):
    v4.3-dev-1561-g357a27760
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: Yes vscode
- Power Supply:        tried both  USB and external 5V

## Problem Description

In the process of development both of my esp32s2 boards will no longer broadcast wifi.
I get a repeated `phy: pll_cap_ext 10` log and when they log that they should be broadcasting they dont.

Provided examples that used to work no longer work.

I have tried erasing flash, building flashing again many times, tried the reset buttons.

I'm afraid to put my code on a new device in the case I brick that one too.

Is there a way to completely reset the board, what might have I done to cause this?
I was modifying the wifi provisioning example with softap, and successfully modified it and configured a device before whatever I did to destroy my devices.

### Expected Behavior

Example wifi ap creates a wireless network

### Actual Behavior

Something in my development process seems to have broken each of my boards so that the wifi cannot be used on either board.
Every sketch I try on both boards have `phy: pll_cap_ext 10` logs forever on any example sketch I've tried and will not broadcast wifi.
Even though this was working previously.

### Code to reproduce this issue

Something in the process of modifying the wifi provisioning example caused this, but now even the vanilla example code displays the issue.

## Debug Logs

```
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0xb (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x17f4
load:0x4004c000,len:0x9d0
load:0x40050000,len:0x2e30
entry 0x4004c1ec
I (46) boot: ESP-IDF v4.3-dev-1561-g357a27760 2nd stage bootloader
I (46) boot: compile time 02:04:02
I (46) boot: chip revision: 0
I (49) boot.esp32s2: SPI Speed      : 80MHz
I (54) boot.esp32s2: SPI Mode       : DIO
I (59) boot.esp32s2: SPI Flash Size : 2MB
I (63) boot: Enabling RNG early entropy source...
I (69) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (87) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  2 factory          factory app      00 00 00010000 00100000
I (102) boot: End of partition table
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x16de4 ( 93668) map
I (135) esp_image: segment 1: paddr=0x00026e0c vaddr=0x3ffc4a20 size=0x038a0 ( 14496) load
I (139) esp_image: segment 2: paddr=0x0002a6b4 vaddr=0x40022000 size=0x00404 (  1028) load
0x40022000: _WindowOverflow4 at /home/nburn42/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (142) esp_image: segment 3: paddr=0x0002aac0 vaddr=0x40022404 size=0x05558 ( 21848) load
0x40022404: _coredump_iram_start at ??:?

I (156) esp_image: segment 4: paddr=0x00030020 vaddr=0x40080020 size=0x69410 (431120) map
0x40080020: _stext at ??:?

I (249) esp_image: segment 5: paddr=0x00099438 vaddr=0x4002795c size=0x0d0b8 ( 53432) load
0x4002795c: tx_pwctrl_background at ??:?

I (274) boot: Loaded app from partition at offset 0x10000
I (274) boot: Disabling RNG early entropy source...
I (286) cache: Instruction cache 	: size 8KB, 4Ways, cache line size 32Byte
I (286) cpu_start: Pro cpu up.
I (339) cpu_start: Pro cpu start user code
I (339) cpu_start: cpu freq: 160000000
I (340) cpu_start: Application information:
I (342) cpu_start: Project name:     wifi_softAP
I (348) cpu_start: App version:      v4.3-dev-1561-g357a27760
I (354) cpu_start: Compile time:     Oct 23 2020 02:03:59
I (360) cpu_start: ELF file SHA256:  5e328a87cca016c6...
I (366) cpu_start: ESP-IDF:          v4.3-dev-1561-g357a27760
I (373) heap_init: Initializing. RAM available for dynamic allocation:
I (380) heap_init: At 3FF9E000 len 00002000 (8 KiB): RTCRAM
I (386) heap_init: At 3FFCBEC0 len 00030140 (192 KiB): DRAM
I (392) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (399) spi_flash: detected chip: generic
I (403) spi_flash: flash io: dio
W (407) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (424) cpu_start: Starting scheduler on PRO CPU.
I (436) wifi softAP: ESP_WIFI_MODE_AP
I (456) wifi:wifi driver task: 3ffd4484, prio:23, stack:6656, core=0
I (456) system_api: Base MAC address is not set
I (456) system_api: read default base MAC address from EFUSE
I (466) wifi:wifi firmware version: a94a000
I (466) wifi:wifi certification version: v7.0
I (466) wifi:config NVS flash: enabled
I (466) wifi:config nano formating: disabled
I (476) wifi:Init data frame dynamic rx buffer num: 32
I (476) wifi:Init management frame dynamic rx buffer num: 32
I (486) wifi:Init management short buffer num: 32
I (486) wifi:Init dynamic tx buffer num: 32
I (496) wifi:Init static rx buffer size: 1600
I (496) wifi:Init static rx buffer num: 10
I (496) wifi:Init dynamic rx buffer num: 32
I (506) wifi_init: rx ba win: 6
I (506) wifi_init: tcpip mbox: 32
I (516) wifi_init: udp mbox: 6
I (516) wifi_init: tcp mbox: 6
I (516) wifi_init: tcp tx win: 5744
I (526) wifi_init: tcp rx win: 5744
I (526) wifi_init: tcp mss: 1440
I (536) wifi_init: WiFi IRAM OP enabled
I (536) wifi_init: WiFi RX IRAM OP enabled
I (546) phy: pll_cap_ext 10
I (546) phy: pll_cap_ext 10
I (546) phy: pll_cap_ext 10
I (556) phy: pll_cap_ext 10
I (556) phy: pll_cap_ext 10
I (556) phy: pll_cap_ext 10
I (566) phy: pll_cap_ext 10
I (566) phy: pll_cap_ext 10
I (566) phy: pll_cap_ext 10
I (576) phy: pll_cap_ext 10
I (576) phy: pll_cap_ext 10
I (586) phy: pll_cap_ext 10
I (596) phy: pll_cap_ext 10
I (606) phy: pll_cap_ext 10
I (616) phy: pll_cap_ext 10
I (636) phy: pll_cap_ext 10
I (656) phy: pll_cap_ext 10
I (656) phy: pll_cap_ext 10
I (656) phy: pll_cap_ext 10
I (656) phy: pll_cap_ext 10
I (656) phy: pll_cap_ext 10
I (656) phy: pll_cap_ext 10
I (666) phy: pll_cap_ext 10
I (666) phy: pll_cap_ext 10
I (666) phy: pll_cap_ext 10
I (676) phy: pll_cap_ext 10
I (676) phy: pll_cap_ext 10
I (696) phy: pll_cap_ext 10
I (716) phy: pll_cap_ext 10
I (726) phy: phy_version: 603, 72dfd77, Jul  7 2020, 19:57:05, 0, 2
I (736) wifi:mode : softAP (7c:df:a1:02:5d:75)
I (736) phy: pll_cap_ext 10
I (736) wifi:Total power save buffer number: 16
I (736) wifi:Init max length of beacon: 752/752
I (746) wifi:Init max length of beacon: 752/752
I (746) wifi softAP: wifi_init_softap finished. SSID:myssid password:mypassword channel:1
I (1156) phy: pll_cap_ext 10
I (1666) phy: pll_cap_ext 10
I (2186) phy: pll_cap_ext 10
I (2696) phy: pll_cap_ext 10
I (3206) phy: pll_cap_ext 10
I (3716) phy: pll_cap_ext 10
I (4226) phy: pll_cap_ext 10
I (4746) phy: pll_cap_ext 10
I (5256) phy: pll_cap_ext 10

```

felmue, 2020-10-23:
Hi @datascraps 

sorry to hear about your troubles. I've tried the softAP example with a fresh github clone (same as yours) on a Kaluga ESP32s2 board and it works fine. So yes, something seems to be not right with your setup I guess.
- Have you tried to make a fresh copy / clone of your development environment?
- Have you tried to increase the log level to debug (or even verbose)?
Thanks
Felix

mbuckaway, 2020-10-25:
This issue happen to be with the same board. The log filled up with the pll_cap message. It turned out, I mistakenly added 

```
esp_log_level_set("WIFI", ESP_LOG_VERBOSE);
```

To my code....turning on verbose logging (all kinds of noise) to the WIFI driver. Once I removed the line, everything worked correctly (which is what @felmue  is suggesting).

datascraps, 2020-10-27:
Thanks, 
Setting wifi logs to info, fixes the problem, and it will broadcast again.

YouDONG-ESP, 2020-10-27:
Hi @datascraps, could you show me your code or a snippet? It's may be a bug, we will take a look at it.

mbuckaway, 2020-10-27:
From what @datascraps  is saying, setting logging to VERBOSE causes the WIFI driver to stop working. Use the following code:

```
esp_log_level_set("WIFI", ESP_LOG_VERBOSE);
```

datascraps, 2020-10-27:
Removing that line did not fix the problem for me,
Instead I had to manually set wifi to INFO.

YouDONG-ESP, 2020-10-28:
Hi @datascraps, I couldn't reproduce this issue locally, but I noticed that your phy version is old.
> `I (726) phy: phy_version: 603, 72dfd77, Jul  7 2020, 19:57:05, 0, 2`
Normally, for your IDF version, phy version is 4500, 0cd6843. Try to use `git submodule update -f` to update phy and see if this issue solved.

YouDONG-ESP, 2020-10-28:
Hi @datascraps. @mbuckaway, according to your description, this issue seems to happen when phy init and a verbose log print process at the same time. 

I added esp_log_level_set("WIFI", ESP_LOG_VERBOSE) to softap example but my wifi raised correctly. 

Meanwhile, I think your esp_log_level_set is invalid because your log level is actually Info, there is no Debug or Verbose print from your log (Remember that esp_log_level_set can not raise log level above the level set using CONFIG_LOG_DEFAULT_LEVEL setting in menuconfig).

Could you check sdkconfig for your log level and provide a code to reproduce this issue?

YouDONG-ESP, 2020-10-28:
By the way, did you use ADC? ADC2 might cause problems.

mbuckaway, 2020-10-29:
I used the mqtt example as a basis for my project. That project defines a few log entries in the beginning of app_main. I moved the wifi code from app_main.c into it's own wifi.c module, and mistakenly added the log line I mentioned. That is when I noticed the noise in the log. I changed the name of my module and the log item from "WIFI" and the problem went away. I have no idea if the noise in the log prevented a wifi connection....I just found all the noise annoying and remove it.

My code can be found here: [ESP32 Project](https://github.com/mbuckaway/esp-moisture). The code should build for the ESP32S2 - which I am using.

Changing any of the esp_log_level_set lines to use WIFI with verbose, causes the issue. I was using IDF 4.2 to see the problem. Yes, my project uses the ADC, but I'm using ADC1 because ADC2 is not available with WIFI enabled.

CONFIG_LOG_DEFAULT_LEVEL is set to INFO...so no idea why using VERBOSE would cause the problem. I will mention that the noisy log entries are green - which I assume means info level.

My suggestion is to setup the MQTT example on a ESP32S2 and add the WIFI line. If this still is not reproducible, I can dig into farther.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5460425/sdkconfig.txt)

YouDONG-ESP, 2020-11-09:
Hi @mbuckaway, I run STA than MQTT example on ESP32S2 with Master version but I didn't reproduce this issue, I also tried your code by removing mqttpublish() which contains a submodule that I don't have access right. The issue can't be reproduced either. Could you give me a code to reproduce this issue with IDF commit id?

YouDONG-ESP, 2020-11-10:
@mbuckaway Could you update IDF version to see if this issue solved? our PHY has been updated a lot.

mbuckaway, 2020-11-10:
So, currently, I cannot duplicate the problem. Setting "esp_log_level_set("WIFI", ESP_LOG_VERBOSE);" has no effect. I do update the IDF randomly, and I am using a device with a ESP32S2. Both current versions of 4.2 and master seem to work correctly when setting the log level. Suggest marking this problem as closed. Anyone experiencing the problem should update the IDF.

Alvin1Zhang, 2020-11-11:
Thanks for reporting and the updates, feel free to reopen.

closed 2020-11-11 by Alvin1Zhang