espressif/esp-idf #6020: ESP32S2: twai_self_test failure with CONFIG_COMPILER_OPTIMIZATION_SIZE=y

jepler, 2020-10-23:
Hi, I'm working for Adafruit on the implementation of the `canio` module for CircuitPython using the ESP32S2's TWAI peripheral, and I've encountered a problem that's stopping me from completing the driver (https://github.com/adafruit/circuitpython/pull/3584).  I have demonstrated what appears to be the same problem but within the twai_self_test example that is bundled with esp-idf.

Software version used: esp-idf v4.3-dev-1561-g357a27760     (the latest at the time of writing; problem also affects v4.2-dev-2968-g8bc19ba89 used in circuitpython 6.0.0-rc.0)

Hardware used: ESP32-S2-Kaluga-1

Testing procedure:
  1. Prepare hardware: Install jumper wire between pins 17 and 18. (NOTE: I do not believe the pin selection is _important_ for reproducing, but it must match the menuconfig below; it's just what I happened to use)
  2. Build working configuration:
        ```
        cd examples/peripherals/twai/twai_self_test/
        git clean -dxf
        idf.py set-target esp32-s2
        idf.py menuconfig
        ```
        In "Example Configuration" set TX GPIO number to 17 and RX GPIO number to 18.  Save and exit
        ```
        idf.py flash
        ```
        Open terminal, press reset button.

        Result: Test succeeds, printing many "Msg received" log lines and finally "Driver uninstalled"

  3. Build non-working configuration (immediately following step 2):
        ```
        idf.py menuconfig
        ````            
        In "Compiler options" set Optimization Level to "Optimize for size (-Os)". save and exit
        ````            
        idf.py flash
        ````            
        Open terminal, press reset button.

        Result: Test fails, hanging after printing "TWAI Self Test: Driver started"

Note: With additional instrumentation added, it appears that when the compiler optimization is "-Os", writes to acceptance_filter.acr and .amr do not succeed, and incorrect values are read back.

Patch to add instrumentation: https://gist.github.com/1c90b383888ca40f3dbafd0b4d68ce9d

Typical instrumentation output when working properly:
```
I (325) TWAI Self Test: Driver installed
I (325) TWAI Self Test: f_config.acceptance_code = 0x00000000
I (335) TWAI Self Test: f_config.acceptance_mask = 0xffffffff
I (345) TWAI Self Test: f_config.single_filter = 0x00000001
I (345) TWAI Self Test: mode_reg = 0x0000000b
I (355) TWAI Self Test: command_reg = 0x00000000
I (355) TWAI Self Test: status_reg = 0x0000003c
I (365) TWAI Self Test: clock_divider_reg = 0x00000300
I (375) TWAI Self Test: acceptance_filter.acr[] =
I (375) TWAI Self Test:  00
I (375) TWAI Self Test:  00
I (385) TWAI Self Test:  00
I (385) TWAI Self Test:  00
I (395) TWAI Self Test: 

I (395) TWAI Self Test: acceptance_filter.amr[] =
I (395) TWAI Self Test:  ff
I (405) TWAI Self Test:  ff
I (405) TWAI Self Test:  ff
I (405) TWAI Self Test:  ff
I (415) TWAI Self Test: 
```

Typical instrumentation output when working improperly ("Optimize for size") - note incorrect values read back for acceptance_filter:
```
I (328) TWAI Self Test: Driver installed
I (328) TWAI Self Test: f_config.acceptance_code = 0x00000000
I (338) TWAI Self Test: f_config.acceptance_mask = 0xffffffff
I (348) TWAI Self Test: f_config.single_filter = 0x00000001
I (348) TWAI Self Test: mode_reg = 0x0000000b
I (358) TWAI Self Test: command_reg = 0x00000000
I (358) TWAI Self Test: status_reg = 0x0000003c
I (368) TWAI Self Test: clock_divider_reg = 0x00000300
I (378) TWAI Self Test: acceptance_filter.acr[] =
I (378) TWAI Self Test:  db
I (378) TWAI Self Test:  ff
I (388) TWAI Self Test:  0d
I (388) TWAI Self Test:  7f
I (398) TWAI Self Test: 

I (398) TWAI Self Test: acceptance_filter.amr[] =
I (398) TWAI Self Test:  7e
I (408) TWAI Self Test:  77
I (408) TWAI Self Test:  fe
I (408) TWAI Self Test:  da
I (418) TWAI Self Test: 
```

Additionally, marking *just* the function `twai_ll_set_acc_filter` in `./components/hal/esp32s2/include/hal/twai_ll.h` as `attribute((noinline,optimize("O0")))` causes the example to succeed when "Optimize for size" is selected.

Dazza0, 2021-07-15:
@jepler Thanks for reporting this issue. I've been able to reproduce the issue. It seems like the root cause is that with size optimization enabled, the compiler is attempting to write to the register field with a `s8` instruction instead of an `s32` because the register field is 8 bits wide instead of usual 32bits.

For now, disabling optimization for `twai_ll_set_acc_filter()` seems to prevent the `s8` instruction from being used. But a lot of the other register fields in TWAI are also 8 bits wide so I'd rather  the `s8` instruction not be used at all when reading/writing to the TWAI registers.

I'll rework the `twai_struct.h` and `twai_ll.h` to prevent `s8` from being used.

Dazza0, 2021-09-01:
@jepler PTAL at commit 2e0ffbd543abc0226f0dbd2f725eb891235abe02 to see if it resolves your issue. The commit should prevent `s8` access on all peripherals.

jepler, 2021-09-01:
I won't be able to test it for 2+ weeks, so you're welcome to close this up if you prefer.

Dazza0, 2021-09-23:
Closed on master branch by 874a720286b78e866684129ce18a6c948c8fa7fe and 14fe6dcaaf6d1bb9a1d1bb1d0e51a8fd184101e7 (for C++ workaround)

closed 2021-09-23 by Dazza0