espressif/esp-idf #6024: Trouble with PPPoS client - Address of function esp_handle_uart_data() become null (in esp_modem.c)

jonathandreyer, 2020-10-25:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.1
- Build System: PlatformIO (2.0.0+sha.3688b7b)
- Operating System: macOS
- Using an IDE?: VSCode with PlatformIO
- Power Supply: USB or external 5V

## Problem Description

In the example of PPPoS (client), there is something troubles and the application crash with a *Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.*. The full message is:
```
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060a30  A0      : 0x800d56de  A1      : 0x3ffbccb0  
A2      : 0x3ffbb428  A3      : 0x00000000  A4      : 0x3ffbb98c  A5      : 0x00000000  
A6      : 0x3ffbb9cc  A7      : 0x00000000  A8      : 0x800d5584  A9      : 0x3ffbcc60  
A10     : 0x3ffbb46c  A11     : 0x00000009  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3ffbb670  A15     : 0x00000009  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: e3e23c687850dc53

Backtrace: 0xfffffffd:0x3ffbccb0 |<-CORRUPTED
```

After multiple debugs with JTAG (ESP-Prog) and debug messages, I have found where the trouble is. The application crash in the method *esp_handle_uart_data* in *esp_modem.c* file. To debug the trouble, I have created a specific repository (https://github.com/jonathandreyer/esp32-sim7000) with the code which exposes the trouble. With this code, the debug log is:
```
D (7516) sim800: STATUS of module is OK
D (7516) pppos-example: Device SIM800 is power_up()
D (7516) sim800: start opening of SIM800 module
D (7516) sim800: try to sync with the module
D (7526) esp-modem: modem>>: DST: 0

E (7526) esp-modem: esp_dte_handle_line(90): no handler for line
W (7526) pppos-example: Unknow line received: DST: 0

D (7636) esp-modem: modem>>: *PSUTTZ: 20/10/25,21:41:30","+04",0

E (7636) esp-modem: esp_dte_handle_line(90): no handler for line
W (7636) pppos-example: Unknow line received: *PSUTTZ: 20/10/25,21:41:30","+04",0

D (8516) sim800: .
D (8516) esp-modem: modem<<: AT
D (8516) esp-modem: handle_uart_data #1, len 3
I (8516) esp-modem: 0x3ffbb46c   41 54 0d                                          |AT.|
D (8516) esp-modem: handle_uart_data #2, callback method is NULL
assertion "esp_dte->receive_cb" failed: file "components/modem/esp_modem.c", line 152, function: esp_handle_uart_data
abort() was called at PC 0x40119e57 on core 0

ELF file SHA256: 74d46404e09b68da

Backtrace: 0x400841e1:0x3ffbcc40 0x40084575:0x3ffbcc60 0x40119e57:0x3ffbcc80 0x400d565e:0x3ffbccb0 0x400d57fb:0x3ffbcce0 0x40085a41:0x3ffbcd10
  #0  0x400841e1:0x3ffbcc40 in invoke_abort at /Users/jonathandreyer/.platformio/packages/framework-espidf/components/esp32/panic.c:157
  #1  0x40084575:0x3ffbcc60 in abort at /Users/jonathandreyer/.platformio/packages/framework-espidf/components/esp32/panic.c:174
  #2  0x40119e57:0x3ffbcc80 in __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)
  #3  0x400d565e:0x3ffbccb0 in esp_handle_uart_data at components/modem/esp_modem.c:152 (discriminator 1)
  #4  0x400d57fb:0x3ffbcce0 in uart_event_task_entry at components/modem/esp_modem.c:172
  #5  0x40085a41:0x3ffbcd10 in vPortTaskWrapper at /Users/jonathandreyer/.platformio/packages/framework-espidf/components/freertos/port.c:143
```

### Expected Behavior

### Actual Behavior

### Steps to reproduce

In this case, I use a SIM7000E but it will probably work with SIM800 or another module.

### Code to reproduce this issue

The code to reproduce this trouble is available here: https://github.com/jonathandreyer/esp32-sim7000

Alvin1Zhang, 2020-10-26:
Thanks for reporting, we will look into.

jonathandreyer, 2020-10-26:
@Alvin1Zhang Thanks for your message. After some other tests, it is maybe related to #5737. I have updated the code in the repository, and the log when the crash appears is:
```
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7192
load:0x40078000,len:13072
ho 0 tail 12 room 4
load:0x40080400,len:3896
entry 0x40080688
I (31) boot: ESP-IDF 3.40100.200827 2nd stage bootloader
I (31) boot: compile time 22:28:53
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x190c4 (102596) map
I (153) esp_image: segment 1: paddr=0x000290ec vaddr=0x3ffb0000 size=0x0229c (  8860) load
I (157) esp_image: segment 2: paddr=0x0002b390 vaddr=0x40080000 size=0x00404 (  1028) load
I (161) esp_image: segment 3: paddr=0x0002b79c vaddr=0x40080404 size=0x0487c ( 18556) load
I (177) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x634e8 (406760) map
I (333) esp_image: segment 5: paddr=0x00093510 vaddr=0x40084c80 size=0x06680 ( 26240) load
I (351) boot: Loaded app from partition at offset 0x10000
I (351) boot: Disabling RNG early entropy source...
I (351) cpu_start: Pro cpu up.
I (355) cpu_start: Application information:
I (360) cpu_start: Project name:     sim800
I (364) cpu_start: App version:      72dc5a1-dirty
I (370) cpu_start: Compile time:     Oct 26 2020 22:23:21
I (376) cpu_start: ELF file SHA256:  15a78295b92cc453...
I (382) cpu_start: ESP-IDF:          3.40100.200827
I (388) cpu_start: Starting app cpu, entry point is 0x40081230
I (0) cpu_start: App cpu up.
I (398) heap_init: Initializing. RAM available for dynamic allocation:
I (405) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (411) heap_init: At 3FFB3D30 len 0002C2D0 (176 KiB): DRAM
I (417) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (423) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (430) heap_init: At 4008B300 len 00014D00 (83 KiB): IRAM
I (436) cpu_start: Pro cpu start user code
I (455) spi_flash: detected chip: generic
I (455) spi_flash: flash io: dio
I (455) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (466) uart: queue free spaces: 30
D (466) esp-modem: flow control is DISABLE
D (476) pppos-example: Device SIM800 is init()
D (476) sim800: start power-up SIM800 module
D (486) sim800: launch STATUS reading
D (986) sim800: .
D (986) sim800: STATUS is 0, inc is 1
D (1486) sim800: .
D (1486) sim800: STATUS is 0, inc is 2
D (1986) sim800: .
D (1986) sim800: STATUS is 0, inc is 3
D (2486) sim800: .
D (2486) sim800: STATUS is 0, inc is 4
D (2986) sim800: .
D (2986) sim800: STATUS is 0, inc is 5
[... 45 lines cut ...]
W (8406) pppos-example: Unknow line received: *PSUTTZ: 20/10/26,22:05:42","+04",0

D (9016) sim800: .
D (9016) esp-modem: modem<<: AT
D (9066) esp-modem: handle_uart_data #1 / esp_dte->receive_cb_ctx : 0x00000000
D (9066) esp-modem: handle_uart_data #2 / length : 9, esp_dte->receive_cb_ctx : 0x00000000
D (9066) esp-modem: handle_uart_data #3 / length : 9
D (9066) esp-modem: handle_uart_data #4 / length : 9, esp_dte->receive_cb_ctx : 0x00000000
D (9076) esp-modem: 0x3ffbb46c   41 54 0d 0d 0a 4f 4b 0d  0a                       |AT...OK..|
D (9086) esp-modem: handle_uart_data #5 / callback method is NULL
assertion "esp_dte->receive_cb" failed: file "components/modem/esp_modem.c", line 155, function: esp_handle_uart_data
abort() was called at PC 0x40119eab on core 0

ELF file SHA256: 15a78295b92cc453

Backtrace: 0x400841e1:0x3ffbcc30 0x40084575:0x3ffbcc50 0x40119eab:0x3ffbcc70 0x400d2fc6:0x3ffbcca0 0x400d3163:0x3ffbcce0 0x40085a41:0x3ffbcd10
  #0  0x400841e1:0x3ffbcc30 in invoke_abort at /Users/jonathandreyer/.platformio/packages/framework-espidf/components/esp32/panic.c:157
  #1  0x40084575:0x3ffbcc50 in abort at /Users/jonathandreyer/.platformio/packages/framework-espidf/components/esp32/panic.c:174
  #2  0x40119eab:0x3ffbcc70 in __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-apple-darwin12/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)
  #3  0x400d2fc6:0x3ffbcca0 in esp_handle_uart_data at components/modem/esp_modem.c:155 (discriminator 1)
  #4  0x400d3163:0x3ffbcce0 in uart_event_task_entry at components/modem/esp_modem.c:175
  #5  0x40085a41:0x3ffbcd10 in vPortTaskWrapper at /Users/jonathandreyer/.platformio/packages/framework-espidf/components/freertos/port.c:143

Rebooting...
```

david-cermak, 2020-10-27:
Hi @jonathandreyer 

Thanks for linking with the #5737,  indeed this might look like the same issue, but I don't think it is.
Here we're getting the data event and subsequent crash *before* setting up the netif https://github.com/jonathandreyer/esp32-sim7000/blob/master/src/main.c#L275 (that's why we ignore them in [v4.2](https://github.com/espressif/esp-idf/blob/f0e87c933ac2ee47640b452dacb7d10343b51eb4/examples/protocols/pppos_client/components/modem/src/esp_modem.c#L142) ) Seems like we haven't backported some important fixes.

Could you please try to update to v4.2 and test again?

jonathandreyer, 2020-10-29:
@david-cermak Thanks for your feedback. I am not sure that I can migrate to v4.2 because PlatformIO support only stable version of esp-idf. It is a bit strange because it doesn’t happen every boot.
I will see what I can with v4.2 do and I will back to you.

david-cermak, 2020-11-03:
@jonathandreyer Could you please try to apply this patch? (not sure if it's possible with PlatformIO, but I think the IDF is supplied as a source so should be doable before rebuilding, the change is against `v4.1`)
[0001-esp_modem-Fix-possible-early-data-event-when-DCE-PPP.patch.txt](https://github.com/espressif/esp-idf/files/5483950/0001-esp_modem-Fix-possible-early-data-event-when-DCE-PPP.patch.txt)

jonathandreyer, 2020-11-03:
@david-cermak Thanks for your this patch. I have taken a look inside the patch and yes, I can apply it to the example (It doesn't touch the ESP-IDF source). I will try that tomorrow.

jonathandreyer, 2020-11-04:
@david-cermak I have tested the patch and that seems fixed crashes. But, with this patch, I have the feeling that another trouble has appeared (at the end of the example of the exit of PPP mode).
Do you think that patch really fixes the trouble or it is only a temporary patch? In fact, it is quite clear that this handle _esp_handle_uart_data_ should not be called at this point, right?

david-cermak, 2020-11-05:
@jonathandreyer Indeed, we shouldn't handle uart data before the DCE/PPP is ready, this is just a temporary patch.

As for the other trouble, I don't think it's related to this change. There's unrelated problem with exiting the PPP mode (see the linked issue), the patch posted [here](https://github.com/espressif/esp-idf/issues/6005#issuecomment-717043807) should fix this.

jonathandreyer, 2020-11-05:
@david-cermak I have tested your new proposition, but the other patch for v4.1 (because I use v4.1 of ESP-IDF), and it doesn't fix the new trouble (crash when PPP exit). The core of example (PPP connexion, MQTT publishing, etc.) doesn't work.
I have created a new branch without all debug messages that I had added. The source code is as similar as possible from the example. In fact, I have leave the custom part to initiate the simcom module. With this new version, without any patch, the behavior is the same which is described in the issue. When I apply the first patch, it is seems fix the initial trouble without generating other trouble.

jonathandreyer, 2020-11-07:
@david-cermak @Alvin1Zhang I have started to investigate to understand the reason why this initial trouble appear sometimes.
When the crash appears (or when the temporary fix proposed by @david-cermak is needed), I have the feeling that the generated what event type is not what is expected. It is here: https://github.com/espressif/esp-idf/blob/5d258af93c668d748a5d190137c1bf8ae58489b4/examples/protocols/pppos_client/components/modem/src/esp_modem.c#L160
In the initialization part (before switch int the PPP mode), this trouble implies that the wrong function to parse received data is called. The method, _esp_handle_uart_data(esp_dte)_ is executed and I have the feeling that at this point of execution, the method _esp_handle_uart_pattern(esp_dte)_ should be called. Moreover, I don't fully understand why there are those two different types of event (data & pattern). 

Maybe, the temporary patch generates also another trouble and ignore some response from the modem (because it is ignore the parsing method of receiving data without processing).

jonathandreyer, 2020-11-08:
@david-cermak @Alvin1Zhang Good news, I have found a solution to fix the solution. I have applied commits 0784426, 745f062 & 6f1b8fe. Probably, it is the second one which is useful with the disabling of *RX interrupt* ([esp_modem.c#L440](https://github.com/espressif/esp-idf/blob/494a124d961c4c755685b22fe986c31826d6f503/examples/protocols/pppos_client/components/modem/src/esp_modem.c#L440)). After some tests (>10 executions), it is working without trouble. I have created a branch with the edited source code on v4.1. It can be seen [here](https://github.com/jonathandreyer/esp32-sim7000/tree/espressif-fix).
I have also updated the PPPoS example (from v4.1) with all possible commits (It can be seen [here](https://github.com/jonathandreyer/esp32-sim7000/tree/espressif-others-patches)). In this example, the program loop every 60sec (in my case I use 10sec). Unfortunately, a trouble appears in the PPP exit part. At the beginning of the second execution, there is the message _esp-modem: Error: Got data event in PPP mode_. Maybe I can open a second issue because that seems a bit different than the first issue. The log can be seen below. 
I have tested with [PR #4849] but it is not resolving the error. Have you already seen those kind of trouble?

Output log:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7192
load:0x40078000,len:13072
ho 0 tail 12 room 4
load:0x40080400,len:3896
entry 0x40080688
I (30) boot: ESP-IDF 3.40100.200827 2nd stage bootloader
I (30) boot: compile time 18:23:06
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x18628 ( 99880) map
I (152) esp_image: segment 1: paddr=0x00028650 vaddr=0x3ffb0000 size=0x0229c (  8860) load
I (156) esp_image: segment 2: paddr=0x0002a8f4 vaddr=0x40080000 size=0x00404 (  1028) load
I (159) esp_image: segment 3: paddr=0x0002ad00 vaddr=0x40080404 size=0x05318 ( 21272) load
I (177) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x62c08 (404488) map
I (331) esp_image: segment 5: paddr=0x00092c30 vaddr=0x4008571c size=0x05be4 ( 23524) load
I (348) boot: Loaded app from partition at offset 0x10000
I (348) boot: Disabling RNG early entropy source...
I (348) cpu_start: Pro cpu up.
I (352) cpu_start: Application information:
I (357) cpu_start: Project name:     sim800
I (362) cpu_start: App version:      21b9bed-dirty
I (367) cpu_start: Compile time:     Nov  8 2020 18:22:13
I (373) cpu_start: ELF file SHA256:  995660aae2d1dd9a...
I (379) cpu_start: ESP-IDF:          3.40100.200827
I (385) cpu_start: Starting app cpu, entry point is 0x40081230
I (0) cpu_start: App cpu up.
I (395) heap_init: Initializing. RAM available for dynamic allocation:
I (402) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (408) heap_init: At 3FFB3D30 len 0002C2D0 (176 KiB): DRAM
I (414) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (421) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (427) heap_init: At 4008B300 len 00014D00 (83 KiB): IRAM
I (433) cpu_start: Pro cpu start user code
I (452) spi_flash: detected chip: gd
I (452) spi_flash: flash io: dio
I (452) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (460) pppos_example: APN settings: <APN_SETTING>
I (470) uart: queue free spaces: 30
E (5920) esp-modem: esp_dte_handle_line(106): no handler for line
W (5920) pppos_example: Unknow line received: RDY

E (6470) esp-modem: esp_dte_handle_line(106): no handler for line
W (6470) pppos_example: Unknow line received: +CFUN: 1

E (6600) esp-modem: esp_dte_handle_line(106): no handler for line
[... 89 lines cut ...]
I (29230) pppos_example: Gateway     : <IP_ADDR>
I (29240) pppos_example: Name Server1: <IP_ADDR>
I (29240) pppos_example: Name Server2: <IP_ADDR>
I (29250) pppos_example: ~~~~~~~~~~~~~~
I (29250) pppos_example: GOT ip event!!!
I (29260) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (29270) pppos_example: MQTT other event id: 7
I (29570) pppos_example: MQTT_EVENT_CONNECTED
I (29580) pppos_example: sent subscribe successful, msg_id=20511
I (29770) pppos_example: MQTT_EVENT_SUBSCRIBED, msg_id=20511
I (29770) pppos_example: sent publish successful, msg_id=0
I (29930) pppos_example: MQTT_EVENT_DATA
TOPIC=/topic/esp-pppos
DATA=esp32-pppos
I (30130) pppos_example: Modem PPP Stopped
I (30180) esp-netif_lwip-ppp: User interrupt
I (30590) pppos_example: Power down
I (30590) pppos_example: Restart after 10 seconds
E (37430) esp-modem: esp_dte_handle_line(101): DTE has not yet bind with DCE
E (37480) esp-modem: esp_dte_handle_line(101): DTE has not yet bind with DCE
E (37980) esp-modem: esp_dte_handle_line(101): DTE has not yet bind with DCE
...
..
.
```

david-cermak, 2020-11-11:
@jonathandreyer  Thanks for the investigation. I've prepared a fix to backport these fixes to `v4.1`:

* 8f0ec8bd50  pppos client: modem netif fix ignoring potential modem-start error
* 6d70b7c352 fix esp_modem switch mode timeout
* 0784426d6b esp_modem: Ensure uart_param_config and uart pins are set before uart_driver_install
* 745f062e07 esp_modem: pppos_client modem to use uart with REF_TICK
* fffdc1d789 esp-netif-ppp: support for setting ppp netif up and down

As for the PPP exit issue, the patch [here](https://github.com/espressif/esp-idf/issues/6005#issuecomment-717043807) should effectively solve it (The linked PR might help in some cases, but need to change to app code to wait for that event).

jonathandreyer, 2020-11-11:
@david-cermak Thanks for your feedback. I have applied the patch ([here](https://github.com/jonathandreyer/esp32-sim7000/tree/espressif-fix-ppp-exit)), but another error has occurred. The log is below and it seems to be related to the size of the Pattern Queue. When I disable the line *res = uart_set_rx_timeout(esp_dte->uart_port, 1);*, the warning *esp-modem: Pattern Queue Size too small* didn't appear but it is blocked at the same place.

```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7192
load:0x40078000,len:13072
ho 0 tail 12 room 4
load:0x40080400,len:3896
entry 0x40080688
I (31) boot: ESP-IDF 3.40100.200827 2nd stage bootloader
I (31) boot: compile time 22:13:03
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x187a8 (100264) map
I (152) esp_image: segment 1: paddr=0x000287d0 vaddr=0x3ffb0000 size=0x0229c (  8860) load
I (156) esp_image: segment 2: paddr=0x0002aa74 vaddr=0x40080000 size=0x00404 (  1028) load
I (160) esp_image: segment 3: paddr=0x0002ae80 vaddr=0x40080404 size=0x05198 ( 20888) load
I (177) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x62e30 (405040) map
I (332) esp_image: segment 5: paddr=0x00092e58 vaddr=0x4008559c size=0x05d64 ( 23908) load
I (348) boot: Loaded app from partition at offset 0x10000
I (349) boot: Disabling RNG early entropy source...
I (349) cpu_start: Pro cpu up.
I (353) cpu_start: Application information:
I (357) cpu_start: Project name:     sim800
I (362) cpu_start: App version:      21b9bed-dirty
I (368) cpu_start: Compile time:     Nov 11 2020 22:12:15
I (374) cpu_start: ELF file SHA256:  58c1b3114941ccef...
I (380) cpu_start: ESP-IDF:          3.40100.200827
I (385) cpu_start: Starting app cpu, entry point is 0x40081230
I (0) cpu_start: App cpu up.
I (396) heap_init: Initializing. RAM available for dynamic allocation:
I (403) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (409) heap_init: At 3FFB3D30 len 0002C2D0 (176 KiB): DRAM
I (415) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (421) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (428) heap_init: At 4008B300 len 00014D00 (83 KiB): IRAM
I (434) cpu_start: Pro cpu start user code
I (452) spi_flash: detected chip: gd
I (453) spi_flash: flash io: dio
I (453) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (461) pppos_example: APN settings: <APN_SETTING>
I (471) uart: queue free spaces: 30
E (6011) esp-modem: esp_dte_handle_line(108): no handler for line
W (6011) pppos_example: Unknow line received: RDY

E (6471) esp-modem: esp_dte_handle_line(108): no handler for line
W (6471) pppos_example: Unknow line received: +CFUN: 1

E (6771) esp-modem: esp_dte_handle_line(108): no handler for line
W (6771) pppos_example: Unknow line received: +CPIN: READY

E (7171) esp-modem: esp_dte_handle_line(108): no handler for line
[... 53 lines cut ...]
W (14551) esp-modem: Pattern Queue Size too small
E (17111) esp-modem: esp_modem_dte_send_cmd(247): process command timeout
W (17111) sim800: Sending "+++" command failed
W (17201) esp-modem: Pattern Queue Size too small
W (17401) esp-modem: Pattern Queue Size too small
E (17611) esp-modem: esp_modem_dte_send_cmd(247): process command timeout
E (17611) sim800: sim800_set_working_mode(303): send command failed
E (17611) esp-modem: esp_modem_dte_change_mode(337): set new working mode:0 failed
E (17621) esp-modem: esp_modem_stop_ppp(529): enter command mode failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40084790
file: "src/main.c" line 309
func: app_main
expression: esp_modem_stop_ppp(dte)

ELF file SHA256: 58c1b3114941ccef

Backtrace: 0x400841e1:0x3ffb5a90 0x40084793:0x3ffb5ab0 0x400d2da2:0x3ffb5ad0 0x400db3fe:0x3ffb5c00 0x40085a41:0x3ffb5c20
  #0  0x400841e1:0x3ffb5a90 in invoke_abort at /Users/<USER>/.platformio/packages/framework-espidf/components/esp32/panic.c:157
  #1  0x40084793:0x3ffb5ab0 in _esp_error_check_failed at /Users/<USER>/.platformio/packages/framework-espidf/components/esp32/panic.c:740
  #2  0x400d2da2:0x3ffb5ad0 in app_main at src/main.c:309 (discriminator 1)
  #3  0x400db3fe:0x3ffb5c00 in main_task at /Users/<USER>/.platformio/packages/framework-espidf/components/esp32/cpu_start.c:565
  #4  0x40085a41:0x3ffb5c20 in vPortTaskWrapper at /Users/<USER>/.platformio/packages/framework-espidf/components/freertos/port.c:143

Rebooting...
```

david-cermak, 2020-11-12:
Well, I thought the patch should address all PPP-exit related issues (at least for the devices I played with), but apparently it didn't. Thanks for sharing your results, clearly seeing the problem in the log that:
```
I (12101) esp-modem-netif: PPP state changed event 5
```
here we exit the ppp client SM, but the further warnings of `Pattern Queue Size too small` indicate the server side of pppd continues (probably some retry/reconnect on the modem side) and trying to send some data. This also causes trouble sending the `+++` (exit data mode) command, as it probably gets mixed with other data and is ignored.

One way to work this around is to send `+++` command while the pppd is still active (as I originally suggested in [the linked issue](https://github.com/espressif/esp-idf/issues/6005) ). This would also brings about another trouble, that is we'd need to `resume` the data mode using different command (maybe not an issue here at `v4.1` as the example doesn't periodically enter/exit data mode)

The ultimate solution is probably using cmux protocol to multiplex terminals, so that one is always in command mode, the other one in the data mode...

I will test the `+++`/ `ATO` option with some other devices soon and hopefully come up with a better fix. Sorry for the inconvenience.

jonathandreyer, 2020-11-12:
@david-cermak Thanks for your reply and your help. If it will be helpful for you, I can investigate the data which are transferred between the chip and the modem (with a logic analyzer).
From my side, I will investigate to test the example with another version of ESP-IDF (v4.2 and master branch). 

Concerning the CMUX, indeed it seems to be an excellent idea and as explain [here](https://github.com/espressif/esp-idf/issues/4688), it will be a huge improvement. It seems that @d3vil-st have created a PoC. If I can do anything, please do not hesitate.

david-cermak, 2020-11-13:
@jonathandreyer I've sketched up a fix using the `+++` / `ATO` option. Could you please test it with your device? This actually exits the data mode *before* the PPP netif closes connection. This seems to me like a viable alternative. (the patch also logs these data in the debug verbosity, no need for a logic analyzer)

[0001-WIP-test-exit-ppp-mode.patch.txt](https://github.com/espressif/esp-idf/files/5537520/0001-WIP-test-exit-ppp-mode.patch.txt)

jonathandreyer, 2020-11-13:
@david-cermak Thanks for the patch, but I didn't understand on which code I need to apply it. It is an incremental version of [this](https://github.com/espressif/esp-idf/issues/6024#issuecomment-725433183) patch or another patch?

jonathandreyer, 2020-11-13:
@david-cermak After some investigation, I have applied the patch to the latest version of example (master branch). Unfortunately, there are some trouble, but I have the feeling that there is a progression. The PPP mode seems to be stopped correctly, but it crashes [here](https://github.com/jonathandreyer/esp32-sim7000/blob/f89d33888d4c1166858d964ada0dd0840c9cf1af/components/modem/esp_modem.c#L282).

Output log:
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7192
load:0x40078000,len:13072
ho 0 tail 12 room 4
load:0x40080400,len:3896
entry 0x40080688
I (31) boot: ESP-IDF 3.40100.200827 2nd stage bootloader
I (31) boot: compile time 17:27:03
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x18728 (100136) map
I (152) esp_image: segment 1: paddr=0x00028750 vaddr=0x3ffb0000 size=0x0229c (  8860) load
I (156) esp_image: segment 2: paddr=0x0002a9f4 vaddr=0x40080000 size=0x00404 (  1028) load
I (159) esp_image: segment 3: paddr=0x0002ae00 vaddr=0x40080404 size=0x05218 ( 21016) load
I (177) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x62f68 (405352) map
I (332) esp_image: segment 5: paddr=0x00092f90 vaddr=0x4008561c size=0x05ce4 ( 23780) load
I (349) boot: Loaded app from partition at offset 0x10000
I (349) boot: Disabling RNG early entropy source...
I (349) cpu_start: Pro cpu up.
I (353) cpu_start: Application information:
I (357) cpu_start: Project name:     sim800
I (362) cpu_start: App version:      2255505-dirty
I (368) cpu_start: Compile time:     Nov 13 2020 17:26:08
I (374) cpu_start: ELF file SHA256:  3baa86396ac095d2...
I (380) cpu_start: ESP-IDF:          3.40100.200827
I (385) cpu_start: Starting app cpu, entry point is 0x40081230
I (377) cpu_start: App cpu up.
I (396) heap_init: Initializing. RAM available for dynamic allocation:
I (403) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (409) heap_init: At 3FFB3D30 len 0002C2D0 (176 KiB): DRAM
I (415) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (422) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (428) heap_init: At 4008B300 len 00014D00 (83 KiB): IRAM
I (434) cpu_start: Pro cpu start user code
I (453) spi_flash: detected chip: gd
I (453) spi_flash: flash io: dio
I (453) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (461) pppos_example: APN settings: <APN_SETTING>
I (471) uart: queue free spaces: 30
E (891) esp-modem: esp_dte_handle_line(106): no handler for line
W (891) pppos_example: Unknow line received: RDY

E (1351) esp-modem: esp_dte_handle_line(106): no handler for line
W (1351) pppos_example: Unknow line received: +CFUN: 1

E (1451) esp-modem: esp_dte_handle_line(106): no handler for line
W (1451) pppos_example: Unknow line received: +CPIN: READY

[... 52 lines cut ...]
Core 0 register dump:
PC      : 0x400d5980  PS      : 0x00060030  A0      : 0x8013023c  A1      : 0x3ffb8b90  
A2      : 0x3ffb9f20  A3      : 0x3ffbdb10  A4      : 0x0000001d  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffb8b50  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffb787c  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x3ffbbc54  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000064  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 3baa86396ac095d2

Backtrace: 0x400d597d:0x3ffb8b90 0x40130239:0x3ffb8bc0 0x4013030b:0x3ffb8be0 0x400dd8b1:0x3ffb8c00 0x400f313c:0x3ffb8c20 0x400f32c5:0x3ffb8c40 0x401307c9:0x3ffb8c70 0x400faad7:0x3ffb8c90 0x400fab2f:0x3ffb8cc0 0x400e99c5:0x3ffb8ce0 0x400e149e:0x3ffb8d00 0x400e1550:0x3ffb8d20 0x40085a41:0x3ffb8d50
  #0  0x400d597d:0x3ffb8b90 in esp_modem_dte_send_data at components/modem/esp_modem.c:282
  #1  0x40130239:0x3ffb8bc0 in esp_modem_dte_transmit at components/modem/esp_modem_netif.c:42
  #2  0x4013030b:0x3ffb8be0 in esp_netif_transmit at /Users/<USER>/.platformio/packages/framework-espidf/components/esp_netif/lwip/esp_netif_lwip.c:704
  #3  0x400dd8b1:0x3ffb8c00 in pppos_low_level_output at /Users/<USER>/.platformio/packages/framework-espidf/components/esp_netif/lwip/esp_netif_lwip_ppp.c:228
  #4  0x400f313c:0x3ffb8c20 in pppos_output_last at /Users/<USER>/.platformio/packages/framework-espidf/components/lwip/lwip/src/netif/ppp/pppos.c:878
  #5  0x400f32c5:0x3ffb8c40 in pppos_write at /Users/<USER>/.platformio/packages/framework-espidf/components/lwip/lwip/src/netif/ppp/pppos.c:244
[... cut ...]
  #9  0x400e99c5:0x3ffb8ce0 in sys_check_timeouts at /Users/<USER>/.platformio/packages/framework-espidf/components/lwip/lwip/src/core/timeouts.c:400
  #10 0x400e149e:0x3ffb8d00 in tcpip_timeouts_mbox_fetch at /Users/<USER>/.platformio/packages/framework-espidf/components/lwip/lwip/src/api/tcpip.c:115
  #11 0x400e1550:0x3ffb8d20 in tcpip_thread at /Users/<USER>/.platformio/packages/framework-espidf/components/lwip/lwip/src/api/tcpip.c:148
  #12 0x40085a41:0x3ffb8d50 in vPortTaskWrapper at /Users/<USER>/.platformio/packages/framework-espidf/components/freertos/port.c:143
```

david-cermak, 2020-11-18:
@jonathandreyer Sorry that wasn't very clear, the patch was meant to be applied to `v4.1` (plus adding those little fixes mentioned [here](https://github.com/espressif/esp-idf/issues/6024#issuecomment-725433183) -- which makes the modem component virtually the same as `master`).

Thanks for testing this! yes, I can reproduce the same crash, the reason is that after we deinit the DTE, PPP still running and trying to transmit data to `nullptr` DTE in `esp_modem_dte_send_data()`. Now we just have to combine these two methods together to wait until PPP closes (make the `modem_exit_ppp()` to block until the app is able to deinitialize it).

Posting a patch against master and hope it finally helps:
[0001-esp_modem-Fixed-race-condition-on-exiting-PPP-mode.patch.txt](https://github.com/espressif/esp-idf/files/5559210/0001-esp_modem-Fixed-race-condition-on-exiting-PPP-mode.patch.txt)

jonathandreyer, 2020-11-20:
@david-cermak Thanks for the patch. I have tested, but unfortunately it is worst. They are two kinds of error: PPP connection is lost at the beginning or some error with MQTT client. The code (with your patch applied) can be seen [here](https://github.com/jonathandreyer/esp32-sim7000/tree/espressif-fix3-ppp-exit).

Output log (when the connection is losted):
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7192
load:0x40078000,len:13072
ho 0 tail 12 room 4
load:0x40080400,len:3896
entry 0x40080688
I (31) boot: ESP-IDF 3.40100.200827 2nd stage bootloader
I (31) boot: compile time 11:11:34
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x187b8 (100280) map
I (152) esp_image: segment 1: paddr=0x000287e0 vaddr=0x3ffb0000 size=0x0229c (  8860) load
I (156) esp_image: segment 2: paddr=0x0002aa84 vaddr=0x40080000 size=0x00404 (  1028) load
I (160) esp_image: segment 3: paddr=0x0002ae90 vaddr=0x40080404 size=0x05188 ( 20872) load
I (177) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x62f18 (405272) map
I (332) esp_image: segment 5: paddr=0x00092f40 vaddr=0x4008558c size=0x05d74 ( 23924) load
I (349) boot: Loaded app from partition at offset 0x10000
I (349) boot: Disabling RNG early entropy source...
I (349) cpu_start: Pro cpu up.
I (353) cpu_start: Application information:
I (357) cpu_start: Project name:     sim800
I (362) cpu_start: App version:      575e996-dirty
I (368) cpu_start: Compile time:     Nov 20 2020 11:10:46
I (374) cpu_start: ELF file SHA256:  a2353556a9ca6338...
I (380) cpu_start: ESP-IDF:          3.40100.200827
I (385) cpu_start: Starting app cpu, entry point is 0x40081230
I (0) cpu_start: App cpu up.
I (396) heap_init: Initializing. RAM available for dynamic allocation:
I (403) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (409) heap_init: At 3FFB3D30 len 0002C2D0 (176 KiB): DRAM
I (415) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (421) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (428) heap_init: At 4008B300 len 00014D00 (83 KiB): IRAM
I (434) cpu_start: Pro cpu start user code
I (452) spi_flash: detected chip: gd
I (453) spi_flash: flash io: dio
I (453) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (461) pppos_example: APN settings: <APN_SETTING>
I (471) uart: queue free spaces: 30
E (5981) esp-modem: esp_dte_handle_line(107): no handler for line
W (5981) pppos_example: Unknow line received: RDY

E (6441) esp-modem: esp_dte_handle_line(107): no handler for line
W (6441) pppos_example: Unknow line received: +CFUN: 1

E (6741) esp-modem: esp_dte_handle_line(107): no handler for line
W (6741) pppos_example: Unknow line received: +CPIN: READY

[... 143 lines cut ...]
I (12370) pppos_example: Name Server2: <IP_ADDR>
I (12380) pppos_example: ~~~~~~~~~~~~~~
I (12380) pppos_example: GOT ip event!!!
I (12390) pppos_example: PPP state changed event 266
I (12400) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (12410) pppos_example: MQTT other event id: 7
E (25470) MQTT_CLIENT: Error transport connect
I (25470) pppos_example: MQTT_EVENT_ERROR
I (25470) pppos_example: MQTT_EVENT_DISCONNECTED
I (40470) pppos_example: MQTT other event id: 7
E (54470) MQTT_CLIENT: Error transport connect
I (54470) pppos_example: MQTT_EVENT_ERROR
I (54470) pppos_example: MQTT_EVENT_DISCONNECTED
I (69470) pppos_example: MQTT other event id: 7
E (83470) MQTT_CLIENT: Error transport connect
I (83470) pppos_example: MQTT_EVENT_ERROR
I (83470) pppos_example: MQTT_EVENT_DISCONNECTED
I (98470) pppos_example: MQTT other event id: 7
E (112470) MQTT_CLIENT: Error transport connect
I (112470) pppos_example: MQTT_EVENT_ERROR
I (112470) pppos_example: MQTT_EVENT_DISCONNECTED
...
..
.
```

david-cermak, 2020-11-26:
@jonathandreyer Thanks for testing this. This is hard to believe as the patch should have no impact on network mode, it only adjusts switching between modes. Could you please post results when logging set to `debug` level (this might have possibly broken the PPP mode, if so we'd see debug data which were not send). Also could you please double-check that the connection issues are really caused by the changes? (reverting, testing, reapplying, testing, etc... I think you've already done so, just checking anyway, since this works on my end even after intensive testing and with applying arbitrary delays...)

jonathandreyer, 2020-11-27:
@david-cermak Thanks for your time. You have right, it is strange but I have strictly apply the patch.
I will double check (with and without) this patch and be back with feedback (and output logs).
Have you an idea when all patches will be available on the master branch because I would like to be sure that I didn't miss anything?

david-cermak, 2020-11-27:
> Have you an idea when all patches will be available on the master branch

This [patch](https://github.com/espressif/esp-idf/issues/6024#issuecomment-729568712) (Fixed exit from PPP mode) is the last outstanding issue, all other fixes have been merged already (might still pending publishing to GitHub). I've just postponed getting this merged to master until we're sure there's no side effect, such as the errors you've reported here.

jonathandreyer, 2020-11-27:
OK, I was obviously talking about the other patches because the one mentioned is actually in progress.

jonathandreyer, 2020-12-06:
@david-cermak I have investigated to understand the reason and I have found the explanation. I have two SIM cards (one with unlimited plan and another with IoT plan) and I have used the wrong SIM card that explains the connection issue.

I have tested again with the correct SIM and all published fixes (but without your latest patch). The code used is available [here](https://github.com/jonathandreyer/esp32-sim7000/tree/2255505a00f29b143d200d73b8b14ad2277b9599). The output log is:
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7192
load:0x40078000,len:13072
ho 0 tail 12 room 4
load:0x40080400,len:3896
entry 0x40080688
I (31) boot: ESP-IDF 3.40100.200827 2nd stage bootloader
I (31) boot: compile time 19:40:12
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x186b8 (100024) map
I (152) esp_image: segment 1: paddr=0x000286e0 vaddr=0x3ffb0000 size=0x0229c (  8860) load
I (156) esp_image: segment 2: paddr=0x0002a984 vaddr=0x40080000 size=0x00404 (  1028) load
I (159) esp_image: segment 3: paddr=0x0002ad90 vaddr=0x40080404 size=0x05288 ( 21128) load
I (177) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x62cd0 (404688) map
I (331) esp_image: segment 5: paddr=0x00092cf8 vaddr=0x4008568c size=0x05c74 ( 23668) load
I (348) boot: Loaded app from partition at offset 0x10000
I (348) boot: Disabling RNG early entropy source...
I (349) cpu_start: Pro cpu up.
I (352) cpu_start: Application information:
I (357) cpu_start: Project name:     sim800
I (362) cpu_start: App version:      2255505-dirty
I (368) cpu_start: Compile time:     Dec  6 2020 19:39:25
I (374) cpu_start: ELF file SHA256:  7e0cbc0a38737daf...
I (380) cpu_start: ESP-IDF:          3.40100.200827
I (385) cpu_start: Starting app cpu, entry point is 0x40081230
I (0) cpu_start: App cpu up.
I (396) heap_init: Initializing. RAM available for dynamic allocation:
I (402) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (409) heap_init: At 3FFB3D30 len 0002C2D0 (176 KiB): DRAM
I (415) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (421) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (428) heap_init: At 4008B300 len 00014D00 (83 KiB): IRAM
I (434) cpu_start: Pro cpu start user code
I (452) spi_flash: detected chip: gd
I (453) spi_flash: flash io: dio
I (453) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (460) pppos_example: APN settings: <APN_SETTING>
I (470) uart: queue free spaces: 30
E (5980) esp-modem: esp_dte_handle_line(106): no handler for line
W (5980) pppos_example: Unknow line received: RDY

E (6440) esp-modem: esp_dte_handle_line(106): no handler for line
W (6440) pppos_example: Unknow line received: +CFUN: 1

E (6740) esp-modem: esp_dte_handle_line(106): no handler for line
W (6740) pppos_example: Unknow line received: +CPIN: READY

[... 374 lines cut ...]
I (25441) pppos_example: PPP state changed event 256
I (25441) esp-netif_lwip-ppp: User interrupt
I (25441) pppos_example: PPP state changed event 5
I (25451) pppos_example: User interrupted event from netif:0x3ffbff40
I (25451) esp-modem-netif: PPP state changed event 5
D (25491) sim800: power down ok
I (25491) pppos_example: Power down
I (25491) pppos_example: Restart after 10 seconds
W (25541) esp-modem: Pattern not found in the pattern queue, uart data length = 0
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d5721  PS      : 0x00060a30  A0      : 0x800d57ce  A1      : 0x3ffbb740  
A2      : 0x3ffb9f08  A3      : 0x00000000  A4      : 0x3ffba478  A5      : 0x00000000  
A6      : 0x3ffba4b0  A7      : 0x00000003  A8      : 0x00000000  A9      : 0x3ffbb710  
A10     : 0x00000000  A11     : 0x3ffbb740  A12     : 0x3ffbae24  A13     : 0x00000000  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000064  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 38aad0446d23fd90

Backtrace: 0x400d571e:0x3ffbb740 0x400d57cb:0x3ffbb770 0x40085a41:0x3ffbb7a0
  #0  0x400d571e:0x3ffbb740 in esp_handle_uart_data at components/modem/esp_modem.c:165
  #1  0x400d57cb:0x3ffbb770 in uart_event_task_entry at components/modem/esp_modem.c:206
  #2  0x40085a41:0x3ffbb7a0 in vPortTaskWrapper at /Users/<USER>/.platformio/packages/framework-espidf/components/freertos/port.c:143
```

david-cermak, 2020-12-07:
Thanks very much for testing this again and the explanation about the connection issue!
I see that the patch actually fixed the PPP-exit issue (as expected), but there's yet another one related to data events after the DCE gets destroyed. I think we already came across this one before (I recall that your device present this *retry* after ppp closed, and we already talked about that ... see this [post](https://github.com/espressif/esp-idf/issues/6024#issuecomment-721365171) sorry keep forgetting about that as I cannot reproduce this issue). I'm adding a very similar patch (on top of your current [version](https://github.com/jonathandreyer/esp32-sim7000/blob/espressif-fix3-ppp-exit) and a very simple one, which just `return`s if `esp_dte->parent.dce` is nullptr):
[0001-esp_modem-Ignore-data-event-on-DTE-with-no-DCE-attac.patch.txt](https://github.com/espressif/esp-idf/files/5655168/0001-esp_modem-Ignore-data-event-on-DTE-with-no-DCE-attac.patch.txt)

Could you please test this once again? This should finally help, as what's happening above is that the modem exits PPP in a clean way, powers down the device and deletes the DCE here https://github.com/jonathandreyer/esp32-sim7000/blob/espressif-fix3-ppp-exit/src/main.c#L317. After waiting on the next line, an uart event comes (and we already know that these kind of events come on your device) calling the `esp_handle_uart_data()` and dereferencing nullptr (dce) here: https://github.com/jonathandreyer/esp32-sim7000/blob/espressif-fix3-ppp-exit/components/modem/esp_modem.c#L165

jonathandreyer, 2020-12-07:
@david-cermak Thanks for your devotion. Unfortunately, this does not seem to totally fix the trouble. You can find the output log with debug level below.

Output log (with `debug` level):
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7192
load:0x40078000,len:13072
ho 0 tail 12 room 4
load:0x40080400,len:3896
entry 0x40080688
I (31) boot: ESP-IDF 3.40100.200827 2nd stage bootloader
I (31) boot: compile time 22:32:46
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x18b18 (101144) map
I (153) esp_image: segment 1: paddr=0x00028b40 vaddr=0x3ffb0000 size=0x0229c (  8860) load
I (157) esp_image: segment 2: paddr=0x0002ade4 vaddr=0x40080000 size=0x00404 (  1028) load
I (160) esp_image: segment 3: paddr=0x0002b1f0 vaddr=0x40080404 size=0x04e28 ( 20008) load
I (177) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x6334c (406348) map
I (332) esp_image: segment 5: paddr=0x00093374 vaddr=0x4008522c size=0x060d4 ( 24788) load
I (349) boot: Loaded app from partition at offset 0x10000
I (349) boot: Disabling RNG early entropy source...
I (350) cpu_start: Pro cpu up.
I (353) cpu_start: Application information:
I (358) cpu_start: Project name:     sim800
I (363) cpu_start: App version:      575e996-dirty
I (369) cpu_start: Compile time:     Dec  7 2020 22:31:56
I (375) cpu_start: ELF file SHA256:  c6f8658c0773beab...
I (381) cpu_start: ESP-IDF:          3.40100.200827
I (386) cpu_start: Starting app cpu, entry point is 0x40081230
I (0) cpu_start: App cpu up.
I (397) heap_init: Initializing. RAM available for dynamic allocation:
I (404) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (410) heap_init: At 3FFB3D30 len 0002C2D0 (176 KiB): DRAM
I (416) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (422) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (429) heap_init: At 4008B300 len 00014D00 (83 KiB): IRAM
I (435) cpu_start: Pro cpu start user code
I (453) spi_flash: detected chip: gd
I (454) spi_flash: flash io: dio
I (454) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (461) pppos_example: APN settings: <APN_SETTING>
I (471) uart: queue free spaces: 30
E (6001) esp-modem: esp_dte_handle_line(108): no handler for line
W (6001) pppos_example: Unknow line received: RDY

E (6461) esp-modem: esp_dte_handle_line(108): no handler for line
W (6461) pppos_example: Unknow line received: +CFUN: 1

E (6761) esp-modem: esp_dte_handle_line(108): no handler for line
W (6761) pppos_example: Unknow line received: +CPIN: READY

[... 110 lines cut ...]
W (39421) esp-modem: Pattern not found in the pattern queue, uart data length = 0
W (39671) esp-modem: Pattern not found in the pattern queue, uart data length = 0
E (39671) esp-modem: esp_modem_dte_send_cmd(271): process command timeout
E (39671) sim800: sim800_get_module_name(438): send command failed
E (39681) sim800: sim800_open(585): get module name failed
CORRUPT HEAP: multi_heap.c:172 detected at 0x3ffbd988
abort() was called at PC 0x40088bc7 on core 0

ELF file SHA256: c6f8658c0773beab

Backtrace: 0x400841e1:0x3ffb59e0 0x40084575:0x3ffb5a00 0x40088bc7:0x3ffb5a20 0x40088ea2:0x3ffb5a40 0x400824e6:0x3ffb5a60 0x40089399:0x3ffb5a80 0x400d4427:0x3ffb5aa0 0x400d2c87:0x3ffb5ad0 0x400db73e:0x3ffb5c00 0x40085a41:0x3ffb5c20
  #0  0x400841e1:0x3ffb59e0 in invoke_abort at /Users/<USER>/.platformio/packages/framework-espidf/components/esp32/panic.c:157
  #1  0x40084575:0x3ffb5a00 in abort at /Users/<USER>/.platformio/packages/framework-espidf/components/esp32/panic.c:174
  #2  0x40088bc7:0x3ffb5a20 in multi_heap_assert at /Users/<USER>/.platformio/packages/framework-espidf/components/heap/multi_heap_platform.h:67
      (inlined by) assert_valid_block at /Users/<USER>/.platformio/packages/framework-espidf/components/heap/multi_heap.c:172
  #3  0x40088ea2:0x3ffb5a40 in multi_heap_free_impl at /Users/<USER>/.platformio/packages/framework-espidf/components/heap/multi_heap.c:539
  #4  0x400824e6:0x3ffb5a60 in heap_caps_free at /Users/<USER>/.platformio/packages/framework-espidf/components/heap/heap_caps.c:272
  #5  0x40089399:0x3ffb5a80 in free at /Users/<USER>/.platformio/packages/framework-espidf/components/newlib/heap.c:47
  #6  0x400d4427:0x3ffb5aa0 in sim800_open at components/modem/sim800.c:594
  #7  0x400d2c87:0x3ffb5ad0 in app_main at src/main.c:266
  #8  0x400db73e:0x3ffb5c00 in main_task at /Users/<USER>/.platformio/packages/framework-espidf/components/esp32/cpu_start.c:565
  #9  0x40085a41:0x3ffb5c20 in vPortTaskWrapper at /Users/<USER>/.platformio/packages/framework-espidf/components/freertos/port.c:143

Rebooting...
```

david-cermak, 2020-12-08:
This means that we're at least moving forward, not there yet though.
Heap corruption looks alarming, but this just point to the wrong deletion in [free(dce)](https://github.com/jonathandreyer/esp32-sim7000/blob/espressif-fix3-ppp-exit/components/modem/sim800.c#L593) (should be [free(sim800_dce)](https://github.com/espressif/esp-idf/blob/master/examples/protocols/pppos_client/components/modem/src/sim800.c#L473) ).
The reason for the crash is actually a failure in `sim800_get_module_name()`, and (now i'm a bit speculating) this message seems to be related, the modem's answer not interpreted correctly (pattern `0x0a` not found in the below data?!)
```
W (39321) esp-modem: Pattern not found in the pattern queue, uart data length = 4
D (39321) esp-modem: debug_data: 0x3ffb9f58   4f 4b 0d 0a                                       |OK..|
```
I'll check if I could somehow simulate the problem (this `ESP-IDF=3.40100.200827`, I think means it's `v4.1`)

fromeijn, 2020-12-08:
I have similar issues, I created a loop so I can test reconnecting. I'm not sure if i should make another issue, let me know if i should. For what i'm seeing it seems that the receiver is a bit out of sync.

I used this version: [4.2](https://github.com/espressif/esp-idf/commit/c40f2590bf759ff60ef122afa79b4ec04e7633d2). My local changes thus far are: 
[0001-esp_modem-4.2.patch.txt](https://github.com/espressif/esp-idf/files/5657941/0001-esp_modem-4.2.patch.txt)

Log output:
```
D (29941) esp_netif_lwip: LwIP stack has been initialized
D (29941) esp_netif_lwip: esp-netif has been successfully initialized
D (29941) event: running task for loop 0x3ffb4930
D (29941) event: created task for loop 0x3ffb4930
D (29951) event: created event loop 0x3ffb4930
D (29951) esp_netif_objects: esp_netif_add_to_list 0x3ffb9958
D (29961) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 1)
D (29971) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3ffb9c54
D (29971) esp-netif_lwip-ppp: Phase Dead
I (29981) uart: queue free spaces: 30
D (29981) intr_alloc: Connected src 35 to int 17 (cpu 0)
D (29991) event: created event loop 0x3ffbb3d8
D (30041) dce_service: sync ok
D (30141) dce_service: disable echo ok
D (30341) sim800: get module name ok
D (30541) sim800: get imei number ok
D (30741) sim800: get imsi number ok
D (30941) sim800: get network operator ok
D (31041) dce_service: set flow control ok
D (31141) dce_service: save settings ok
I (31141) pppos_example: Module: SIMCOM_SIM7000G
I (31141) pppos_example: Operator: <Operator>
I (31141) pppos_example: IMEI: <imei>
I (31141) pppos_example: IMSI: <imsi>
D (31341) sim800: inquire signal quality ok
I (31341) pppos_example: rssi: 28, ber: 99
D (31541) sim800: inquire battery status ok
I (31541) pppos_example: Battery voltage: 0 mV
D (31641) dce_service: define pdp context ok
D (31741) sim800: enter ppp mode ok
D (31741) event: running post ESP_MODEM_EVENT:0 with handler 0x400d5614 and context 0x3ffbc9e0 on loop 0x3ffbb3d8
I (31741) pppos_example: Modem PPP Started
D (31741) event: running post ESP_MODEM_EVENT:0 with handler 0x400dc554 and context 0x3ffbcac0 on loop 0x3ffbb3d8
D (31751) esp_netif_handlers: esp_netif action has started with netif0x3ffb9958 from event_id=0
D (31761) esp-netif_lwip-ppp: esp_netif_start_ppp: Starting PPP connection: 0x3ffb9c54
D (31771) esp-netif_lwip-ppp: Phase Start
D (31771) event: running post NETIF_PPP_STATUS:259 with handler 0x400d549c and context 0x3ffb991c on loop 0x3ffb4930
D (31771) esp-netif_lwip-ppp: Phase Establish
I (31781) pppos_example: PPP state changed event 259
D (31791) esp_netif_lwip: check: remote, if=0x3ffb9958 fn=0x400dd854

D (31791) event: running post NETIF_PPP_STATUS:259 with handler 0x400d6a88 and context 0x3ffbcb50 on loop 0x3ffb4930
D (31801) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffb9958
D (31811) event: running post NETIF_PPP_STATUS:262 with handler 0x400d549c and context 0x3ffb991c on loop 0x3ffb4930
D (31821) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (31831) pppos_example: PPP state changed event 262
D (31841) event: running post NETIF_PPP_STATUS:262 with handler 0x400d6a88 and context 0x3ffbcb50 on loop 0x3ffb4930
D (31891) esp-netif_lwip-ppp: Phase Authenticate
D (31891) event: running post NETIF_PPP_STATUS:263 with handler 0x400d549c and context 0x3ffb991c on loop 0x3ffb4930
I (31891) pppos_example: PPP state changed event 263
D (31901) event: running post NETIF_PPP_STATUS:263 with handler 0x400d6a88 and context 0x3ffbcb50 on loop 0x3ffb4930
D (31951) esp-netif_lwip-ppp: Phase Network
D (31951) event: running post NETIF_PPP_STATUS:265 with handler 0x400d549c and context 0x3ffb991c on loop 0x3ffb4930
I (31951) pppos_example: PPP state changed event 265
D (31961) event: running post NETIF_PPP_STATUS:265 with handler 0x400d6a88 and context 0x3ffbcb50 on loop 0x3ffb4930
I (32111) esp-netif_lwip-ppp: Connected
I (32111) esp-netif_lwip-ppp: Name Server1: <ip addr>
I (32111) esp-netif_lwip-ppp: Name Server2: <ip addr>
D (32111) event: running post IP_EVENT:5 with handler 0x400d5670 and context 0x3ffb98f0 on loop 0x3ffb4930
D (32121) esp-netif_lwip-ppp: Phase Running
D (32121) pppos_example: IP event! 5
I (32131) pppos_example: Modem Connect to PPP Server
I (32141) pppos_example: ~~~~~~~~~~~~~~
[... 123 lines cut ...]
I (49301) pppos_example: Module: SIMCOM_SIM7000G
I (49301) pppos_example: Operator: <Operator>
I (49301) pppos_example: IMEI: <imei>
I (49301) pppos_example: IMSI: <imsi>
W (49451) esp-modem: Pattern not found in the pattern queue, uart data length = 4
D (49451) esp-modem: debug_data: 0x3ffbaf2c   4f 4b 0d 0a                                       |OK..|
W (49601) esp-modem: Pattern not found in the pattern queue, uart data length = 0
W (49651) esp-modem: Pattern not found in the pattern queue, uart data length = 0
W (49701) esp-modem: Pattern not found in the pattern queue, uart data length = 0
E (49811) esp-modem: esp_modem_dte_send_cmd(254): process command timeout
E (49811) sim800: sim800_get_signal_quality(248): send command failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400844e4
file: "../main/pppos_client_main.c" line 477
func: app_main
expression: dce->get_signal_quality(dce, &rssi, &ber)

abort() was called at PC 0x400844e7 on core 0

Backtrace:0x4008569b:0x3ffb5780 0x40085d39:0x3ffb57a0 0x40089a06:0x3ffb57c0 0x400844e7:0x3ffb5830 0x400d5dba:0x3ffb5850 0x400d406a:0x3ffb5990 0x40085d41:0x3ffb59c0

ELF file SHA256: 52b7228114d21075

Rebooting...
ets Jun  8 2016 00:22:57
```

david-cermak, 2020-12-08:
@fromeijn Thanks for your report! The version, you've referred missed [the timeout fix](https://github.com/espressif/esp-idf/commit/6d70b7c352d8cf0340ac14e127372a7a83f4779a), but I was able to reproduce this issue, because of that! There is a bug in my last patch: [`uart_pattern_pop_pos()`](https://github.com/jonathandreyer/esp32-sim7000/blob/espressif-fix3-ppp-exit/components/modem/esp_modem.c#L174) should be `uart_pattern_get_pos()` as the former *consumes* the position and the upcoming checks won't find the pattern!
Here's a small patch on top your changes:
[0001-esp-modem-PPP-exit-additional-pattern-fixes-v4.2.patch.txt](https://github.com/espressif/esp-idf/files/5660339/0001-esp-modem-PPP-exit-additional-pattern-fixes-v4.2.patch.txt)

It would be great if you could test it again with the fix.

david-cermak, 2020-12-08:
@jonathandreyer As mentioned above, I was able to reproduce the problem and think it's the same issue. your changes contained the [rx-timeout fix](https://github.com/jonathandreyer/esp32-sim7000/blob/espressif-fix3-ppp-exit/components/modem/esp_modem.c#L466), though. (but this is `v4.1` and the timeout unit might be a little different -- speculation again, will check...)

Adding another version of the patch, this is an *absolute* patch against current master, so replacing the previous one:
[0001-esp_modem-Fixed-exit-PPP-mode-Dec-8.patch.txt](https://github.com/espressif/esp-idf/files/5660369/0001-esp_modem-Fixed-exit-PPP-mode-Dec-8.patch.txt)
(just amended the previous commit)

Again, I'd be grateful if you can test this once more

jonathandreyer, 2020-12-08:
@david-cermak Thanks for the new patch. We move forward but, there is still a strange behaviour. The program is blocked (without crashing) in the second loop. I have updated my "current master" of the [repository](https://github.com/jonathandreyer/esp32-sim7000/tree/espressif-others-patches) where the example is stored (fix error explain [here](https://github.com/espressif/esp-idf/issues/6024#issuecomment-740447511) and apply a part of the missing commit [```fffdc1d```](https://github.com/espressif/esp-idf/commit/fffdc1d789a8ed830dd7f19b690fb5c79c375200)). 

The code tested are available [here](https://github.com/jonathandreyer/esp32-sim7000/commit/43cf7b36401ba8e85f2192c1e28512837b77a563). You can find the output log with debug level below.

Output log (with debug level):
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7192
load:0x40078000,len:13072
ho 0 tail 12 room 4
load:0x40080400,len:3896
entry 0x40080688
I (31) boot: ESP-IDF 3.40100.200827 2nd stage bootloader
I (31) boot: compile time 22:05:05
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x18d18 (101656) map
I (153) esp_image: segment 1: paddr=0x00028d40 vaddr=0x3ffb0000 size=0x0229c (  8860) load
I (157) esp_image: segment 2: paddr=0x0002afe4 vaddr=0x40080000 size=0x00404 (  1028) load
I (160) esp_image: segment 3: paddr=0x0002b3f0 vaddr=0x40080404 size=0x04c28 ( 19496) load
I (177) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x63b00 (408320) map
I (333) esp_image: segment 5: paddr=0x00093b28 vaddr=0x4008502c size=0x062d4 ( 25300) load
I (350) boot: Loaded app from partition at offset 0x10000
I (350) boot: Disabling RNG early entropy source...
I (351) cpu_start: Pro cpu up.
I (354) cpu_start: Application information:
I (359) cpu_start: Project name:     sim800
I (364) cpu_start: App version:      43cf7b3-dirty
I (370) cpu_start: Compile time:     Dec  8 2020 22:04:19
I (376) cpu_start: ELF file SHA256:  9b0ca8f1139da9f0...
I (382) cpu_start: ESP-IDF:          3.40100.200827
I (387) cpu_start: Starting app cpu, entry point is 0x40081230
I (0) cpu_start: App cpu up.
I (398) heap_init: Initializing. RAM available for dynamic allocation:
I (405) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (411) heap_init: At 3FFB3D30 len 0002C2D0 (176 KiB): DRAM
I (417) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (423) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (430) heap_init: At 4008B300 len 00014D00 (83 KiB): IRAM
I (436) cpu_start: Pro cpu start user code
I (454) spi_flash: detected chip: gd
I (455) spi_flash: flash io: dio
I (455) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (463) pppos_example: APN settings: <APN_SETTING>
I (473) uart: queue free spaces: 30
E (6003) esp-modem: esp_dte_handle_line(108): no handler for line
W (6003) pppos_example: Unknow line received: RDY

E (6453) esp-modem: esp_dte_handle_line(108): no handler for line
W (6453) pppos_example: Unknow line received: +CFUN: 1

E (6753) esp-modem: esp_dte_handle_line(108): no handler for line
[... 137 lines cut ...]
D (40923) sim800: enter ppp mode ok
I (40923) pppos_example: Modem PPP Started
I (40923) pppos_example: PPP state changed event 259
I (40923) pppos_example: PPP state changed event 262
I (41023) pppos_example: PPP state changed event 263
I (41073) pppos_example: PPP state changed event 265
W (41123) esp-modem: Pattern not found in the pattern queue, uart data length = 26
D (41123) esp-modem-pattern: debug_data: 0x3ffb9f58   7e 80 21 01 00 00 04 67  c3 7e 7e 80 21 04 04 00  |~.!....g.~~.!...|
D (41123) esp-modem-pattern: debug_data: 0x3ffb9f68   0a 02 06 00 2d 0f 01 e0  42 7e                    |....-...B~|
W (41333) esp-modem: Pattern not found in the pattern queue, uart data length = 0
W (41533) esp-modem: Pattern not found in the pattern queue, uart data length = 0
W (41733) esp-modem: Pattern not found in the pattern queue, uart data length = 0
W (41833) esp-modem: Pattern not found in the pattern queue, uart data length = 0
W (41933) esp-modem: Pattern not found in the pattern queue, uart data length = 0
W (42133) esp-modem: Pattern not found in the pattern queue, uart data length = 0
W (42333) esp-modem: Pattern not found in the pattern queue, uart data length = 0
I (101073) pppos_example: PPP state changed event 267
I (101073) pppos_example: PPP state changed event 262
I (113073) pppos_example: PPP state changed event 268
I (113073) pppos_example: PPP state changed event 256
I (113073) esp-netif_lwip-ppp: User interrupt
I (113073) pppos_example: PPP state changed event 5
I (113083) pppos_example: User interrupted event from netif:0x3ffbff80
I (113083) esp-modem-netif: PPP state changed event 5
```

fromeijn, 2020-12-09:
@david-cermak Thanks for you input, much appreciated! 

It is working better already, only still having problems to connect it up to netif. Not sure how to solve this. See the log below:

```
D (29951) esp_netif_lwip: LwIP stack has been initialized
D (29951) esp_netif_lwip: esp-netif has been successfully initialized
D (29951) event: running task for loop 0x3ffb4930
D (29951) event: created task for loop 0x3ffb4930
D (29961) event: created event loop 0x3ffb4930
D (29961) esp_netif_objects: esp_netif_add_to_list 0x3ffb9958
D (29971) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 1)
D (29981) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3ffb9c54
D (29981) esp-netif_lwip-ppp: Phase Dead
I (29991) uart: queue free spaces: 30
D (29991) intr_alloc: Connected src 35 to int 17 (cpu 0)
D (30001) event: created event loop 0x3ffbb3d8
D (30051) dce_service: sync ok
D (30151) dce_service: disable echo ok
D (30351) sim800: get module name ok
D (30551) sim800: get imei number ok
D (30751) sim800: get imsi number ok
D (30951) sim800: get network operator ok
D (31051) dce_service: set flow control ok
D (31151) dce_service: save settings ok
I (31151) pppos_example: Module: SIMCOM_SIM7000G
I (31151) pppos_example: Operator: <Operator>
I (31151) pppos_example: IMEI: <imei>
I (31151) pppos_example: IMSI: <imsi>
D (31351) sim800: inquire signal quality ok
I (31351) pppos_example: rssi: 31, ber: 99
D (31551) sim800: inquire battery status ok
I (31551) pppos_example: Battery voltage: 0 mV
D (41621) dce_service: define pdp context ok
D (41721) sim800: enter ppp mode ok
D (41721) event: running post ESP_MODEM_EVENT:0 with handler 0x400d562c and context 0x3ffbc9e0 on loop 0x3ffbb3d8
0x400d562c: modem_event_handler at ~/esp/esp-idf-v4.2/examples/protocols/pppos_client/build/../main/pppos_client_main.c:134

I (41721) pppos_example: Modem PPP Started
D (41721) event: running post ESP_MODEM_EVENT:0 with handler 0x400dc60c and context 0x3ffbcabc on loop 0x3ffbb3d8
0x400dc60c: esp_netif_action_start at ~/esp/esp-idf-v4.2/components/esp_netif/esp_netif_handlers.c:32

D (41731) esp_netif_handlers: esp_netif action has started with netif0x3ffb9958 from event_id=0
D (41741) esp-netif_lwip-ppp: esp_netif_start_ppp: Starting PPP connection: 0x3ffb9c54
D (41751) esp-netif_lwip-ppp: Phase Start
D (41751) event: running post NETIF_PPP_STATUS:259 with handler 0x400d54b4 and context 0x3ffb991c on loop 0x3ffb4930
0x400d54b4: on_ppp_changed at ~/esp/esp-idf-v4.2/examples/protocols/pppos_client/build/../main/pppos_client_main.c:193

D (41751) esp-netif_lwip-ppp: Phase Establish
I (41761) pppos_example: PPP state changed event 259
D (41771) esp_netif_lwip: check: remote, if=0x3ffb9958 fn=0x400dd90c
0x400dd90c: esp_netif_update_default_netif_lwip at ~/esp/esp-idf-v4.2/components/esp_netif/lwip/esp_netif_lwip.c:163

D (41771) event: running post NETIF_PPP_STATUS:259 with handler 0x400d6b40 and context 0x3ffbcb4c on loop 0x3ffb4930
0x400d6b40: on_ppp_changed at ~/esp/esp-idf-v4.2/examples/protocols/pppos_client/build/../components/modem/src/esp_modem_netif.c:31

D (41781) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffb9958
D (41791) event: running post NETIF_PPP_STATUS:262 with handler 0x400d54b4 and context 0x3ffb991c on loop 0x3ffb4930
0x400d54b4: on_ppp_changed at ~/esp/esp-idf-v4.2/examples/protocols/pppos_client/build/../main/pppos_client_main.c:193

D (41801) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (41811) pppos_example: PPP state changed event 262
D (41821) event: running post NETIF_PPP_STATUS:262 with handler 0x400d6b40 and context 0x3ffbcb4c on loop 0x3ffb4930
0x400d6b40: on_ppp_changed at ~/esp/esp-idf-v4.2/examples/protocols/pppos_client/build/../components/modem/src/esp_modem_netif.c:31

D (41871) esp-netif_lwip-ppp: Phase Authenticate
D (41871) event: running post NETIF_PPP_STATUS:263 with handler 0x400d54b4 and context 0x3ffb991c on loop 0x3ffb4930
0x400d54b4: on_ppp_changed at ~/esp/esp-idf-v4.2/examples/protocols/pppos_client/build/../main/pppos_client_main.c:193

I (41871) pppos_example: PPP state changed event 263
[... 216 lines cut ...]
D (70051) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffb9958
D (70061) event: running post NETIF_PPP_STATUS:262 with handler 0x400d54b4 and context 0x3ffb991c on loop 0x3ffb4930
0x400d54b4: on_ppp_changed at ~/esp/esp-idf-v4.2/examples/protocols/pppos_client/build/../main/pppos_client_main.c:193

D (70071) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (70081) pppos_example: PPP state changed event 262
D (70091) event: running post NETIF_PPP_STATUS:262 with handler 0x400d6b40 and context 0x3ffbcb4c on loop 0x3ffb4930
0x400d6b40: on_ppp_changed at ~/esp/esp-idf-v4.2/examples/protocols/pppos_client/build/../components/modem/src/esp_modem_netif.c:31

Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x002f998f  PS      : 0x00060230  A0      : 0x800de178  A1      : 0x3ffb8770  
A2      : 0x3ffbcbd4  A3      : 0x3ffbd1e4  A4      : 0x0000002e  A5      : 0x3ffb22cc  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x801332ca  A9      : 0x3ffb8740  
A10     : 0x06050000  A11     : 0x3ffbd1e4  A12     : 0x0000002e  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0x002f998c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace:0x002f998c:0x3ffb8770 |<-CORRUPTED

ELF file SHA256: 3037f8e322441ee7

Rebooting...
```

david-cermak, 2020-12-10:
@jonathandreyer Does it hang every second iteration or just randomly? I think it happens time to time that the PPP client gets disconnected and user application needs to take care of reconnection. (it's not stuck, it just waits in the main loop for connection flag here: https://github.com/jonathandreyer/esp32-sim7000/blob/43cf7b36401ba8e85f2192c1e28512837b77a563/src/main.c#L293)
The trouble here is unfortunately the misleading event we're getting:
```
I (113073) esp-netif_lwip-ppp: User interrupt
I (113073) pppos_example: PPP state changed event 5
I (113083) pppos_example: User interrupted event from netif:0x3ffbff80
I (113083) esp-modem-netif: PPP state changed event 5
```
User interrupt (i.e. disconnect raising from `esp_netif_close()`?!) instead of disconnection event `IP_EVENT_PPP_LOST_IP` which we'd normally expect and could make corrective actions on application level.
The actual problem seems to be directly in the lwip, as the ppp's `err_code` is not cleaned between restarts with close/open, so the user flag is set from the previous iteration. 
A quick fix might be setting
```
pcb->err_code == PPPERR_NONE
```
after starting the ppp session somewhere in here: https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/netif/ppp/ppp.c#L730

Plus of course handling disconnection in the app code
https://github.com/jonathandreyer/esp32-sim7000/blob/43cf7b36401ba8e85f2192c1e28512837b77a563/src/main.c#L206

david-cermak, 2020-12-10:
@fromeijn I think creating the `modem_netif_adapter` just once and attaching it to the netif in the loop should solve this issue. I believe that lwip's PPP is not very restart-friendly, but this should work at least:
1) Create the netif adapter before the reconnection while loop (e.g. here https://github.com/jonathandreyer/esp32-sim7000/blob/43cf7b36401ba8e85f2192c1e28512837b77a563/src/main.c#L255-L256)
2) Run only the  `esp_netif_attach()` in the loop (similar to https://github.com/jonathandreyer/esp32-sim7000/blob/43cf7b36401ba8e85f2192c1e28512837b77a563/src/main.c#L291) which would in turn call `pppos_start()`

jonathandreyer, 2020-12-10:
@david-cermak Thanks for your message. I didn't test to last version more than 3-5 time, but at each test I have always had the same trouble.
If you want, I can test it again (multiple time, maybe 10 times).

david-cermak, 2020-12-11:
For completeness, I'm also adding a preliminary version of the uart driver fix to be backported to `v4.1`. The example should work (better) with this or the above workaround (or both)
[0001-driver-backport-fix-driver-set-rx-timeout-feature-of.patch.txt](https://github.com/espressif/esp-idf/files/5680580/0001-driver-backport-fix-driver-set-rx-timeout-feature-of.patch.txt)

jonathandreyer, 2020-12-13:
@david-cermak Thanks for your message. I started the tests again with the previous [patch](https://github.com/espressif/esp-idf/issues/6024#issuecomment-740717189). The code of this version is available [here](https://github.com/jonathandreyer/esp32-sim7000/tree/espressif-fix4-ppp-exit). Unfortunately and after multiple restart, it is always the same error. You can find the output log with debug level right below.
I have tried to use your backported patch, but with PlatformIO, there is a problem. The repository is available [here](https://github.com/jonathandreyer/esp-idf/tree/release/v4.1). I have opened an issue (platformio/platform-espressif32#450) and I will be back when it is OK.

Output log (with debug level) / Multiple startup:
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7192
load:0x40078000,len:13072
ho 0 tail 12 room 4
load:0x40080400,len:3896
entry 0x40080688
I (31) boot: ESP-IDF 3.40100.200827 2nd stage bootloader
I (31) boot: compile time 22:18:30
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (98) boot_comm: chip revision: 1, min. application chip revision: 0
I (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x18d18 (101656) map
I (153) esp_image: segment 1: paddr=0x00028d40 vaddr=0x3ffb0000 size=0x0229c (  8860) load
I (157) esp_image: segment 2: paddr=0x0002afe4 vaddr=0x40080000 size=0x00404 (  1028) load
I (160) esp_image: segment 3: paddr=0x0002b3f0 vaddr=0x40080404 size=0x04c28 ( 19496) load
I (177) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x63b00 (408320) map
I (333) esp_image: segment 5: paddr=0x00093b28 vaddr=0x4008502c size=0x062d4 ( 25300) load
I (350) boot: Loaded app from partition at offset 0x10000
I (350) boot: Disabling RNG early entropy source...
I (351) cpu_start: Pro cpu up.
I (354) cpu_start: Application information:
I (359) cpu_start: Project name:     sim800
I (364) cpu_start: App version:      43cf7b3-dirty
I (370) cpu_start: Compile time:     Dec 13 2020 22:17:37
I (376) cpu_start: ELF file SHA256:  9d1bb788f8186b65...
I (382) cpu_start: ESP-IDF:          3.40100.200827
I (387) cpu_start: Starting app cpu, entry point is 0x40081230
I (0) cpu_start: App cpu up.
I (398) heap_init: Initializing. RAM available for dynamic allocation:
I (405) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (411) heap_init: At 3FFB3D30 len 0002C2D0 (176 KiB): DRAM
I (417) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (423) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (430) heap_init: At 4008B300 len 00014D00 (83 KiB): IRAM
I (436) cpu_start: Pro cpu start user code
I (454) spi_flash: detected chip: gd
I (455) spi_flash: flash io: dio
I (455) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (462) pppos_example: APN settings: <APN_SETTING>
I (472) uart: queue free spaces: 30
E (6002) esp-modem: esp_dte_handle_line(108): no handler for line
W (6002) pppos_example: Unknow line received: RDY

E (6462) esp-modem: esp_dte_handle_line(108): no handler for line
W (6462) pppos_example: Unknow line received: +CFUN: 1

E (6762) esp-modem: esp_dte_handle_line(108): no handler for line
W (6762) pppos_example: Unknow line received: +CPIN: READY
[... 1886 lines cut ...]
I (25793) esp-netif_lwip-ppp: User interrupt
I (25793) pppos_example: PPP state changed event 5
I (25803) pppos_example: User interrupted event from netif:0x3ffbff80
I (25803) esp-modem-netif: PPP state changed event 5
D (25893) sim800: power down ok
I (25893) pppos_example: Power down
I (25893) pppos_example: Restart after 10 seconds
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d2fc8  PS      : 0x00060a30  A0      : 0x800d3265  A1      : 0x3ffbb740
A2      : 0x3ffb9f08  A3      : 0x00000000  A4      : 0x3ffba478  A5      : 0x00000000
A6      : 0x3ffba558  A7      : 0x00000003  A8      : 0x00000000  A9      : 0x3ffbb710
A10     : 0xffffffff  A11     : 0x3ffb0058  A12     : 0xffffffff  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x00000013  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000064  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

ELF file SHA256: 794a744b87d3c00e

Backtrace: 0x400d2fc5:0x3ffbb740 0x400d3262:0x3ffbb770 0x40085a41:0x3ffbb7a0
  #0  0x400d2fc5:0x3ffbb740 in esp_handle_uart_pattern at components/modem/esp_modem.c:136
  #1  0x400d3262:0x3ffbb770 in uart_event_task_entry at components/modem/esp_modem.c:268
  #2  0x40085a41:0x3ffbb7a0 in vPortTaskWrapper at /Users/<USER>/.platformio/packages/framework-espidf/components/freertos/port.c:143

Rebooting...
```

david-cermak, 2020-12-14:
@jonathandreyer Thanks again for testing this! It looks like this fails every second iteration in a regular way, that's interesting as I couldn't possibly reproduce it even with the "unpatched" `v4.1`.

As for the workaround patch. Could you please still give it another try with the below?
[0001-esp_modem-Fixed-exit-PPP-mode-Dec-14.patch.txt](https://github.com/espressif/esp-idf/files/5688783/0001-esp_modem-Fixed-exit-PPP-mode-Dec-14.patch.txt)
(again cumulative patch replacing the previous one)
The issue was that I (have again) forgotten to ignore the uart event (sorry we've had this issue before for other events, this time it was the newly updated pattern event)

jonathandreyer, 2020-12-14:
@david-cermak Thanks for your message. I have found a solution to build with an upstream version. I will be back as soon as possible (maybe tomorrow) with the results for your new [patch](https://github.com/espressif/esp-idf/issues/6024#issuecomment-744414011) and the previous [one](https://github.com/espressif/esp-idf/issues/6024#issuecomment-743316219).

dragan-cvetkov, 2020-12-15:
Hi @david-cermak and @jonathandreyer,
I'm currently using V4.2 and facing the same issues on pppos using Sara R4 (race condition and UART issue when calling esp_modem_stop_ppp). I was wondering if the [patch](https://github.com/espressif/esp-idf/issues/6024#issuecomment-744414011) can be used also on V4.2. Thank you in advance

david-cermak, 2020-12-15:
@dragan-cvetkov Yes, the patch `0001-esp_modem-Fixed-exit-PPP-mode-Dec-14.patch.txt` posted [here](https://github.com/espressif/esp-idf/issues/6024#issuecomment-74441401) is applicable to `v4.2` and master, as well. (i'm actually testing it mainly on master). For the `v4.2` you might need to cherry-pick the https://github.com/espressif/esp-idf/commit/e8834b66c583fdca31ef006d2a5cdb1e1af171de (it's backported already, haven't been pushed to github yet)

jonathandreyer, 2020-12-15:
@david-cermak Thanks for your messages and patches. I have a good news! It seems to work with the latest [patch](https://github.com/espressif/esp-idf/issues/6024#issuecomment-744414011) and the upstreamed ESP-IDF with your backported [patch](https://github.com/espressif/esp-idf/issues/6024#issuecomment-743316219). The code of this version is available [here](https://github.com/jonathandreyer/esp32-sim7000/tree/espressif-fix6-ppp-exit-upstream). I have let the program run more than 20 loops with any crash. You can find the output log just below (only the first three loops).
I have also [tested](https://github.com/jonathandreyer/esp32-sim7000/tree/espressif-fix6-ppp-exit) without the upstreamed version (released v4.1 version) and it is also crashes. Maybe, I can add that the version of ESP-IDF used is the release one. The output log is available in the after the working log.

Output log (with debug level) / Working loops:
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7460
load:0x40078000,len:13072
load:0x40080400,len:3896
entry 0x40080688
I (29) boot: ESP-IDF 3.40100.201212+sha.c5fba7c 2nd stage bootloader
I (29) boot: compile time 21:47:01
I (29) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x18df0 (101872) map
I (152) esp_image: segment 1: paddr=0x00028e18 vaddr=0x3ffb0000 size=0x022fc (  8956) load
I (156) esp_image: segment 2: paddr=0x0002b11c vaddr=0x40080000 size=0x00404 (  1028) load
I (159) esp_image: segment 3: paddr=0x0002b528 vaddr=0x40080404 size=0x04af0 ( 19184) load
I (176) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x63ffc (409596) map
I (333) esp_image: segment 5: paddr=0x00094024 vaddr=0x40084ef4 size=0x06514 ( 25876) load
I (350) boot: Loaded app from partition at offset 0x10000
I (351) boot: Disabling RNG early entropy source...
I (351) cpu_start: Pro cpu up.
I (355) cpu_start: Application information:
I (359) cpu_start: Project name:     sim800
I (364) cpu_start: App version:      01cb940-dirty
I (370) cpu_start: Compile time:     Dec 15 2020 21:46:19
I (376) cpu_start: ELF file SHA256:  32e2c7ecf9f016e0...
I (382) cpu_start: ESP-IDF:          3.40100.201212+sha.c5fba7c
I (388) cpu_start: Starting app cpu, entry point is 0x40081248
I (0) cpu_start: App cpu up.
I (399) heap_init: Initializing. RAM available for dynamic allocation:
I (406) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (412) heap_init: At 3FFB3CF8 len 0002C308 (176 KiB): DRAM
I (418) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (424) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (431) heap_init: At 4008B408 len 00014BF8 (82 KiB): IRAM
I (437) cpu_start: Pro cpu start user code
I (455) spi_flash: detected chip: gd
I (456) spi_flash: flash io: dio
I (456) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (464) pppos_example: APN settings: <APN_SETTING>
I (474) uart: queue free spaces: 30
D (5944) esp-modem: No handler for line: RDY

W (5944) pppos_example: Unknow line received: RDY

D (6394) esp-modem: No handler for line: +CFUN: 1

W (6394) pppos_example: Unknow line received: +CFUN: 1

D (6504) esp-modem: No handler for line: +CPIN: READY
[... 775 lines cut ...]
I (24293) pppos_example: PPP state changed event 256
I (24293) esp-netif_lwip-ppp: User interrupt
I (24293) pppos_example: PPP state changed event 5
I (24303) pppos_example: User interrupted event from netif:0x3ffbad88
I (24303) esp-modem-netif: PPP state changed event 5
D (24313) esp-modem: Pattern not found in the pattern queue, uart data length = 1
D (24323) esp-modem-pattern: debug_data: 0x3ffb9f58   4e                                                |N|
E (25313) esp-modem: esp_modem_dte_send_cmd(306): process command timeout
E (25313) sim800: sim800_power_down(418): send command failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40084790
file: "src/main.c" line 319
func: app_main
expression: dce->power_down(dce)

ELF file SHA256: cb7ca90174ed3eaf

Backtrace: 0x400841e1:0x3ffb5a90 0x40084793:0x3ffb5ab0 0x400d2e85:0x3ffb5ad0 0x400db8a2:0x3ffb5c00 0x40085a41:0x3ffb5c20
  #0  0x400841e1:0x3ffb5a90 in invoke_abort at /Users/<USER>/.platformio/packages/framework-espidf/components/esp32/panic.c:157
  #1  0x40084793:0x3ffb5ab0 in _esp_error_check_failed at /Users/<USER>/.platformio/packages/framework-espidf/components/esp32/panic.c:740
  #2  0x400d2e85:0x3ffb5ad0 in app_main at src/main.c:319 (discriminator 1)
  #3  0x400db8a2:0x3ffb5c00 in main_task at /Users/<USER>/.platformio/packages/framework-espidf/components/esp32/cpu_start.c:565
  #4  0x40085a41:0x3ffb5c20 in vPortTaskWrapper at /Users/<USER>/.platformio/packages/framework-espidf/components/freertos/port.c:143

Rebooting...
```

jonathandreyer, 2020-12-15:
@david-cermak Maybe if you would like, I can let run the program more longer (several hours) to track if it is OK after a long time and multiple (PPP enter & PPP exit).

dragan-cvetkov, 2020-12-16:
Hi @david-cermak, thank you a lot for the information. I'll patch and be back with an update.

david-cermak, 2020-12-16:
@jonathandreyer Thanks for the good new 👍  
I was going to give up and refactor the UART part to use only the rx mode (not switching between pattern and rx, which I'm probably going to do anyway, but not a good changeset to be backported...).

> Maybe if you would like, I can let run the program more longer (several hours) to track if it is OK after a long time and multiple (PPP enter & PPP exit).

Yes, please it would be nice if you can give it a bit more time. Just note, that i might get stuck in waiting for the connection flag (for the reasons I mentioned before), the fix for this is twofold:
* lwip fix https://github.com/espressif/esp-idf/issues/6024#issuecomment-742829601 to clean up the state
* user app to check for disconnection (IP-lost event)
(well, if the current version hangs waiting for the connection it's kind of expected)

Also I would be curious about the last patch without the UART fix (posted here https://github.com/espressif/esp-idf/issues/6024#issuecomment-743316219), as this should work as well.

fromeijn, 2020-12-16:
@david-cermak Thanks for your feedback, had some issues testing because i could not reach the broker `mqtt.eclipse.org`. Tought is had something to do with my setup. I used this patch https://github.com/espressif/esp-idf/issues/6024#issuecomment-744414011 and added https://github.com/espressif/esp-idf/commit/e8834b66c583fdca31ef006d2a5cdb1e1af171de now it seems stable. Also added a 1 second wait after mqtt disconnection, otherwise will get `handle line failed`.

Any suggestions to test more or to improve?

so all my changes from [V4.2](https://github.com/espressif/esp-idf/tree/v4.2) are : 
[0001-esp_modem-Fixed-exit-PPP-mode-Dec-14-V4.2.patch.txt](https://github.com/espressif/esp-idf/files/5702840/0001-esp_modem-Fixed-exit-PPP-mode-Dec-14-V4.2.patch.txt)

my log over a few loops:
```
D (35935) esp_netif_lwip: LwIP stack has been initialized
D (35935) esp_netif_lwip: esp-netif has been successfully initialized
D (35935) event: running task for loop 0x3ffb8834
D (35935) event: created task for loop 0x3ffb8834
D (35945) event: created event loop 0x3ffb8834
D (35945) esp_netif_objects: esp_netif_add_to_list 0x3ffb97dc
D (35955) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 1)
D (35965) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3ffb9ae0
D (35965) esp-netif_lwip-ppp: Phase Dead
I (35975) uart: queue free spaces: 30
D (35975) intr_alloc: Connected src 35 to int 13 (cpu 0)
D (35985) event: created event loop 0x3ffbb264
D (35985) dce_service: sync ok
D (35995) dce_service: disable echo ok
D (35995) sim800: get module name ok
D (36005) sim800: get imei number ok
D (36005) sim800: get imsi number ok
D (36015) sim800: get network operator ok
D (36015) dce_service: set flow control ok
D (36015) dce_service: save settings ok
I (36015) pppos_example: Module: SIMCOM_SIM7000G
I (36025) pppos_example: Operator: <operator>
I (36025) pppos_example: IMEI: <imei>
I (36035) pppos_example: IMSI: <imsi>
D (36045) sim800: inquire signal quality ok
I (36045) pppos_example: rssi: 29, ber: 99
D (36065) sim800: inquire battery status ok
I (36065) pppos_example: Battery voltage: 3309 mV
D (36095) dce_service: define pdp context ok
D (36105) sim800: enter ppp mode ok
D (36205) event: running post ESP_MODEM_EVENT:0 with handler 0x400d5698 and context 0x3ffbc86c on loop 0x3ffbb264
0x400d5698: modem_event_handler at /examples/protocols/pppos_client/build/../main/pppos_client_main.c:113

I (36205) pppos_example: Modem PPP Started
D (36205) event: running post ESP_MODEM_EVENT:0 with handler 0x400dc500 and context 0x3ffbc948 on loop 0x3ffbb264
0x400dc500: esp_netif_action_start at /components/esp_netif/esp_netif_handlers.c:32

D (36215) esp_netif_handlers: esp_netif action has started with netif0x3ffb97dc from event_id=0
D (36225) esp-netif_lwip-ppp: esp_netif_start_ppp: Starting PPP connection: 0x3ffb9ae0
D (36235) esp-netif_lwip-ppp: Phase Start
D (36235) event: running post NETIF_PPP_STATUS:259 with handler 0x400d5520 and context 0x3ffb97a0 on loop 0x3ffb8834
0x400d5520: on_ppp_changed at /examples/protocols/pppos_client/build/../main/pppos_client_main.c:172

D (36235) esp-netif_lwip-ppp: Phase Establish
I (36245) pppos_example: PPP state changed event 259
D (36255) esp_netif_lwip: check: remote, if=0x3ffb97dc fn=0x400dd800
0x400dd800: esp_netif_update_default_netif_lwip at /components/esp_netif/lwip/esp_netif_lwip.c:163

D (36255) event: running post NETIF_PPP_STATUS:259 with handler 0x400d6a48 and context 0x3ffbc9d8 on loop 0x3ffb8834
0x400d6a48: on_ppp_changed at /examples/protocols/pppos_client/build/../components/modem/src/esp_modem_netif.c:31

D (36265) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffb97dc
D (36275) event: running post NETIF_PPP_STATUS:262 with handler 0x400d5520 and context 0x3ffb97a0 on loop 0x3ffb8834
0x400d5520: on_ppp_changed at /examples/protocols/pppos_client/build/../main/pppos_client_main.c:172

D (36285) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (36295) pppos_example: PPP state changed event 262
D (36305) event: running post NETIF_PPP_STATUS:262 with handler 0x400d6a48 and context 0x3ffbc9d8 on loop 0x3ffb8834
0x400d6a48: on_ppp_changed at /examples/protocols/pppos_client/build/../components/modem/src/esp_modem_netif.c:31

D (36305) esp-netif_lwip-ppp: Phase Authenticate
[... 561 lines cut ...]
0x400d5520: on_ppp_changed at /examples/protocols/pppos_client/build/../main/pppos_client_main.c:172

I (97365) pppos_example: PPP state changed event 268
D (97365) event: running post NETIF_PPP_STATUS:268 with handler 0x400d6a48 and context 0x3ffbc9d8 on loop 0x3ffb8834
0x400d6a48: on_ppp_changed at /examples/protocols/pppos_client/build/../components/modem/src/esp_modem_netif.c:31

D (97365) esp-netif_lwip-ppp: Phase Dead
D (97385) event: running post NETIF_PPP_STATUS:256 with handler 0x400d5520 and context 0x3ffb97a0 on loop 0x3ffb8834
0x400d5520: on_ppp_changed at /examples/protocols/pppos_client/build/../main/pppos_client_main.c:172

I (97385) esp-netif_lwip-ppp: User interrupt
I (97395) pppos_example: PPP state changed event 256
D (97405) event: running post NETIF_PPP_STATUS:256 with handler 0x400d6a48 and context 0x3ffbc9d8 on loop 0x3ffb8834
0x400d6a48: on_ppp_changed at /examples/protocols/pppos_client/build/../components/modem/src/esp_modem_netif.c:31

D (97415) event: running post NETIF_PPP_STATUS:5 with handler 0x400d5520 and context 0x3ffb97a0 on loop 0x3ffb8834
0x400d5520: on_ppp_changed at /examples/protocols/pppos_client/build/../main/pppos_client_main.c:172

I (97425) pppos_example: PPP state changed event 5
I (97435) pppos_example: User interrupted event from netif:0x3ffbbc58
D (97435) event: running post NETIF_PPP_STATUS:5 with handler 0x400d6a48 and context 0x3ffbc9d8 on loop 0x3ffb8834
0x400d6a48: on_ppp_changed at /examples/protocols/pppos_client/build/../components/modem/src/esp_modem_netif.c:31

I (97445) esp-modem-netif: PPP state changed event 5
```

dragan-cvetkov, 2020-12-16:
Hi @david-cermak I was able to test the patch. In my case it is working (much better) but I'm still having an issue. 
My application works as follow.

- By removing the antenna, an internal connection error is triggered
- If an error is detected all the task are deinitialized (safely)
- At last the PPP is stopped and GSM/LTEM module is deinitialized

Sometimes the ppp_stop is working as expected, and all is deinitialized properly, here the ouptut:

```
W (85305) ltem_task: LTEM DETACH State Entered ---> In DETACH I call esp_modem_stop_ppp(dte_modem);
D (93605) esp-modem: debug_data: 0x3ffd7bc8   0d 0a 4f 4b 0d 0a                                 |..OK..|
D (93605) saraR4: EXIT DATA Handle Response:
OK

D (93615) saraR4: enter command mode ok
D (93615) event: running post ESP_MODEM_EVENT:3 with handler 0x400d90f8 and context 0x3ffda234 on loop 0x3ffd80f8
I (93625) ltem_task: Modem PPP Stopped
D (93625) event: running post ESP_MODEM_EVENT:3 with handler 0x400e5d08 and context 0x3ffda304 on loop 0x3ffd80f8
D (93635) esp_netif_handlers: esp_netif action stopped with netif0x3ffd780c from event_id=3
D (93645) esp-netif_lwip-ppp: esp_netif_stop_ppp: Stopped PPP connection: 0x3f8001cc
ppp_close[1]: kill_link -> lcp_close
ppp phase changed[1]: phase=11
D (93665) esp-netif_lwip-ppp: Phase Terminate
D (93665) event: running post NETIF_PPP_STATUS:267 with handler 0x400d90c0 and context 0x3ffd77d4 on loop 0x3ffc7fb8
I (93675) ltem_task: PPP state changed event 267
D (93685) event: running post NETIF_PPP_STATUS:267 with handler 0x400dbbdc and context 0x3ffda504 on loop 0x3ffc7fb8
ppp phase changed[1]: phase=9
D (93695) esp-netif_lwip-ppp: Phase Network
D (93695) event: running post NETIF_PPP_STATUS:265 with handler 0x400d90c0 and context 0x3ffd77d4 on loop 0x3ffc7fb8
I (93715) ltem_task: PPP state changed event 265
D (93715) event: running post NETIF_PPP_STATUS:265 with handler 0x400dbbdc and context 0x3ffda504 on loop 0x3ffc7fb8
sifvjcomp[1]: VJ compress enable=0 slot=0 max slot=0
sifdown[1]: err_code=5
ppp phase changed[1]: phase=6
D (93735) esp-netif_lwip-ppp: Phase Establish
D (93745) event: running post NETIF_PPP_STATUS:262 with handler 0x400d90c0 and context 0x3ffd77d4 on loop 0x3ffc7fb8
I (93755) ltem_task: PPP state changed event 262
D (93755) event: running post NETIF_PPP_STATUS:262 with handler 0x400dbbdc and context 0x3ffda504 on loop 0x3ffc7fb8
pppos_send_config[1]: out_accm=FF FF FF FF
ppp_send_config[1]
pppos_recv_config[1]: in_accm=0 0 0 0
ppp_recv_config[1]
sent [LCP TermReq id=0x2 "User request"]
pppos_write[1]: len=20
D (93785) esp_netif_lwip: check: remote, if=0x3ffd780c fn=0x400e6db4

D (93785) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffd780c
D (93795) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (93805) dce_service: hang up ok
sent [LCP TermReq id=0x3 "User request"]
pppos_write[1]: len=20
ppp phase changed[1]: phase=12
D (105785) esp-netif_lwip-ppp: Phase Disconnect
D (105785) event: running post NETIF_PPP_STATUS:268 with handler 0x400d90c0 and context 0x3ffd77d4 on loop 0x3ffc7fb8
I (105785) ltem_task: PPP state changed event 268
D (105795) event: running post NETIF_PPP_STATUS:268 with handler 0x400dbbdc and context 0x3ffda504 on loop 0x3ffc7fb8
Connection terminated.
ppp_link_terminated[1]
ppp_link_end[1]
ppp phase changed[1]: phase=0
D (105815) esp-netif_lwip-ppp: Phase Dead
D (105815) event: running post NETIF_PPP_STATUS:256 with handler 0x400d90c0 and context 0x3ffd77d4 on loop 0x3ffc7fb8
I (105825) ltem_task: PPP state changed event 256
D (105835) event: running post NETIF_PPP_STATUS:256 with handler 0x400dbbdc and context 0x3ffda504 on loop 0x3ffc7fb8
I (105845) esp-netif_lwip-ppp: User interrupt
D (105845) event: running post NETIF_PPP_STATUS:5 with handler 0x400d90c0 and context 0x3ffd77d4 on loop 0x3ffc7fb8
I (105855) ltem_task: PPP state changed event 5
I (105865) ltem_task: User interrupted event from netif:0x0
D (105865) event: running post NETIF_PPP_STATUS:5 with handler 0x400dbbdc and context 0x3ffda504 on loop 0x3ffc7fb8
I (105875) esp-modem-netif: PPP state changed event 5
ppp_link_terminated[1]: finished.
W (105885) ltem_task: LTEM DEINIT State Entered
D (105895) saraR4: power down ok
D (105905) event: deleted loop 0x3ffd80f8
D (105905) esp_netif_objects: esp_netif_remove_from_list netif successfully removed (total netifs: 0)
ppp_free[1]
I (105915) ltem_task: LTEM task has exited
```

And sometimes the ppp_stop is not working properly and the output is the following, and hanging:

```
W (83645) ltem_task: LTEM DETACH State Entered
D (118075) esp-modem: Not sending data in transition mode
pppos_netif_output[1]: output failed proto=0x21, len = 71
D (118075) esp-modem: Not sending data in transition mode
pppos_netif_output[1]: output failed proto=0x21, len = 71
....
....
D (122585) esp-modem: Not sending data in transition mode
pppos_netif_output[1]: output failed proto=0x21, len = 71
```

I do believe that there are two errors that can happen:

- DTE never sends the "+++"
  - I will check with an oscilloscope if this can possibly happen
  - Before proceeding I would like to ask you if this scenario can happen

- DCE receives/recognize sporadically the escape sequence "+++" 
  - I've contacted uBlox in order to understand if this can happen, as the datasheet is not really well written.

david-cermak, 2020-12-16:
@fromeijn Thanks for your feedback, too.

> Also added a 1 second wait after mqtt disconnection, otherwise will get handle line failed.

That's a good point and could be added to the example, but in fact we have to ensure that the modem component could actually exit the PPP mode any time, even in the middle of active communication. then some error messages could appear, some command would fail but in the end the PPP mode stops.

> Any suggestions to test more or to improve?

I don't think so, but I'm open to suggestions. Thanks for offering to help. (as mentioned above I'd look into reworking the uart modes and possibly make use of the CMUX mode, but that's future consideration). Also these periodic switching between command and data mode is not a typical use-case of a modem component.

david-cermak, 2020-12-16:
@dragan-cvetkov 
> And sometimes the ppp_stop is not working properly and the output is the following, and hanging:

This is I think what would happen if the device gets lost the connection, so normally we would get the `IP_LOST` event and could programatically react to it, but it it's the second and later iteration of the loop we never get this event. Need to fix reset states in lwip (mentioned ^^^)

> DTE never sends the "+++"

I don't think this is happening, the escape sequence should be send always when exiting the PPP mode

> DCE receives/recognize sporadically the escape sequence "+++"

I think it's possible, depending on the device used. For example datasheet of the `SIM7600` says to repeat the `"+++"` sequence (including the 1 second delays) until it's successfully processed.

dragan-cvetkov, 2020-12-17:
@david-cermak, again thank you for your feedback.
> > DTE never sends the "+++"

> I don't think this is happening, the escape sequence should be send always when exiting the PPP mode

Indeed this is correct, I've tested it and escape sequence is always being sent.

> > DCE receives/recognize sporadically the escape sequence "+++"

> I think it's possible, depending on the device used. For example datasheet of the SIM7600 says to repeat the "+++" sequence (including the 1 second delays) until it's successfully processed.

This is the case. In my case (Sara R4) the DCE sporadically never answers back (I've also increased the timeout to 30s). It looks like the DCE is stuck.

Nevertheless, with the [PATCH](https://github.com/espressif/esp-idf/issues/6024#issuecomment-744414011), esp_modem_stop_ppp(dte) works now correctly.

In regards to the LWIP reset states issue, is there an timeframe when this will be patched?

fromeijn, 2020-12-17:
Hey also something I've noticed, that the `pppos_client` example is not working using `UART_NUM_2`. Something to keep in mind. see issue #5746

closed 2021-01-05 by espressif-bot

david-cermak, 2021-01-06:
The fix is now available on master and will be backported to other releases soon.

> In regards to the LWIP reset states issue, is there an timeframe when this will be patched?

As for the lwIP issue, we first have to look into the options to fix it somehow in the port/esp-netif layers if possible.

jonathandreyer, 2021-01-06:
@david-cermak Thanked for the feedback. I didn't forgot to test more longer as requested [here](https://github.com/espressif/esp-idf/issues/6024#issuecomment-746011809)) but I need more time. Currently, I am switching of dev. board to have a better test bench (based on [this one](https://github.com/Xinyuan-LilyGO/LilyGo-T-PCIE) with some modifications). I will be back with results in a couple of days.

tcpipchip, 2021-11-12:
@david-cermak 
First of all, thanks for all the info!
I would like to tests your ESP32 ppp routine in my U-blox modem SARA U200 and SARA G350.
I was using the ESP-IDF 4.2 got by the Platformio.
Must i apply the PATCH on that ? If yes, where are the patchs ? Or, how i can i do the patch command ?
Thank you!

david-cermak, 2021-11-12:
@tcpipchip The fix has been merged to `v4.2` branch as 84039dc77d1939a681879a057ce6fe959a5bc670. You can check if the version referenced by PlatformIO contains this commit.

```bash
git tag --contains 84039dc77d1939a681879a057ce6fe959a5bc670
```
is telling me, that the fix is available on both bugfix releases `v4.2.1` and `v4.2.2`

tcpipchip, 2021-11-12:
@david-cermak 
I am installing the ESP-IDF using Platformio 4.2.2 releases
Hope to compile and test on 3 u-blox modules soon!
I hope not get the x.509 certificate expire error too!
Thanks again!

tcpipchip, 2021-11-12:
@david-cermak 
Merged 1 ELF section
Generated C:/Users/maw/esp/esp-idf/examples/protocols/pppos_client/build/bootloader/bootloader.bin
[623/946] Generating x509_crt_bundle
FAILED: esp-idf/mbedtls/x509_crt_bundle
cmd.exe /C "cd /D C:\Users\maw\esp\esp-idf\examples\protocols\pppos_client\build\esp-idf\mbedtls && C:\Users\maw\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe C:/Users/maw/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input C:/Users/maw/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem -q"
gen_crt_bundle.py: Invalid certificate in C:/Users/maw/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem
Invalid certificate
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Must i disable the certificates ?

tcpipchip, 2021-11-12:
@david-cermak 
[image]
Looks that all is fine!
I hope to test soon on U-BLOX
This version solve the auto-connect problem ? btw, all problems in this big discussion ?
thanks again!

tcpipchip, 2021-11-13:
starting tests with u-blox modules

[5 more replies]