espressif/esp-idf #6025: ESP32 error to set breakpoints

jarcos-al, 2020-10-26:
## Environment

- Development Kit:      none
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used: ESP32-WROVER-E
- IDF version (run ``git describe --tags`` to find it): ESP-IDF v4.3-dev-dirty
- Build System:   CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: Eclipse with Espressif plugging
- Power Supply:      external 3.3V

## Problem Description

I have an error when I put a breakpoint while I debug my code.
If I run the debugger and I don't set any breakpoint, the issue dont appear.
When this issue appear, the console throws Ilegal instruction errors in the initialization.
The error only appears when I start the debugger with a breakpoin or more

The error that appear in the console:
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400ea098: e2006136 d11004c1 00a0c2b1
Setting breakpoint at 0x400ea09e and returning...

### Steps to reproduce

1. launch openocd enviroment
2. put a breakpoint
3. init debug
4. run the code

- [x] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/5438516/sdkconfig.zip)

Alvin1Zhang, 2020-10-27:
Thanks for reporting, we will look into.

gerekon, 2020-11-05:
@jarcos-al Could you collect OpenOCD and GDB logs as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/tips-and-quirks.html#reporting-issues-with-openocd-gdb) and attach them?

Alvin1Zhang, 2021-01-08:
@jarcos-al Thanks for reporting, would you please help share if an updates for this issue? Thanks.

jarcos-al, 2021-01-08:
Now the problem occurs only when I put more than 1 breakpoint. Software breakpoints should not be enabled.

gerekon, 2021-01-08:
@jarcos-al Could you post the logs I asked above? Without them it is hard to help you.

tom-borcin, 2021-07-19:
Hello @jarcos-al, could you please provide the logs @gerekon asked for? Thank you!

gerekon, 2024-09-11:
Closing due to no response from the author