espressif/esp-idf #6026: https_async issue on eth

KaeLL, 2020-10-26:
## Environment

- Module or chip used:  ESP32-D0WDQ6
- IDF version: v4.1
- Build System: CMake
- Compiler version: esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: external 5V

## Problem Description

`https_async()` from the esp_http_client example works fine on WiFi, but not on my W5100 port, and I can't figure out why.
I mean, apparently it works, as it returns 200, but I get the `TRANS_SSL: esp_tls_conn_read error, errno=No more processes` error every time, along with the failed attempts to connect to postman-echo, and I'd like to know why.

### Expected Behavior

Connection to postman-echo works fine, and no `esp_tls_conn_read error`.

### Code to reproduce this issue and steps to reproduce

Build and run [this](https://github.com/KaeLL/ESP32-W5100/tree/httpc_async_issue) PoC app I made.

[Logs](https://gist.github.com/KaeLL/d027708d20519e35b58e864fe5ef3f3a)

Alvin1Zhang, 2020-10-27:
Thanks for reporting.

suda-morris, 2020-11-26:
we have pushed the W5500 driver to master, the example you mentioned works well there. W5100 might have much similarity with W5500.

KaeLL, 2020-11-26:
Will check it out.

Alvin1Zhang, 2020-12-08:
@KaeLL Thanks for reporting, would you please help share if any updates for this issue? Thanks.

KaeLL, 2020-12-08:
I compared the implementations, and I didn't find any outstanding differences, hardware ones aside. Something I'm not aware of must be slipping by. Since I don't have a W5500 and I can't do a meticulous investigation on this right now, I'll close the issue, and reopen in case I have something new related to it to share.

closed 2020-12-08 by KaeLL