espressif/esp-idf #6028: DNS solving problem when more than 1 type A answers received

JB-DX, 2020-10-26:
ESP-IDF:  v4.3-dev-1521-g0289d1cc8

During OTA tests I stumbled across a weird DNS solving problem.

OTA fails when trying to connect to my local pc "celsius" with this url:
`http://celsius:81/my_main-debug.bin`

celsius' IP-adr is 192.168.0.6
However ESP32' debug-log shows that it is trying to connect to 192.168.0.96, that adr is not used.

Pinging to "celsius" from several other PCs is successful however.

So why does ESP32 fail?

Wireshark reveals that a DNS query to the router (Fritzbox) returns two type A answers,
  192.168.0.96
and
  192.168.0.6

ESP32 just uses the 1st entry, and doesn't try the 2nd entry when the first one fails.
Apparently Windows is smarter and tries all until a connection is successful.

I have not found a configuration or parameter for ESP32 to behave the same.

I was puzzled why 192.168.0.96 made it into the routers DNS table.
Most likely that was the adr when the PC first connected via DHPC.
Later on I forced the router to use the fixed adr 192.168.0.6. 
Apparently the first adr remained in the table and never was removed.
Since Windows handles this situation without problems, one can assume that a similar situation may exists in the field and should not lead to failures. Therefore I like to have it fixed.

Any ideas how to do that?

boarchuz, 2020-10-27:
https://github.com/espressif/esp-lwip/blob/61d840ff4778f4946c8743f7e412345abcd537f1/src/api/netdb.c#L247-L267

LWIP will only return the first answer. 

I think it's reasonable to fail here given the server is at fault returning an invalid (expired) answer. Attempting multiple connections is arguably unexpected behaviour in an embedded context, and even early desktop browsers (Safari 4.0, Firefox 2.0, Chrome 3.0) did not handle multiple answers. (Source: https://back.nber.org/sys-admin/dns-failover.html)
Easiest fix in the field is to restart your router.

Alvin1Zhang, 2020-10-27:
Thanks for reporting.

JB-DX, 2020-10-27:
@boarchuz
I checked Win10 a bit more using our own app that uses TCP, not HTTP.
For name resolving we just had switched from gethostbyname()  to getaddrinfo() due to some other problems. MS recommends not using gethostbyname()  anymore.

Now I tried that app in this situation.
gethostbyname() returns the 1st entry, which failed.
getaddrinfo() returns the working adr, in this case #2

So the app can rely on the DNS solver function and does not need to handle such situation itself. Browser do the same and don't see that problem as you asumed.

Resetting the router does not help, it keeps the adresses. I will query the maker about the reasons.

JB-DX, 2020-10-28:
Update:
After resetting the router the order of ip-addresses have changed. 
Now the working 192.168.0.6 is at position 1, so ESP32 can get it.

david-cermak, 2021-07-09:
@JB-DX Could you please close the issue if no more updates? If I understand correctly you were able to correctly resolve multiple dns entries using `getaddrinfo()` (but only one with `gethostbyname()`) which is the expected behaviour.

JB-DX, 2021-07-11:
gethostbyname() solved it only on the Windows side, for ESP32 there's no solution available, and will not as @boarchuz suggested. Restarting the router fixed it here. I only hope this situation will never happen with our customers.
So I will close this issue now.

closed 2021-07-11 by JB-DX