espressif/esp-idf #6030: Inaccurate timer tiimeout when light sleep

AxelLin, 2020-10-27:
## Environment

- Module or chip used: ESP32-WROOM-32E
- IDF version: v4.3-dev-1561-g357a27760322
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Power Supply: USB

## Problem Description

I setup a periodic timer which will timeout every 1 hour.
In my code, I use esp_timer_get_next_alarm to calculate the sleep time required,
then set esp_sleep_enable_timer_wakeup(time_until_next_alarm); and enter sleep
by esp_light_sleep_start();

When I check the log (with timestamp enabled), I found the esp32 timer seems faster
for about 12 seconds every hour.

Below I grep out the TIMEOUT part of the log:
The first field is the PC log time (with +8 TZ offset, you can find the problme by checking the log time v.s. TIMEOUT time)

[16:30:41:851] <0x1b>[0;32mI (5756) example: TIMEOUT: The current time is: Mon Oct 26 08:30:53 2020<0x1b>[0m␍␊
[17:30:29:695] <0x1b>[0;32mI (5796) example: TIMEOUT: The current time is: Mon Oct 26 09:30:53 2020<0x1b>[0m␍␊
[18:30:17:829] <0x1b>[0;32mI (5836) example: TIMEOUT: The current time is: Mon Oct 26 10:30:53 2020<0x1b>[0m␍␊
[19:30:05:888] <0x1b>[0;32mI (5876) example: TIMEOUT: The current time is: Mon Oct 26 11:30:53 2020<0x1b>[0m␍␊
[20:29:52:905] <0x1b>[0;32mI (5916) example: TIMEOUT: The current time is: Mon Oct 26 12:30:53 2020<0x1b>[0m␍␊
[21:29:40:050] <0x1b>[0;32mI (5956) example: TIMEOUT: The current time is: Mon Oct 26 13:30:53 2020<0x1b>[0m␍␊
[22:29:25:413] <0x1b>[0;32mI (5996) example: TIMEOUT: The current time is: Mon Oct 26 14:30:53 2020<0x1b>[0m␍␊
[23:29:12:101] <0x1b>[0;32mI (6036) example: TIMEOUT: The current time is: Mon Oct 26 15:30:53 2020<0x1b>[0m␍␊
[00:28:59:107] <0x1b>[0;32mI (6076) example: TIMEOUT: The current time is: Mon Oct 26 16:30:53 2020<0x1b>[0m␍␊
[01:28:45:694] <0x1b>[0;32mI (6116) example: TIMEOUT: The current time is: Mon Oct 26 17:30:53 2020<0x1b>[0m␍␊
[02:28:31:409] <0x1b>[0;32mI (6156) example: TIMEOUT: The current time is: Mon Oct 26 18:30:53 2020<0x1b>[0m␍␊
[03:28:17:145] <0x1b>[0;32mI (6196) example: TIMEOUT: The current time is: Mon Oct 26 19:30:53 2020<0x1b>[0m␍␊
[04:28:02:888] <0x1b>[0;32mI (6236) example: TIMEOUT: The current time is: Mon Oct 26 20:30:53 2020<0x1b>[0m␍␊
[05:27:48:443] <0x1b>[0;32mI (6276) example: TIMEOUT: The current time is: Mon Oct 26 21:30:53 2020<0x1b>[0m␍␊
[06:27:34:223] <0x1b>[0;32mI (6316) example: TIMEOUT: The current time is: Mon Oct 26 22:30:53 2020<0x1b>[0m␍␊
[07:27:20:867] <0x1b>[0;32mI (6356) example: TIMEOUT: The current time is: Mon Oct 26 23:30:53 2020<0x1b>[0m␍␊
[08:27:07:893] <0x1b>[0;32mI (6396) example: TIMEOUT: The current time is: Tue Oct 27 00:30:53 2020<0x1b>[0m␍␊
[09:26:53:463] <0x1b>[0;32mI (6436) example: TIMEOUT: The current time is: Tue Oct 27 01:30:53 2020<0x1b>[0m␍␊
[10:26:40:147] <0x1b>[0;32mI (6476) example: TIMEOUT: The current time is: Tue Oct 27 02:30:53 2020<0x1b>[0m␍␊
[11:26:31:322] <0x1b>[0;32mI (6516) example: TIMEOUT: The current time is: Tue Oct 27 03:30:53 2020<0x1b>[0m␍␊
[12:26:22:342] <0x1b>[0;32mI (6556) example: TIMEOUT: The current time is: Tue Oct 27 04:30:53 2020<0x1b>[0m␍␊
[13:26:12:455] <0x1b>[0;32mI (6596) example: TIMEOUT: The current time is: Tue Oct 27 05:30:53 2020<0x1b>[0m␍␊

### Expected Behavior
I would expected the timer timeout happen every hour in my set up.

### Actual Behavior
The timer seems happens too early (about 12 seconds before 1 hour) for each timeout.
You can find the PC log timestamp v.s. the TIMEOUT time.

### Code to reproduce this issue

Attached the full log and the code I used.
I modify sntp_example_main.c, I just need ntp sync once during boot so the esp32 time matches my pc time.
Then from the log you can find esp32 time and pc time becomes different.

To run the test code, copy it to examples/protocols/sntp/main/sntp_example_main.c and build it.

[sntp_example_main.c.txt](https://github.com/espressif/esp-idf/files/5443185/sntp_example_main.c.txt)
[timer_lightsleep_log.txt](https://github.com/espressif/esp-idf/files/5443193/timer_lightsleep_log.txt)

igrr, 2020-10-27:
@AxelLin Which RTC clock source are you using? You can check for `CONFIG_ESP32_RTC_CLOCK_SOURCE_*` in sdkconfig.

12 seconds per hour, or 0.3%, is within the frequency (in-)stability of both internal RC oscillator options in the ESP32 (150kHz and 8MHz/256).

AxelLin, 2020-10-27:
Hi @igrr 

I use the default setting in sntp example:
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1=y
CONFIG_ESP32_RTC_CLK_SRC_INT_RC=y
CONFIG_ESP32_RTC_CLK_CAL_CYCLES=1024

Note: If I don't enter light sleep, the timeout looks fine. That is why I send this bug report.

igrr, 2020-10-27:
In light sleep and deep sleep modes, timekeeping is done with the help of the RTC clock, as the high-resolution timers driven from XTAL are either clock gated or powered off. The only thing I may suggest to get better timekeeping accuracy in this scenario is to a) either use an external 32k XTAL or an oscillator as clock source, or b) choose "8M/256" RTC clock source in menuconfig, and do more frequent wake-up and NTP sync.

AxelLin, 2020-10-28:
I got about 2 ~ 3 seconds time drift per hour if change to "8M/256" RTC clock source,
which is better than 12 seconds drift per hour when using internal 150kHz RC osillator.

BTW, I think frequently NTP sync does not help because the issue is related to the periodic timer timeout.
(The timer timeout just happened faster than expected time.)

closed 2020-11-21 by AxelLin