espressif/esp-idf #6031: PSRAM initialization failed error

Sanketa54, 2020-10-27:
## Environment

Environment
Module or chip used: ESP32D2WD
IDF version: v4.3-dev-1561-g357a27760322
Build System: idf.py and vscode idf extention
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
Operating System: WIndows 10
Power Supply: USB
PSRAM = External ( 64 MB)

## Problem Description

I have been configured external psram with ESP32 D2WD chip tried on idf platform with esp-idf extention on vs code.
I flashed the example of https://github.com/espressif/espidf/tree/master/examples/system/himem but not getting the expected output.

### Expected Behavior
I (254) spiram: SPI RAM mode: flash 40m sram 40m
I (259) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
### Actual Behavior
10:52:02.435 -> [0;32mI (203) psram: This chip is ESP32-D2WD[0m
10:52:02.435 -> [0;31mE (209) cpu_start: Failed to init external RAM![0m

PLease help me with the solution or suggest me easy example just to check the psram

Spritetm, 2020-10-27:
What hardware are you using? How is the PSRAM connected? What GPIO pins do you have in use?

Sanketa54, 2020-10-28:
@Spritetm : PSRAM : APS6404L-SQR-ZR (64Mb)
PSRAM is connected externally to ESP32D2WD chip revision1 with GPIO pins CLK - GPIO6 and CS- GPIO10

Spritetm, 2020-10-28:
CLK for GPIO6 is non-standard, I think - by default it should be GPIO9, unless you modified this in Menuconfig.

Sanketa54, 2020-10-28:
I tried modifying that in menuconfig but it didn't worked for me.
It was showing the same message "cpu_start: Failed to init external RAM!"