espressif/esp-idf #6032: idf.py set-started esp32 err: no module named click

huang443765159, 2020-10-27:
Hello.
I am using mac for idf tool installation.I installed the tutorial https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/get-started/index.html#get-started-set-up-tools and they were fine until I did step 7.
I got the error when i did idf.py set-started esp32 like this:
[image]
I have installed this library, but there is still something wrong, could you please help me？
[image]
Thanks.

huang443765159, 2020-10-27:
I have solved the problem！It took me hours... The problem is that I need to use python2.7 instead of 3.8.

closed 2020-10-27 by huang443765159

dobairoland, 2020-10-27:
Hi @huang443765159. I'm glad that you've resolved your issue. In case anybody else encounters this issue, I just want to note that it should have to work with Python 3.8 as well. As a matter of fact, I'm using it with that Python version right now. 

Usually, this kind of issue is caused by invoking not the same Python for which the package was installed. The output of `which python` changes to a Python environment with "espressif" in the path. After that, using `python -m pip` can be used to ensure that packages are manipulated for the right Python interpreter. It is worth noting that even `python $(which idf.py)` can be used to ensure or check if the same behavior is encountered in case `idf.py` is invoked with a different Python.

I hope this helps for any future similar issue.