espressif/esp-idf #6033: Bluetooth HID micropython example

street-grease-coder, 2020-10-27:
**Is your feature request related to a problem? Please describe.**
It's frustrating to use micropython for a BLE mouse using ESP32, example code is missing. Micropython strives for a cross-platform solution, so this might fall into espressif interest to develop, in fact, I think this would make ESP32 (even) more attractive as a platform.

**Describe the solution you'd like**

example code in micropython or importable library for a bluetooth mouse and/or keyboard, that works in linux, apple and windows environments. Basically [this](https://github.com/T-vK/ESP32-BLE-Mouse), for micropython.

**Describe alternatives you've considered**
Please give as many details as you can. Include suggestions for useful APIs or interfaces if relevant.

Arduino code for this is available: https://github.com/T-vK/ESP32-BLE-Mouse
Others have played around with the micropython code: https://gitee.com/walkline/esp32-ble/blob/master/ble/ble_hid.py

**Additional context**

I think this would make a compelling feature for esp32 (to use as wireless mice/keyboards), if there was more micropython support for the ESP32 BLE features.

chegewara, 2020-10-27:
This is esp-idf issue tracker, not micropython. Micropython has its own repository and as i saw they are adding BLE HID to library.

street-grease-coder, 2020-10-28:
they are adding general support - I don't think it's anyone's intention (and by anyone I mean the 1-2 people actively maintaining that repo) to implement it building on the Espressif's IDF. That is the issue

Alvin1Zhang, 2020-10-29:
Thanks for reporting.

closed 2020-10-29 by Alvin1Zhang