espressif/esp-idf #6035: Strange C++ exception behaviour

mcilloni, 2020-10-27:
## Environment

- Development Kit: none
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it): v4.0.1-465-g1c7607d4a
- Build System: CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB (ESP-Prog)

## Problem Description

I've been encountering a very weird issue with exceptions after updating my toolchain to 2020r3 (I'm on the latest commits of release/v4.0).

Our code has some parts that invoke an external library (nlohmann::json); we've been using it on the esp32 for almost a year now, without no relevant issues. This library provides an STL-like basic_json class that supports allocators (we use it to parse JSON on external SPIRAM memory on our WROVER chips) and STL-like access.

More in detail, it provides an at() method that behaves quite much like the std::map counterpart, i.e. it either returns a deserialized value of a given type or it throws a json::out_of_range exception. We have code that checks the existence of a key in a JSON object by invoking `json::at()` inside a block and then catching any `json::out_of_range` it throws; see

```C++
        template<typename RetType, JSON_TEMPLATE_PARAMS> 
        inline std::optional<RetType> json_read_opt(const GENERIC_JSON &j, const char *const key) {
            try {
                return j.at(key).template get<RetType>();
            // attempt workaround for lousy bug
            } catch (const typename GENERIC_JSON::out_of_range &e) {
                return std::nullopt;
            }
        }
```
where JSON_TEMPLATE_PARAMS and GENERIC_JSON are #defines that make the extremely long declaration of a generic basic_json more bearable.

`json::at()` internally calls `std::map::at()` in order to fetch from an internal instance of `std::map` the item associated with the given key, rethrowing `std::out_of_range` as `json::out_of_range` if the search fails. The at() function somewhat resembles the following code:

```C++
    const_reference at(const typename object_t::key_type& key) const
    {
        // at only works for objects
        if (JSON_HEDLEY_LIKELY(is_object()))
        {
            JSON_TRY
            {
                return m_value.object->at(key); // <- invocation of std::map::at() and source of the std::out_of_range exception
            }
            JSON_CATCH (std::out_of_range&)
            {
                // create better exception explanation
                JSON_THROW(out_of_range::create(403, "key '" + std::string(key) + "' not found"));
            }
        }
        else
        {
            JSON_THROW(type_error::create(304, "cannot use at() with " + std::string(type_name())));
        }
    }
```

The issue I'm facing is that _sometimes_ the std::out_of_range coming from inside `json::at()` _seemingly_ ends up being caught in a `catch (const std::exception&)` several frames up in the call stack (in one of the callers), even though it should not be possible at all due to it being wrapped in a handling try-catch block. 
I'm certain it's `std::out_of_range` and not another class because the exception caught by the `catch(const std::exception&)` handler has .what() == `"map::at"`, which in libstdc++ only occours when a `std::map` throws a `std::out_of_range`:

```rg
$  rg -F 'map::at' .espressif/
.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_map.h
539:      __throw_out_of_range(__N("map::at"));
548:      __throw_out_of_range(__N("map::at"));
``` 

I've also observed this behaviour to consistently happen in a precise point of the code (i.e. when an exception is thrown inside of these 3 nested try-catch, with the outer one catching for std::exception), and more weirdly that __changing unrelated parts of code__ (i.e. adding function calls or adding anything even in different components) __can prevent this behaviour from happening at all__; the code then functions correctly (as it always did before updating IDF and toolchain) until something more is added, which may cause the issue to reappear.

The caller function looks somewhat like this:

```
esp_err_t load_defs(const ext::json &input_json)  { 
		try {
			for (const auto &[name, j_val] : j.at("items").items()) {
				const auto d_t { dev_type::from_str(name) };
				
                                // here j_val is converted to the correct value by nlohmann::json
				if (!some_map.try_emplace(d_t, j_val).second) {
					// prints a log
				}
			}
		} catch (const our_exception_type &e) {
			// print what()
			return ESP_ERR_INVALID_ARG;
		} catch (const ext::json::exception &e) {
			// print what()

			return ESP_ERR_INVALID_ARG;
		} catch (const std::exception &e) { / / <- the catch() that's being hit
			// print what()

			return ESP_ERR_INVALID_ARG;
		}

		return ESP_OK;
}
```

The library then calls 

```
JSON_TEMPLATE
    void from_json(const GENERIC_JSON &j, some_type &t) {
        t.val1 = check_value(j, "R");
        t.val2 = check_value(j, "W");
        t.val3 = check_value(j, "N");
    }
```

`check_value` then calls `json_read_opt<>` ( see my previous comment), which has a try-catch inside, and then it itself calls json::at(), which has another nested try-catch() for std::out_of_range. Trying to catch std::exception inside of check_value has no effect; the exception ends up in "load_defs" anyway.

Given that I'm running this firmware on a custom board, I do not have functioning JTAG access, so I had to get creative in order to get a backtrace pinpointing to the exact spot the exception is thrown. Luckily, `std::map::at()` only throws `std::out_of_range` in two specific points, so I could simply modify the STL header to add an invocation of `terminate()` gated by a global boolean flag; this allowed me to get more or less determine where the exception is thrown, which is the point I marked in the aforementioned `json::at()` method (i.e. as soon as `map::at` throws due to the value not being found in the map).

```stacktrace
0x402a6022: std::terminate() at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:57

0x400f2ab0: nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, ext::allocator<char> >, bool, long long, unsigned long long, double, ext::allocator, nlohmann::adl_serializer, std::vector<unsigned char, std::allocator<unsigned char> > >::at(std::__cxx11::basic_string<char, std::char_traits<char>, ext::allocator<char> > const&) const at <redacted>/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_map.h:541
 (inlined by) nlohmann::basic_json<std::map, std::vector, std::__cxx11::basic_string<char, std::char_traits<char>, ext::allocator<char> >, bool, long long, unsigned long long, double, ext::allocator, nlohmann::adl_serializer, std::vector<unsigned char, std::allocator<unsigned char> > >::at(std::__cxx11::basic_string<char, std::char_traits<char>, ext::allocator<char> > const&) const at <redacted>/build/../components/nlohmann_json/upstream/include/nlohmann/json.hpp:3461
```

This, together with the weird fact that shuffling unrelated code around makes the "bad" catch disappear, makes me suspect there could be something wrong with either my board, the exception handling as generated by the compiler, or maybe my code.

Thanks a lot for the patience.

### Expected Behavior

The exception gets caught in the right handler, or the issue repeats itself deterministically.

### Actual Behavior

The exception apparently gets caught in the wrong handler, and only occours randomly, varying from one build to another.

Alvin1Zhang, 2020-10-28:
Thanks for reporting, we will look into.

0xjakob, 2021-09-02:
@mcilloni Does this issue still persist with the new tool chain (2021-r1)? Do you have a reproducer without involving the `json` library?

AxelLin, 2021-09-02:
> @mcilloni Does this issue still persist with the new tool chain (2021-r1)? Do you have a reproducer without involving the `json` library?

The reported issue is on esp-idf v4.0.1. I don't think toolchain 2021-r1 can work for v4.0 branch.

mcilloni, 2021-09-02:
> @mcilloni Does this issue still persist with the new tool chain (2021-r1)? Do you have a reproducer without involving the `json` library?

No, sorry, but I worked around the issue ages ago (by moving some code around) and it didn't reappear afterwards. I can't test with the 2021r1 toolchain because as @AxelLin said in his reply, we will be still stuck on IDF 4.0 for the time being.

0xjakob, 2021-09-02:
@AxelLin You're right, I missed that. If the problem occurs again, please let us know, in particular if it happens without the `json`. Otherwise it will be quite difficult to reproduce for us, in particular if it occurs sporadically. We haven't observed anything similar. We did have an exception handling bug in the upstream xtensa gcc at one point, though, but it has been fixed in 2020r3 or before.

0xjakob, 2021-10-09:
@mcilloni I'm closing this issue for now. If you encounter this behavior again, feel free to re-open it or open another issue.

closed 2021-10-09 by 0xjakob