espressif/esp-idf #6036: In IDF-4 wifi search does not work in connected state (STA - mode).

SergeySlyusarEZlo, 2020-10-27:
In IDF-4 wifi search does not work in connected state (STA - mode). in IDF-3 this functionality works fine.

In this case connection was allready established and got ip-address.
when and when i try to run wifi search  by esp_wifi_scan_start() 
returned warning:

W (94394) wifi:sta_scan: STA is connecting, scan are not allowed!

YouDONG-ESP, 2020-10-28:
Hi @SergeySlyusarEZlo, this log shows that STA is still connecting instead of in a connected state. This doc might help your issues: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html?highlight=scan#scan-when-wi-fi-is-connecting. Make sure STA is connected and then try scan, if this issue still happens, please provide us your commit ID and a complete log, we will take a look.

SergeySlyusarEZlo, 2020-10-29:
Hello, I checked again, but still reproduce

Git commit ID of using IDF: 4fe18a9258f437048cef19ddca31a2c16d5880e0

my log:

I (5571) wifi:mode : sta (b4:e6:2d:f9:8e:21)
D (5571) event: running post WIFI_EVENT:2 with handler 0x40231c8c on loop 0x3f80eb20
0x40231c8c: wifi_create_and_start_sta at /home/sergey/Work/ESP32/lib/esp-idf/components/tcpip_adapter/tcpip_adapter_compat.c:67

V (5571) esp_adapter: thread sem get: sem=0x3f819bf4
V (5571) esp_netif_objects: esp_netif_next_unsafe 0x0
V (5571) esp_netif_objects: esp_netif_next_unsafe 0x0
D (5571) esp_netif_objects: esp_netif_add_to_list 0x3f81abe4
D (5571) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 1)
D (5581) event: running post WIFI_EVENT:2 with handler 0x4025fd98 on loop 0x3f80eb20
0x4025fd98: wifi_default_action_sta_start at /home/sergey/Work/ESP32/lib/esp-idf/components/esp_wifi/src/wifi_default.c:70

D (5581) wifi_init_default: wifi_start esp-netif:0x3f81abe4 event-id2
D (5581) wifi_init_default: WIFI mac address: b4 e6 2d f9 8e 21
D (5581) esp_netif_handlers: esp_netif action has started with netif0x3f81abe4 from event_id=2
D (5581) esp_netif_lwip: check: remote, if=0x3f81abe4 fn=0x40233178
0x40233178: esp_netif_start_api at /home/sergey/Work/ESP32/lib/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:569

D (5591) esp_netif_lwip: esp_netif_start_api 0x3f81abe4
V (5591) esp_netif_objects: esp_netif_next_unsafe 0x0
V (5591) esp_netif_objects: esp_netif_next_unsafe 0x3f81abe4
V (5591) esp_netif_lwip: esp_netif_is_netif_up esp_netif:0x3f81abe4
D (5591) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (5591) event: running post SYSTEM_EVENT:2 with handler 0x40231dec on loop 0x3f80eb20
0x40231dec: esp_event_post_to_user at /home/sergey/Work/ESP32/lib/esp-idf/components/esp_event/event_loop_legacy.c:31

V (5671) esp_adapter: thread sem get: sem=0x3f819bf4
I (5671) start: WiFiManager started
V (5681) esp_adapter: thread sem create: sem=0x3f819bf4
V (5681) esp_adapter: thread sem get: sem=0x3f819bf4
I (7721) wifi:new:<13,2>, old:<1,0>, ap:<255,255>, sta:<13,2>, prof:1
I (8271) wifi:state: init -> auth (b0)
I (8271) wifi:state: auth -> assoc (0)
I (8281) wifi:state: assoc -> run (10)
D (8381) wpa: IEEE 802.1X RX: version=1 type=3 length=95

D (8381) wpa:   EAPOL-Key type=2

D (8381) wpa:   key_info 0x8a (ver=2 keyidx=0 rsvd=0 Pairwise Ack)

D (8381) wpa:   key_length=16 key_data_length=0

D (8381) wpa: WPA: RX EAPOL-Key - hexdump(len=99):
D (8381) wpa: 01 03 00 5f 02 00 8a 00 10 00 00 00 00 00 00 00
D (8381) wpa: 01 94 0a 75 58 7f db e9 a7 cf ee 74 22 23 c3 2f
D (8381) wpa: 4a 51 0e 89 c8 5b 54 0d fb d8 fb 66 43 a6 7c 22
D (8391) wpa: 8f 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
D (8391) wpa: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
D (8391) wpa: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
D (8391) wpa: 00 00 00
D (8391) wpa: WPA 1/4-Way Handshake

D (8391) wpa: RSN: msg 1/4 key data - hexdump(len=0):
D (8391) wpa: WPA: PTK derivation - A1=b4:e6:2d:f9:8e:21 A2=00:e0:80:00:6b:28

D (8401) wpa: WPA: PMK - hexdump(len=32):
D (8401) wpa: 6a 15 e2 2d e2 5e ae 32 f6 23 ba e6 17 35 2e 63
D (8401) wpa: a2 2a a6 06 8e 8a 4e e2 61 3c 8b e7 f7 0c fb 95
D (8401) wpa: WPA: PTK - hexdump(len=48):
D (8401) wpa: 14 65 19 bc 64 00 18 a7 2a ef 9b 4b bf 6d c6 96
D (8401) wpa: 2c bc 29 92 38 36 22 95 0d 2f f0 2b 49 9a 00 34
D (8401) wpa: 3b 72 71 4a 9d 75 ef 3d e5 37 02 f3 c1 0e 4f a5
D (8401) wpa: WPA: WPA IE for msg 2/4
 - hexdump(len=22):
[... 78 lines cut ...]
D (8491) esp_netif_lwip: if0x3f81abe4 start ip lost tmr: no need start because netif=0x3f81adc0 interval=120 ip=0
D (8491) esp_netif_lwip: starting dhcp client
D (8491) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (8501) event: running post SYSTEM_EVENT:4 with handler 0x40231dec on loop 0x3f80eb20
0x40231dec: esp_event_post_to_user at /home/sergey/Work/ESP32/lib/esp-idf/components/esp_event/event_loop_legacy.c:31

D (8501) wifi_eventHandler: wifi_eventHandler
D (8501) wifi_eventHandler: event: sta_connected
I (8541) wifi:AP's beacon interval = 102400 us, DTIM period = 1
V (10691) esp_adapter: thread sem get: sem=0x3f819bf4
D (11131) esp_netif_lwip: esp_netif_dhcpc_cb lwip-netif:0x3f81adc0
D (11131) esp_netif_lwip: if0x3f81abe4 ip changed=1
D (11131) event: running post IP_EVENT:0 with handler 0x4025fdc8 on loop 0x3f80eb20
0x4025fdc8: wifi_default_action_sta_got_ip at /home/sergey/Work/ESP32/lib/esp-idf/components/esp_wifi/src/wifi_default.c:124

D (11131) wifi_init_default: Got IP wifi default handler entered
D (11131) esp_netif_handlers: esp_netif action got_ip with netif0x3f81abe4 from event_id=0
I (11131) esp_netif_handlers: sta ip: 192.168.119.60, mask: 255.255.255.0, gw: 192.168.119.1
D (11131) event: running post SYSTEM_EVENT:7 with handler 0x40231dec on loop 0x3f80eb20
0x40231dec: esp_event_post_to_user at /home/sergey/Work/ESP32/lib/esp-idf/components/esp_event/event_loop_legacy.c:31

D (11141) wifi_eventHandler: wifi_eventHandler
D (11141) wifi_eventHandler: event: sta_got_ip
V (11141) esp_adapter: thread sem get: sem=0x3f81b458
W (11141) wifi:sta_scan: STA is connecting, scan are not allowed!

YouDONG-ESP, 2020-11-03:
Hi @SergeySlyusarEZlo, could you show me your code or the event handler? Seems like something goes wrong with your event_handler. You can compare it with mine. Replace scan.c by it than run scan example.

`/* Scan Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

/*
    This example shows how to scan for available set of APs.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "esp_log.h"
#include "esp_event.h"
#include "nvs_flash.h"
#define WIFI_CONNECTED_BIT BIT0
#define WIFI_FAIL_BIT      BIT1

#define DEFAULT_SCAN_LIST_SIZE CONFIG_EXAMPLE_SCAN_LIST_SIZE

static const char *TAG = "scan";
static int s_retry_num = 0; 
static EventGroupHandle_t s_wifi_event_group;

static void print_auth_mode(int authmode)
{
    switch (authmode) {
    case WIFI_AUTH_OPEN:
        ESP_LOGI(TAG, "Authmode \tWIFI_AUTH_OPEN");
        break;
    case WIFI_AUTH_WEP:
        ESP_LOGI(TAG, "Authmode \tWIFI_AUTH_WEP");
        break;
    case WIFI_AUTH_WPA_PSK:
        ESP_LOGI(TAG, "Authmode \tWIFI_AUTH_WPA_PSK");
        break;
    case WIFI_AUTH_WPA2_PSK:
        ESP_LOGI(TAG, "Authmode \tWIFI_AUTH_WPA2_PSK");
        break;
    case WIFI_AUTH_WPA_WPA2_PSK:
        ESP_LOGI(TAG, "Authmode \tWIFI_AUTH_WPA_WPA2_PSK");
        break;
    case WIFI_AUTH_WPA2_ENTERPRISE:
        ESP_LOGI(TAG, "Authmode \tWIFI_AUTH_WPA2_ENTERPRISE");
        break;
    default:
        ESP_LOGI(TAG, "Authmode \tWIFI_AUTH_UNKNOWN");
        break;
    }
}

static void print_cipher_type(int pairwise_cipher, int group_cipher)
{
    switch (pairwise_cipher) {
    case WIFI_CIPHER_TYPE_NONE:
        ESP_LOGI(TAG, "Pairwise Cipher \tWIFI_CIPHER_TYPE_NONE");
        break;
    case WIFI_CIPHER_TYPE_WEP40:
        ESP_LOGI(TAG, "Pairwise Cipher \tWIFI_CIPHER_TYPE_WEP40");
        break;
    case WIFI_CIPHER_TYPE_WEP104:
        ESP_LOGI(TAG, "Pairwise Cipher \tWIFI_CIPHER_TYPE_WEP104");
        break;
    case WIFI_CIPHER_TYPE_TKIP:
[... 122 lines cut ...]
    if (bits & WIFI_CONNECTED_BIT) {
        ESP_LOGI(TAG, "connected to ap SSID: password:");
    } else if (bits & WIFI_FAIL_BIT) {
        ESP_LOGI(TAG, "Failed to connect to SSID:, password:");
    } else {
        ESP_LOGE(TAG, "UNEXPECTED EVENT");
    }

    ESP_ERROR_CHECK(esp_event_handler_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, &event_handler));
    ESP_ERROR_CHECK(esp_event_handler_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, &event_handler));
    vEventGroupDelete(s_wifi_event_group);
}

void app_main(void)
{
    // Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );
    wifi_init_sta();
}
`

SergeySlyusarEZlo, 2020-11-03:
Checked, the same result, and i don't understand how can event handler affected the driver scan operations. I just invoke scan after got ip-addres and have error:

I (11131) esp_netif_handlers: **sta ip: 192.168.119.60**, mask: 255.255.255.0, gw: 192.168.119.1
D (11131) event: running post SYSTEM_EVENT:7 with handler 0x40231dec on loop 0x3f80eb20
0x40231dec: esp_event_post_to_user at /home/sergey/Work/ESP32/lib/esp-idf/components/esp_event/event_loop_legacy.c:31

D (11141) wifi_eventHandler: wifi_eventHandler
D (11141) wifi_eventHandler: event: sta_got_ip
V (11141) esp_adapter: thread sem get: sem=0x3f81b458
W (11141) **wifi:sta_scan: STA is connecting, scan are not allowed!**

Please note: This issue reproduced only on IDF-4.
**on IDF-3 it works fine.**

YouDONG-ESP, 2020-11-03:
@SergeySlyusarEZlo Just checked again, Invoke scan after sta is connected won't cause any problems in your IDF commit 4fe18a9 . `W (11141) wifi:sta_scan: STA is connecting, scan are not allowed!` shows that your wifi is not connected, but you've git Ip address. So I think you might do some other operations affect it. Can you show me your code or what else did you do besides scan and station?

closed 2020-12-09 by Alvin1Zhang