espressif/esp-idf #6037: Setting CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256=y in sdkconfig.defaults does not work

AxelLin, 2020-10-28:
## Environment

 - Module or chip used: ESP32-WROOM-32E
 - IDF version: v4.3-dev-1561-g357a27760322
 - Build System: idf.py
 - Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
 - Operating System: Linux
 - Power Supply: USB

## Problem Description

I want to use CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256=y instead of CONFIG_ESP32_RTC_CLK_SRC_INT_RC.

Edit sdkconfig.defaults file, with below settings:
#CONFIG_ESP32_RTC_CLK_SRC_INT_RC=n
CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256=y

$ rm sdkconfig
$ idf.py reconfigure -DSDKCONFIG_DEFAULTS="sdkconfig.defaults"
Executing action: reconfigure
Running cmake in directory /home/axel/esp/esp-idf/examples/protocols/sntp/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DSDKCONFIG_DEFAULTS=sdkconfig.defaults -DCCACHE_ENABLE=0 /home/axel/esp/esp-idf/examples/protocols/sntp"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Component directory /home/axel/esp/esp-idf/components/aws_iot does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/axel/esp/esp-idf/components/micro-ecc does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/axel/esp/esp-idf/components/nimble does not contain a CMakeLists.txt file. No component will be added
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/axel/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/axel/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/axel/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/axel/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/axel/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/axel/esp/esp-idf/examples/protocols/sntp/sdkconfig
Loading defaults file /home/axel/esp/esp-idf/examples/protocols/sntp/sdkconfig.defaults...
Traceback (most recent call last):
  File "/home/axel/esp/esp-idf/tools/kconfig_new/confgen.py", line 636, in <module>
    main()
  File "/home/axel/esp/esp-idf/tools/kconfig_new/confgen.py", line 322, in main
    output_function(deprecated_options, config, temp_file)
  File "/home/axel/esp/esp-idf/tools/kconfig_new/confgen.py", line 469, in write_json
    config_dict = get_json_values(config)
  File "/home/axel/esp/esp-idf/tools/kconfig_new/confgen.py", line 464, in get_json_values
    write_node(n)
  File "/home/axel/esp/esp-idf/tools/kconfig_new/confgen.py", line 461, in write_node
    val = int(val)
ValueError: invalid literal for int() with base 10: ''
CMake Error at /home/axel/esp/esp-idf/tools/cmake/kconfig.cmake:228 (message):
  Failed to run confgen.py
  (/home/axel/.espressif/python_env/idf4.3_py3.8_env/bin/python;/home/axel/esp/esp-idf/tools/kconfig_new/confgen.py;--kconfig;/home/axel/esp/esp-idf/Kconfig;--sdkconfig-rename;/home/axel/esp/esp-idf/sdkconfig.rename;--config;/home/axel/esp/esp-idf/examples/protocols/sntp/sdkconfig;--defaults;/home/axel/esp/esp-idf/examples/protocols/sntp/sdkconfig.defaults;--env-file;/home/axel/esp/esp-idf/examples/protocols/sntp/build/config.env).
  Error 1
Call Stack (most recent call first):
  /home/axel/esp/esp-idf/tools/cmake/build.cmake:442 (__kconfig_generate_config)
  /home/axel/esp/esp-idf/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:10 (project)

-- Configuring incomplete, errors occurred!
See also "/home/axel/esp/esp-idf/examples/protocols/sntp/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1

### Expected Behavior
idf.py reconfigure -DSDKCONFIG_DEFAULTS="sdkconfig.defaults" should work and
the sdkconfig should include CONFIG_ESP32_RTC_CLK_SRC_INT_8MD256=y setting.

### Actual Behavior
idf.py reconfigure -DSDKCONFIG_DEFAULTS="sdkconfig.defaults" fails
### Steps to reproduce

I use sntp example code to test:
1. cd examples/protocols/sntp
2. rm sdkconfig
3. add sdkconfig.defaults (as I mentioned above)
4. idf.py reconfigure -DSDKCONFIG_DEFAULTS="sdkconfig.defaults"

KonstantinKondrashov, 2020-10-28:
Hi @AxelLin! Thank you for the report. I see this happens on my side as well, I will fix it.

KonstantinKondrashov, 2020-10-28:
Hi @AxelLin!
If it is urgent, you can use this patch to fix it. It will be merged ASAP.
[11011.zip](https://github.com/espressif/esp-idf/files/5450458/11011.zip)
Thanks.

AxelLin, 2020-10-28:
Hi @KonstantinKondrashov 
It works with your patch.
Thanks a lot for the prompt response.

AxelLin, 2020-11-04:
Just to remind that v4.0, v4.1 and v4.2 also need the same fix.

AxelLin, 2020-11-23:
@Alvin1Zhang 
Please reopen the issue.
v4.0, v4.1 and v4.2-rc still hit the same error.

projectgus, 2020-11-23:
Hi @AxelLin,

Thanks for the reminder about this. Backports were planned to v4.2 and v4.1, and will be coming soon. We'll check v4.0 as well - thanks for letting us know.

Angus

closed 2021-03-12 by AxelLin

xobs, 2023-12-26:
For completeness, here is the patch from the zipfile so that you don't have to download it to see what the fix was:

```patch
diff --git a/components/esp32/Kconfig b/components/esp32/Kconfig
index 9362207f91acc5f949a47143fb687b56803e79bf..dcf4c57be915f14a54962e161804e65972d0e334 100644
--- a/components/esp32/Kconfig
+++ b/components/esp32/Kconfig
@@ -585,7 +585,7 @@ menu "ESP32-specific"
 
     config ESP32_RTC_CLK_CAL_CYCLES
         int "Number of cycles for RTC_SLOW_CLK calibration"
-        default 3000 if ESP32_RTC_CLK_SRC_EXT_CRYS
+        default 3000 if ESP32_RTC_CLK_SRC_EXT_CRYS || ESP32_RTC_CLK_SRC_EXT_OSC || ESP32_RTC_CLK_SRC_INT_8MD256 # Freq ~32kHz
         default 1024 if ESP32_RTC_CLK_SRC_INT_RC
         range 0 27000 if ESP32_RTC_CLK_SRC_EXT_CRYS || ESP32_RTC_CLK_SRC_EXT_OSC || ESP32_RTC_CLK_SRC_INT_8MD256
         range 0 32766 if ESP32_RTC_CLK_SRC_INT_RC
diff --git a/components/esp32s2/Kconfig b/components/esp32s2/Kconfig
index 3464590d4b153ca7d8137e598c8f0fe33b906b7f..2fc2f591cd98310d85e65e5b688126222e04bdce 100644
--- a/components/esp32s2/Kconfig
+++ b/components/esp32s2/Kconfig
@@ -441,7 +441,7 @@ menu "ESP32S2-specific"
 
     config ESP32S2_RTC_CLK_CAL_CYCLES
         int "Number of cycles for RTC_SLOW_CLK calibration"
-        default 3000 if ESP32S2_RTC_CLK_SRC_EXT_CRYS
+        default 3000 if ESP32S2_RTC_CLK_SRC_EXT_CRYS || ESP32S2_RTC_CLK_SRC_EXT_OSC || ESP32S2_RTC_CLK_SRC_INT_8MD256 # Freq ~32kHz
         default 576 if ESP32S2_RTC_CLK_SRC_INT_RC
         range 0 125000
         help
diff --git a/components/esp32s3/Kconfig b/components/esp32s3/Kconfig
index e1a1a7996f64b380351d5d093e0861c35a567ee9..124b60bf842315d0b859542690cf5cd3c370df48 100644
--- a/components/esp32s3/Kconfig
+++ b/components/esp32s3/Kconfig
@@ -462,7 +462,7 @@ menu "ESP32S3-Specific"
 
     config ESP32S3_RTC_CLK_CAL_CYCLES
         int "Number of cycles for RTC_SLOW_CLK calibration"
-        default 3000 if ESP32S3_RTC_CLK_SRC_EXT_CRYS
+        default 3000 if ESP32S3_RTC_CLK_SRC_EXT_CRYS || ESP32S3_RTC_CLK_SRC_EXT_OSC || ESP32S3_RTC_CLK_SRC_INT_8MD256 # Freq ~32kHz
         default 1024 if ESP32S3_RTC_CLK_SRC_INT_RC
         range 0 125000
         help
```