espressif/esp-idf #6038: Multiple root node in one mesh network.

jarus0, 2020-10-28:
As stated in esp/mdf idf mesh documentation one mesh network can have one root node. but I want multiple roots in the same mesh network.

In my requirement sometimes I have 50+ devices interconnected, so probably root will have congestion of messages. but at the location I could have multiple routers.

What kind of features I am expecting?
If any node tries to send a message to the outer network. It will first search for the shortest path with the strongest signal connection in the mesh.

If you suggest me to have two mesh network then I lose interconnected communication between nodes (in that scenario one mesh network node can't communicate with another mesh node directly).

zhangyanjiaoesp, 2023-08-25:
@jarus0 Now we only support one root node in the same network. If you are worried that the root node will be congested, you can adjust the `xon qsize` appropriately by calling `esp_mesh_set_xon_qsize()`, the max value of the `xon qsize` is 128.

Sherry616, 2024-03-04:
Hi @jarus0, could you please share the latest update of this issue? Thanks.