espressif/esp-idf #6039: ULP wakeup from deep sleep

fpgamaster, 2020-10-28:
Hi All,
The issue is not related to idf version or a dev kit. The environment used is based on ESP32 rev 1 SoC.
Looking at some examples, e.g. https://github.com/espressif/esp-iot-solution:
waking up from deep sleep is like:
....
'
exit:
	halt

	.global wake_up
wake_up:
	/* Check if the SoC can be woken up */
	READ_RTC_REG(RTC_CNTL_DIAG0_REG, 19, 1)
	and r0, r0, 1
	jump exit, eq

	/* Wake up the SoC and stop ULP program */
	wake
	/* Stop the wakeup timer so it does not restart ULP */
	WRITE_RTC_FIELD(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN, 0)
	halt
' 

At some places instead of READ_RTC_REG(RTC_CNTL_DIAG0_REG, 19, 1) -> READ_RTC_FIELD(RTC_CNTL_LOW_POWER_ST_REG, RTC_CNTL_MAIN_STATE_IN_IDLE) is used which is same.

In the code above if the RTC_CNTL_MAIN_STATE_IN_IDLE bit is zero the SoC will be never woken up.
I can not find in the ESP32 Technical Reference Manual any description of the RTC_CNTL_LOW_POWER_ST_REG. According to the esp-idf source codes it is DR_REG_RTCCNTL_BASE + 0xC0.

Can someone point me to a document explaining this register?
We a looking for a way to wake up the SoC in any cases from a deep sleep.

Thanks!

boarchuz, 2020-11-02:
>`READ_RTC_FIELD(RTC_CNTL_LOW_POWER_ST_REG, RTC_CNTL_MAIN_STATE_IN_IDLE)`

Where are you seeing this? MAIN_STATE_IN_IDLE is bit 27; bit 19 is RTC_CNTL_RDY_FOR_WAKEUP which is likely what you want instead.

>In the code above if the RTC_CNTL_MAIN_STATE_IN_IDLE bit is zero the SoC will be never woken up

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/ulp_instruction_set.html#wake-wake-up-the-chip

You can poll the bit until set, eg.
```
wake_up:
/* Check if the SoC can be woken up */
READ_RTC_REG(RTC_CNTL_LOW_POWER_ST_REG, RTC_CNTL_RDY_FOR_WAKEUP_S, 1)
/* Loop while not ready */
jumpr wake_up, 1, LT
/* Ready bit is now set, wake up the SoC */
wake
/* Stop the wakeup timer so it does not restart ULP */
WRITE_RTC_FIELD(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN, 0)
halt
```

fpgamaster, 2020-11-02:
HI @boarchuz ,
Just a wrong code copied, sorry!
READ_RTC_REG(RTC_CNTL_DIAG0_REG, 19, 1) is equal to READ_RTC_FIELD(RTC_CNTL_LOW_POWER_ST_REG, RTC_CNTL_RDY_FOR_WAKEUP)

Here is the actual code I use:

wake_up:
        READ_RTC_FIELD(RTC_CNTL_LOW_POWER_ST_REG, RTC_CNTL_MAIN_STATE_IN_IDLE)
        and r0, r0, 1
        jumpr wake_up_normal, 0, gt

wait_for_rdy:
        /* Wake up from deep sleep*/
        /* Check if the system can be woken up */
        READ_RTC_FIELD(RTC_CNTL_LOW_POWER_ST_REG, RTC_CNTL_RDY_FOR_WAKEUP)
        and r0, r0, 1
        jump wait_for_rdy, eq

wake_up_normal:
        /* Wake up the SoC, end program */
        wake
        /* Stop ULP timer */
        WRITE_RTC_FIELD(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN, 0)
        halt

krzychb, 2022-03-03:
Hi @fpgamaster, description of RTC_CNTL_LOW_POWER_ST_REG has been added to [ESP32 Technical Reference Manual](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf) (page 695). Please feel to reopen this report if more explanation is needed. I am sorry for missing updating this ticket.

closed 2022-03-03 by krzychb