espressif/esp-idf #6042: esp_wifi_80211_tx does not transmit packets after woken up from light sleep

zinke-ct-video, 2020-10-29:
### Hardware:
Board: 2 x ESP32-WROVER-IB (Rev. 3)
IDF: ESP-IDF, v4.3-dev-1561-g357a27760
Flash Frequency: 80Mhz
PSRAM enabled: yes

### Description:
I have two ESP32-WROVER modules (Rev. 3) which communicate over WiFi. One ESP is sending customized data packets using esp_wifi_80211_tx and the other one is receiving these data packets using the WiFi promiscuous mode. If there is nothing to send, the sender goes to light sleep. After the sender woke up from light sleep, new data is transmitted to the receiver. But this does not work. The function esp_wifi_80211_tx returns no error code, but the packet is actually NOT transmitted. This is the same issue as #3852, which was never solved.

Alvin1Zhang, 2020-10-30:
Thanks for reporting.

YouDONG-ESP, 2020-11-25:
Hi @zinke-ct-video, It actually works. First thing need to know, you can't call light sleep when your sta got connection, modem sleep will invoke it automatically.
If you don't got connection, Remember to call esp_wifi_stop() before light sleep and call esp_wifi_start() after wake up to make sure hardware is initialized. 
I didn't see your code so I give you my code here: 

`#include <string.h>
#include <time.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_sleep.h"
#include "esp_system.h"
#include "driver/uart.h"
#include "driver/gpio.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "lwip/err.h"
#include "lwip/sys.h"

#define PKT2SEND probe_req

        static const char probe_req[] = {
		 0x40, 0x00, //Frame Control version=0 type=0(management)  subtype=100 (probe request)
		 0x00, 0x00, //Duration
		 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, //A1 Destination address  broadcast
		 0x50, 0x02, 0x91, 0x00, 0x00, 0x00, //A2 Source address (STA Mac i.e. transmitter address)
		 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, //A3 BSSID
		 0x40, 0x9e, //sequence number (This would be overwritten if 'en_sys_seq' is set to true in fn call esp_wifi_80211_tx )
		 	 	 0x00,	//wlan.tag.number => 0
				 0x00,	//wlan.tag.length
				 0x01,	//wlan.tag.number => 1
				 0x08,	//wlan.tag.length => 8
				 0x8b,	//Supported Rates: 5.5(B) (0x8b)	(wlan.supported_rates)
				 0x96,	//Supported Rates: 11(B) (0x96)		(wlan.supported_rates)
				 0x82,	//Supported Rates: 1(B) (0x82)		(wlan.supported_rates)
				 0x84,	//Supported Rates: 2(B) (0x84)		(wlan.supported_rates)
				 0x0c,	//Supported Rates: 6 (0x0c)			(wlan.supported_rates)
				 0x18,	//Supported Rates: 12 (018)			(wlan.supported_rates)
				 0x30,	//Supported Rates: 24 (0x30)		(wlan.supported_rates)
				 0x60,	//Supported Rates: 48 (0x60)		(wlan.supported_rates)
				 0x32,	//wlan.tag.number => 50
				 0x04,	//wlan.tag.length => 4
				 0x6c,	//Extended Supported Rates: 54 (0x6c)	(wlan.extended_supported_rates)
				 0x12,	//Extended Supported Rates: 9 (0x12)	(wlan.extended_supported_rates
				 0x24,	//Extended Supported Rates: 18 (0x24)	(wlan.extended_supported_rates
				 0x48,	//Extended Supported Rates: 36 (0x48)	(wlan.extended_supported_rates
};

static EventGroupHandle_t s_wifi_event_group;

static const char *TAG = "wifi station";

void sleepa(void)
{
    esp_sleep_enable_timer_wakeup(2000000);
    printf("Entering light sleep\n");
    uart_wait_tx_idle_polling(CONFIG_ESP_CONSOLE_UART_NUM);
    int64_t t_before_us = esp_timer_get_time();
    esp_light_sleep_start();
    int64_t t_after_us = esp_timer_get_time();
    printf("Returned from light sleep, reason: timer, t=%lld ms, slept for %lld ms\n", t_after_us / 1000, (t_after_us - t_before_us) / 1000);
}

static void event_handler(void* arg, esp_event_base_t event_base,
                                int32_t event_id, void* event_data)
{
    if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) {
        printf("start\n");
        esp_wifi_80211_tx(ESP_IF_WIFI_STA, PKT2SEND, sizeof(PKT2SEND), false);
        vTaskDelay(5000/portTICK_PERIOD_MS);
        esp_wifi_stop();
    }else if( event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_STOP){
        printf("stop\n");
        sleepa();
        esp_wifi_start();
    }
}

void wifi_init_sta(void)
{
    s_wifi_event_group = xEventGroupCreate();

    ESP_ERROR_CHECK(esp_netif_init());

    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_create_default_wifi_sta();

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    esp_event_handler_instance_t instance_any_id;
    esp_event_handler_instance_t instance_got_ip;
    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                                                        ESP_EVENT_ANY_ID,
                                                        &event_handler,
                                                        NULL,
                                                        &instance_any_id));
    ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT,
                                                        IP_EVENT_STA_GOT_IP,
                                                        &event_handler,
                                                        NULL,
                                                        &instance_got_ip));

    wifi_config_t wifi_config = {
        .sta = {
            .ssid = "xxxx",
            .password = "",
	     .threshold.authmode = WIFI_AUTH_OPEN,

            .pmf_cfg = {
                .capable = true,
                .required = false
            },
        },
    };
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
    ESP_ERROR_CHECK(esp_wifi_start() );

    ESP_LOGI(TAG, "wifi_init_sta finished.");
    esp_wifi_80211_tx(ESP_IF_WIFI_STA, PKT2SEND, sizeof(PKT2SEND), false);

}

void app_main(void)
{
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    wifi_init_sta();
}
`

(remember to set the channel of packet capture software to 1 bgn!!!)

zinke-ct-video, 2020-11-26:
Hi @YouDONG-ESP, this is a very helpful hint. I didn't call esp_wifi_start / stop before and after light sleep.

Thanks!

closed 2020-11-26 by zinke-ct-video