espressif/esp-idf #6046: ESP-IDF v 2.3 installer is showing an error and failing to install in Windows 10?

amulya-incorrigible, 2020-10-30:
I am posting my error log below. Please help me!
2020-10-30 10:45:36.736   --------
2020-10-30 10:45:36.745   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\asio\asio\objects\info\alternates
2020-10-30 10:45:36.747   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\bootloader\subproject\components\micro-ecc\micro-ecc\objects\info\alternates
2020-10-30 10:45:36.748   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\bt\controller\lib\objects\info\alternates
2020-10-30 10:45:36.750   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\bt\host\nimble\nimble\objects\info\alternates
2020-10-30 10:45:36.752   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\coap\libcoap\modules\ext\tinydtls\objects\info\alternates
2020-10-30 10:45:36.753   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\coap\libcoap\objects\info\alternates
2020-10-30 10:45:36.755   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\esptool_py\esptool\objects\info\alternates
2020-10-30 10:45:36.756   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\esp_wifi\lib_esp32\objects\info\alternates
2020-10-30 10:45:36.758   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\expat\expat\objects\info\alternates
2020-10-30 10:45:36.759   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\json\cJSON\objects\info\alternates
2020-10-30 10:45:36.761   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\libsodium\libsodium\objects\info\alternates
2020-10-30 10:45:36.762   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\lwip\lwip\objects\info\alternates
2020-10-30 10:45:36.764   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\mbedtls\mbedtls\objects\info\alternates
2020-10-30 10:45:36.765   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\mqtt\esp-mqtt\objects\info\alternates
2020-10-30 10:45:36.768   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\nghttp\nghttp2\modules\third-party\mruby\objects\info\alternates
2020-10-30 10:45:36.771   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\nghttp\nghttp2\modules\third-party\neverbleed\objects\info\alternates
2020-10-30 10:45:36.772   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\nghttp\nghttp2\objects\info\alternates
2020-10-30 10:45:36.773   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\protobuf-c\protobuf-c\objects\info\alternates
2020-10-30 10:45:36.775   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\spiffs\spiffs\objects\info\alternates
2020-10-30 10:45:36.776   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\components\unity\unity\objects\info\alternates
2020-10-30 10:45:36.778   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\modules\examples\build_system\cmake\import_lib\main\lib\tinyxml2\objects\info\alternates
2020-10-30 10:45:36.779   Removing C:\Users\amuly\Desktop\esp-idf-2\.git\objects\info\alternates
2020-10-30 10:45:36.780   idf_tools.py exists in IDF directory
2020-10-30 10:45:36.780   idf_tools.py command: C:\Users\amuly\anaconda3\python.exe "C:\Users\amuly\Desktop\esp-idf-2\tools\idf_tools.py" --idf-path C:\Users\amuly\Desktop\esp-idf-2
2020-10-30 10:45:36.780   Installing tools:C:\Users\amuly\anaconda3\python.exe "C:\Users\amuly\Desktop\esp-idf-2\tools\idf_tools.py" --idf-path C:\Users\amuly\Desktop\esp-idf-2 install
2020-10-30 10:45:37.102   Done, exit code=1
2020-10-30 10:45:37.102   --------
2020-10-30 10:45:37.102   Running command: C:\Users\amuly\anaconda3\python.exe "C:\Users\amuly\Desktop\esp-idf-2\tools\idf_tools.py" --idf-path C:\Users\amuly\Desktop\esp-idf-2 install
                          Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
                          Installing xtensa-esp32-elf@esp-2020r3-8.4.0
                          Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          WARNING: Download failure <urlopen error unknown url type: https>
                          WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          WARNING: Download failure <urlopen error unknown url type: https>
                          WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          Downloading xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          WARNING: Download failure <urlopen error unknown url type: https>
                          WARNING: Failed to download https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip to C:\.espressif\dist\xtensa-esp32-elf-gcc8_4_0-esp-2020r3-win64.zip.tmp
                          ERROR: Failed to download, and retry count has expired
                          Traceback (most recent call last):
                            File "C:\Users\amuly\Desktop\esp-idf-2\tools\idf_tools.py", line 1387, in <module>
                              main(sys.argv[1:])
                            File "C:\Users\amuly\Desktop\esp-idf-2\tools\idf_tools.py", line 1383, in main
                              action_func(args)
                            File "C:\Users\amuly\Desktop\esp-idf-2\tools\idf_tools.py", line 1167, in action_install
                              tool_obj.download(tool_version)
                            File "C:\Users\amuly\Desktop\esp-idf-2\tools\idf_tools.py", line 584, in download
                              raise DownloadError()
                          __main__.DownloadError
                          
2020-10-30 10:45:37.102   --------
2020-10-30 10:45:37.102   Message box (OK):
                          Installation has failed with exit code 1
2020-10-30 10:46:25.858   User chose OK.
2020-10-30 10:46:25.865   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2020-10-30 10:46:27.373   User chose Yes.
2020-10-30 10:46:27.404   CurStepChanged raised an exception.

kingaza, 2021-02-06:
@amulya-incorrigible , have you solved the problem or there is any progress? I met the same problem

krzychb, 2021-02-06:
Hi @kingaza and @amulya-incorrigible please check solution in https://github.com/espressif/esp-idf/issues/6367#issuecomment-771787725. The issue root cause is described earlier in https://github.com/espressif/esp-idf/issues/6367#issuecomment-755947923

kingaza, 2021-02-07:
@krzychb thank you. now it works!

georgik, 2021-04-16:
@amulya-incorrigible Thank you for reporting the problem.
@kingaza Thank you for confirming the solution.

ESP-IDF Tools Installer for Windows was updated, based on feedback from GitHub. 
Please use newer version which contains several corrections and system checks.
You can download new installer from: https://dl.espressif.com/dl/esp-idf/

closed 2021-04-16 by tom-borcin