espressif/esp-idf #6047: esp-netif_lwip-ppp: pppos_input_tcpip failed with -1 is not handled and does not result in Phase Change

teekaytk, 2020-10-30:
# Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it):
   v4.2-dev
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
 xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [Battery]

## Problem Description
I am running a PPPoS link to a Linux server using UART. Linux is running pppd and I have no problem connecting and exchanging data with the linux server. ESP32 is PPP client. However after a few minutes or a few hours, or after a lot of data is exchanged, I start to get 
esp-netif_lwip-ppp: pppos_input_tcpip failed with -1 
I am printing out the PPP Phases but they don't change. Seems like an underlying layer has broken, may be TCP or UART level but I am unable to recover from it unless I reboot the device.

### Expected Behavior
PPP link to keep working , self recover , or expose PHASE_DISOCNNECT or PHASE_DEAD so I can recreate the link.
### Actual Behavior
Multiple
esp-netif_lwip-ppp: pppos_input_tcpip failed with -1 
recieved

Alvin1Zhang, 2020-11-02:
Thanks for reporting.

MrCurio, 2020-12-09:
Hello @teekaytk ,
I think i may have solved this issue here #6094 . 

It seems to be a memory leak somewhere in your code or in the stack. I would firstly check your code 

Thanks

closed 2021-08-12 by Alvin1Zhang