espressif/esp-idf #6048: Possible dangling pointer in esp_netif_set_hostname_api

boarchuz, 2020-10-31:
- IDF version: (master)

## Problem Description

In esp_netif_set_hostname_api, p_netif->hostname may no longer be valid if the function returns for any reason after freeing esp_netif->hostname but before setting p_netif->hostname to the new pointer

https://github.com/espressif/esp-idf/blob/357a2776032299b8bc4044900a8f1d6950d7ce89/components/esp_netif/lwip/esp_netif_lwip.c#L1106-L1141

### Steps to reproduce

- Set a valid hostname so that esp_netif_set_hostname_api copies the string, sets p_netif->hostname, and returns ESP_OK
- Now repeat with an invalid hostname (eg. with strlen > ESP_NETIF_HOSTNAME_MAX_SIZE). The previous hostname will be freed, but p_netif->hostname still points to this memory as the function returns early.
- esp_netif_get_hostname will return ESP_OK but the pointer is not valid; also presumably lwip functions that use this pointer may send junk (eg. dhcp_option_hostname)

### Expected Behavior

- Keep p_netif->hostname in sync with esp_netif->hostname.

- Perhaps skip strdup (or free) if there is any error. Only copy string if it will be used.

Alvin1Zhang, 2020-11-02:
Thanks for reporting.

Alvin1Zhang, 2020-11-23:
Thanks for reporting and sorry for late reply, we have a fix under internal reviewing and the issue will close automatically once the fix is available on GitHub. Thanks.

closed 2020-12-13 by espressif-bot