espressif/esp-idf #6050: ESP32-E SecureBoot V2

lajollalerner, 2020-10-31:
Environment V4.2beta1 running on custom hardware using the ESP32-VROOM-E (V3 chip) and on DEVkit-ESP32-E

Hi all,

I followed the manual procedures for signing and buring the efuses for V2 secure boot. I am able to verify that the signed code (both bootloader and app) are signed using the espsecure verify_signature command. my two questions are,

1. Do I use espsecure.py sign_data on the bootloader.bin and the app.bin or is there a different way to sign the bootloader.bin?

`python espsecure.py sign_data -v 2 -k testkey.pem -o bootloader_signed.bin bootloader.bin`

2. When I used espefuse.py burn_key I noticed the printout of blk2 data is reversed from the bin file produced by espsecure.py digest_rsa_public_key. Is that correct?

BIN File:
00000000:  4d 89 59 e2 d5 b3 58 4e 0b d2 64 9b 86 78 bd 60  :M.Y...XN..d..x.`
00000010:  58 6b 58 a0 0b e7 f5 c6 10 9f c2 db 72 40 df 8b  :XkX.........r@..

Output of `python espefuse.py -p COMx burn_key secure_boot pub.bin`:

Burn keys to blocks:
 - BLK2 -> [8b df 40 72 db c2 9f 10 c6 f5 e7 0b a0 58 6b 58 60 bd 78 86 9b 64 d2 0b 4e 58 b3 d5 e2 59 89 4d]
        Disabling read to key block
        Disabling write to key block

Burn keys in efuse blocksThe key block will be read and write protected (no further changes or readback)

The device serial output is:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:2, clock div:2
secure boot v2 enabled
Sig block 0 invalid: Image digest does not match
secure boot verification failed

Thank you,
Ian

Alvin1Zhang, 2020-11-02:
Thanks for reporting.

supreetd21, 2020-11-02:
Hi @lajollalerner,
Question 1: `python espsecure.py sign_data -v 2` is the right method to sign your application and bootloader binaries.

With regards to question 2, your observation is correct, using the `secure boot` parameter to `burn_key` reverses & r/w protects the digest to be programmed. The `secure boot` parameter is relevant for secure boot v1, though.

Secure boot v2 does not require the digest to be reverse programmed or to be read-protected.
Thus, for secure boot v2, you can use -
1. `python espefuse.py -p COMx burn_key BLK2 pub.bin`
2. `python espefuse.py burn_key_digest secure_boot_signing_key.pem`

Thank you for reporting, will add a helper note regarding its usage.

lajollalerner, 2020-11-03:
Hi all,

Using the `python espefuse.py burn_key_digest secure_boot_signing_key.pem` still produced a reverse order in the espefuse summary output. I continue to get the following after using espsecure sign_data -v 2 -k mykey.pem.

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:9348
load:0x40078000,len:16376
load:0x40080400,len:4000
entry 0x40080688
[0;32mI (29) boot: ESP-IDF v4.2-beta1-dirty 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 10:02:07[0m
[0;32mI (29) boot: chip revision: 3[0m
[0;32mI (33) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (38) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (42) boot.esp32: SPI Flash Size : 8MB[0m
[0;32mI (47) boot: Enabling RNG early entropy source...[0m
[0;32mI (52) boot: Partition Table:[0m
[0;32mI (56) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (63) boot:  0 nvs              WiFi data        01 02 0000b000 00002000[0m
[0;32mI (70) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (78) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (85) boot:  3 app0             OTA app          00 10 00010000 001e0000[0m
[0;32mI (93) boot:  4 app1             OTA app          00 11 001f0000 001e0000[0m
[0;32mI (100) boot:  5 spiffs           Unknown data     01 82 003d0000 00010000[0m
[0;32mI (108) boot: End of partition table[0m
[0;32mI (112) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x291d0 (168400) map[0m
[0;32mI (185) esp_image: segment 1: paddr=0x000391f8 vaddr=0x3ffbdb60 size=0x04440 ( 17472) load[0m
[0;32mI (193) esp_image: segment 2: paddr=0x0003d640 vaddr=0x40080000 size=0x00404 (  1028) load[0m
[0;32mI (194) esp_image: segment 3: paddr=0x0003da4c vaddr=0x40080404 size=0x025cc (  9676) load[0m
[0;32mI (205) esp_image: segment 4: paddr=0x00040020 vaddr=0x400d0020 size=0xf0430 (984112) map[0m
[0;32mI (584) esp_image: segment 5: paddr=0x00130458 vaddr=0x400829d0 size=0x1cb8c (117644) load[0m
[0;32mI (636) esp_image: segment 6: paddr=0x0014cfec vaddr=0x00000000 size=0x02fe4 ( 12260) [0m
[0;32mI (641) esp_image: Verifying image signature...[0m
[0;32mI (641) secure_boot: Secure Boot eFuse bit(ABS_DONE_1) not yet programmed.[0m
[0;32mI (646) secure_boot: Verifying with RSA-PSS...[0m
[0;32mI (672) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (672) secure_boot_v2: enabling secure boot v2...[0m
[0;32mI (672) esp_image: segment 0: paddr=0x00001020 vaddr=0x3fff0030 size=0x00004 (     4) [0m
[0;32mI (681) esp_image: segment 1: paddr=0x0000102c vaddr=0x3fff0034 size=0x02484 (  9348) [0m
[0;32mI (693) esp_image: segment 2: paddr=0x000034b8 vaddr=0x40078000 size=0x03ff8 ( 16376) [0m
[0;32mI (704) esp_image: segment 3: paddr=0x000074b8 vaddr=0x40080400 size=0x00fa0 (  4000) [0m
[0;32mI (708) esp_image: Verifying image signature...[0m
[0;32mI (713) secure_boot: Secure Boot eFuse bit(ABS_DONE_1) not yet programmed.[0m
[0;32mI (720) secure_boot: Verifying with RSA-PSS...[0m
Sig block 0 invalid: Image digest does not match
[0;31mE (729) secure_boot: Secure Boot V2 verification failed.[0m
[0;31mE (735) esp_image: Secure boot signature verification failed[0m
[0;32mI (741) esp_image: Calculating simple hash to check for corruption...[0m
[0;31mE (757) esp_image: Image hash failed - image is corrupt[0m
[0;33mW (757) esp_image: image corrupted on flash[0m
[0;31mE (759) secure_boot_v2: bootloader image appears invalid! error 8194[0m
[0;31mE (766) boot: Secure Boot v2 failed (8194)[0m
[0;31mE (771) boot: OTA app partition slot 0 is not bootable[0m
[0;32mI (777) esp_image: segment 0: paddr=0x001f0020 vaddr=0x3f400020 size=0x291d0 (168400) map[0m
[0;32mI (850) esp_image: segment 1: paddr=0x002191f8 vaddr=0x3ffbdb60 size=0x04440 ( 17472) load[0m
[0;32mI (857) esp_image: segment 2: paddr=0x0021d640 vaddr=0x40080000 size=0x00404 (  1028) load[0m
[0;32mI (858) esp_image: segment 3: paddr=0x0021da4c vaddr=0x40080404 size=0x025cc (  9676) load[0m
[0;32mI (870) esp_image: segment 4: paddr=0x00220020 vaddr=0x400d0020 size=0xf0430 (984112) map[0m
[0;32mI (1249) esp_image: segment 5: paddr=0x00310458 vaddr=0x400829d0 size=0x1cb8c (117644) load[0m
[0;32mI (1300) esp_image: segment 6: paddr=0x0032cfec vaddr=0x00000000 size=0x02fe4 ( 12260) [0m
[0;32mI (1305) esp_image: Verifying image signature...[0m
[0;32mI (1305) secure_boot: Secure Boot eFuse bit(ABS_DONE_1) not yet programmed.[0m
[0;32mI (1311) secure_boot: Verifying with RSA-PSS...[0m
[0;32mI (1337) boot: Loaded app from partition at offset 0x1f0000[0m
[0;32mI (1337) secure_boot_v2: enabling secure boot v2...[0m
[0;32mI (1337) esp_image: segment 0: paddr=0x00001020 vaddr=0x3fff0030 size=0x00004 (     4) [0m
[0;32mI (1346) esp_image: segment 1: paddr=0x0000102c vaddr=0x3fff0034 size=0x02484 (  9348) [0m
[0;32mI (1358) esp_image: segment 2: paddr=0x000034b8 vaddr=0x40078000 size=0x03ff8 ( 16376) [0m
[0;32mI (1369) esp_image: segment 3: paddr=0x000074b8 vaddr=0x40080400 size=0x00fa0 (  4000) [0m
[0;32mI (1373) esp_image: Verifying image signature...[0m
[0;32mI (1378) secure_boot: Secure Boot eFuse bit(ABS_DONE_1) not yet programmed.[0m
[0;32mI (1385) secure_boot: Verifying with RSA-PSS...[0m
[... cut ...]
[0;31mE (1400) esp_image: Secure boot signature verification failed[0m
[0;32mI (1407) esp_image: Calculating simple hash to check for corruption...[0m
[0;31mE (1423) esp_image: Image hash failed - image is corrupt[0m
[0;33mW (1423) esp_image: image corrupted on flash[0m
[0;31mE (1425) secure_boot_v2: bootloader image appears invalid! error 8194[0m
[0;31mE (1432) boot: Secure Boot v2 failed (8194)[0m
[0;31mE (1437) boot: OTA app partition slot 1 is not bootable[0m
[0;31mE (1443) boot: No bootable app partitions in the partition table[0m
```

lajollalerner, 2020-11-06:
Hi All,

Any update on this? I sure could use some guidance. 

Thanks,
Ian

supreetd21, 2020-11-09:
Hi @lajollalerner,
Okay, got the issue from the logs. I could reproduce the issue on my end.

There seems to be an issue in the software bootloader handling the app verification during the first boot after manually enabled secure boot v2 on ESP32 V3. Can you check if the attached patch solves your problem? It will also be available upstream soon.

The key burnt through `python espefuse.py burn_key_digest secure_boot_signing_key.pem` has the apt order and with the patch, should not be the reason for stopping the booting of the app.

[0001-Secure-Boot-V2-Fix-an-issue-leading-to-manual-enable.txt](https://github.com/espressif/esp-idf/files/5507428/0001-Secure-Boot-V2-Fix-an-issue-leading-to-manual-enable.txt)

closed 2020-12-13 by espressif-bot

08Sahil, 2026-05-23:
Issue resolved after performing this command

espefuse -p /dev/ttyUSB0 burn-efuse ABS_DONE_0