espressif/esp-idf #6051: esp_adc_cal_raw_to_voltage() change proposal, as actual parameters and described parameters are inconsistent.

jlrtrevenga, 2020-11-01:
This is not an error, but just a suggestion: 

According to esp_adc_cal.h parameters description, "esp_adc_cal_raw_to_voltage" was though to have 3 parameters (including  "@param[out]  voltage") + standard error_t as function return code to check valid result. Instead, voltage has been implemented as return code, and no error code can be validated. 

Suggestion is to use return error_code, and return voltage as out parameter, as described.

Extracted from  esp_adc_cal.h:

______________________________________________________________________________________________________________
uint32_t esp_adc_cal_raw_to_voltage(uint32_t adc_reading, const esp_adc_cal_characteristics_t *chars);

/**
 * @brief   Reads an ADC and converts the reading to a voltage in mV
 *
 * This function reads an ADC then converts the raw reading to a voltage in mV
 * based on the characteristics provided. The ADC that is read is also
 * determined by the characteristics.
 *
 * @note    The Characteristics structure must be initialized before this
 *          function is called (call esp_adc_cal_characterize())
 *
 * @param[in]   channel     ADC Channel to read
 * @param[in]   chars       Pointer to initialized ADC characteristics structure
 * @param[out]  voltage     Pointer to store converted voltage
 *
 * @return
 *      - ESP_OK: ADC read and converted to mV
 *      - ESP_ERR_TIMEOUT: Error, timed out attempting to read ADC
 *      - ESP_ERR_INVALID_ARG: Error due to invalid arguments
______________________________________________________________________________________________________________

Alvin1Zhang, 2020-11-02:
Thanks for reporting.

boarchuz, 2020-11-02:
I think you might be looking at the wrong description - the quoted is for the function declaration _after_ esp_adc_cal_raw_to_voltage in the header file (ie. esp_adc_cal_get_voltage).

esp_adc_cal_raw_to_voltage has this:
https://github.com/espressif/esp-idf/blob/6c17e3a64c02eff3a4f726ce4b7248ce11810833/components/esp_adc_cal/include/esp_adc_cal.h#L97-L111

fuAlex, 2020-11-19:
@jlrtrevenga Thank you for your suggestion. This function is used for unit conversion. We will consider adding return values ​​for it in later versions

closed 2023-06-20 by Alvin1Zhang