espressif/esp-idf #6052: Release 4.1V WiFi error on heap_caps_malloc MALLOC_CAP_SPIRAM

filzek, 2020-11-01:
Now the 4.1 release has a new bug, while using memory on SPIRAM, and delete a task, and recreate task it start to have problems.

If just use internal memory there is no problem, but once use SPIRAM memory the problem start to happen.

W (44750) wifi:tx null, bss is null
W (44920) wifi:tx null, bss is null
W (45150) wifi:add mismatch

so, this error just draw if the PSIRAM is used, so, a FIX for the 4.1 sdk is urgently needed.

 MALLOC_CAP_8BIT:
Heap summary for capabilities 0x00000004:
  At 0x3ffb2730 len 15448 free 0 allocated 15080 min_free 0
    largest_free_block 0 alloc_blocks 20 free_blocks 1 total_blocks 21
  At 0x3ffca5ec len 32767 free 200 allocated 32212 min_free 12
    largest_free_block 200 alloc_blocks 19 free_blocks 1 total_blocks 20
  At 0x3f804ad4 len 4175147 free 3630180 allocated 543564 min_free 3622888
    largest_free_block 3629584 alloc_blocks 82 free_blocks 11 total_blocks 93
  At 0x3ffaff10 len 240 free 0 allocated 96 min_free 0
    largest_free_block 0 alloc_blocks 7 free_blocks 0 total_blocks 7
  At 0x3ffb6388 len 7288 free 0 allocated 6988 min_free 0
    largest_free_block 0 alloc_blocks 16 free_blocks 1 total_blocks 17
  At 0x3ffb9a20 len 16648 free 4 allocated 15116 min_free 0
    largest_free_block 0 alloc_blocks 92 free_blocks 3 total_blocks 95
  At 0x3ffca5c8 len 88632 free 0 allocated 86952 min_free 0
    largest_free_block 0 alloc_blocks 103 free_blocks 0 total_blocks 103
  At 0x3ffe0440 len 15072 free 16 allocated 12904 min_free 0
    largest_free_block 4 alloc_blocks 131 free_blocks 3 total_blocks 134
  At 0x3ffe4350 len 113840 free 23508 allocated 89104 min_free 20192
    largest_free_block 23348 alloc_blocks 73 free_blocks 4 total_blocks 77
  Totals:
    free 3653908 allocated 802016 min_free 3643092 largest_free_block 3629584

 MALLOC_CAP_32BIT:
Heap summary for capabilities 0x00000002:
  At 0x3ffb2730 len 15448 free 0 allocated 15080 min_free 0
    largest_free_block 0 alloc_blocks 20 free_blocks 1 total_blocks 21
  At 0x3ffca5ec len 32767 free 200 allocated 32212 min_free 12
    largest_free_block 200 alloc_blocks 19 free_blocks 1 total_blocks 20
  At 0x3f804ad4 len 4175147 free 3630180 allocated 543564 min_free 3622888
    largest_free_block 3629584 alloc_blocks 82 free_blocks 11 total_blocks 93
  At 0x3ffaff10 len 240 free 0 allocated 96 min_free 0
    largest_free_block 0 alloc_blocks 7 free_blocks 0 total_blocks 7
  At 0x3ffb6388 len 7288 free 0 allocated 6988 min_free 0
    largest_free_block 0 alloc_blocks 16 free_blocks 1 total_blocks 17
  At 0x3ffb9a20 len 16648 free 4 allocated 15116 min_free 0
    largest_free_block 0 alloc_blocks 92 free_blocks 3 total_blocks 95
  At 0x3ffca5c8 len 88632 free 0 allocated 86952 min_free 0
    largest_free_block 0 alloc_blocks 103 free_blocks 0 total_blocks 103
  At 0x3ffe0440 len 15072 free 16 allocated 12904 min_free 0
    largest_free_block 4 alloc_blocks 131 free_blocks 3 total_blocks 134
  At 0x3ffe4350 len 113840 free 23508 allocated 89104 min_free 20192
    largest_free_block 23348 alloc_blocks 73 free_blocks 4 total_blocks 77
  At 0x4009d100 len 12032 free 11984 allocated 0 min_free 11964
    largest_free_block 11984 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 3665892 allocated 802016 min_free 3655056 largest_free_block 3629584

 MALLOC_CAP_INTERNAL:
Heap summary for capabilities 0x00000800:
  At 0x3ffb2730 len 15448 free 0 allocated 15080 min_free 0
    largest_free_block 0 alloc_blocks 20 free_blocks 1 total_blocks 21
  At 0x3ffca5ec len 32767 free 200 allocated 32212 min_free 12
    largest_free_block 200 alloc_blocks 19 free_blocks 1 total_blocks 20
  At 0x3ffaff10 len 240 free 0 allocated 96 min_free 0
    largest_free_block 0 alloc_blocks 7 free_blocks 0 total_blocks 7
  At 0x3ffb6388 len 7288 free 0 allocated 6988 min_free 0
    largest_free_block 0 alloc_blocks 16 free_blocks 1 total_blocks 17
  At 0x3ffb9a20 len 16648 free 4 allocated 15116 min_free 0
    largest_free_block 0 alloc_blocks 92 free_blocks 3 total_blocks 95
  At 0x3ffca5c8 len 88632 free 0 allocated 86952 min_free 0
    largest_free_block 0 alloc_blocks 103 free_blocks 0 total_blocks 103
  At 0x3ffe0440 len 15072 free 16 allocated 12904 min_free 0
    largest_free_block 4 alloc_blocks 131 free_blocks 3 total_blocks 134
  At 0x3ffe4350 len 113840 free 23508 allocated 89104 min_free 20192
    largest_free_block 23348 alloc_blocks 73 free_blocks 4 total_blocks 77
  At 0x4009d100 len 12032 free 11984 allocated 0 min_free 11964
    largest_free_block 11984 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 35712 allocated 258452 min_free 32168 largest_free_block 23348

 MALLOC_CAP_8BIT|MALLOC_CAP_INTERNAL:
Heap summary for capabilities 0x00000804:
  At 0x3ffb2730 len 15448 free 0 allocated 15080 min_free 0
    largest_free_block 0 alloc_blocks 20 free_blocks 1 total_blocks 21
  At 0x3ffca5ec len 32767 free 200 allocated 32212 min_free 12
    largest_free_block 200 alloc_blocks 19 free_blocks 1 total_blocks 20
  At 0x3ffaff10 len 240 free 0 allocated 96 min_free 0
    largest_free_block 0 alloc_blocks 7 free_blocks 0 total_blocks 7
  At 0x3ffb6388 len 7288 free 0 allocated 6988 min_free 0
    largest_free_block 0 alloc_blocks 16 free_blocks 1 total_blocks 17
  At 0x3ffb9a20 len 16648 free 4 allocated 15116 min_free 0
    largest_free_block 0 alloc_blocks 92 free_blocks 3 total_blocks 95
  At 0x3ffca5c8 len 88632 free 0 allocated 86952 min_free 0
    largest_free_block 0 alloc_blocks 103 free_blocks 0 total_blocks 103
  At 0x3ffe0440 len 15072 free 16 allocated 12904 min_free 0
    largest_free_block 4 alloc_blocks 131 free_blocks 3 total_blocks 134
  At 0x3ffe4350 len 113840 free 23508 allocated 89104 min_free 20192
    largest_free_block 23348 alloc_blocks 73 free_blocks 4 total_blocks 77
  Totals:
    free 23728 allocated 258452 min_free 20204 largest_free_block 23348

Alvin1Zhang, 2020-11-02:
Thanks for reporting.

YouDONG-ESP, 2020-11-02:
Hi @filzek, I didn't reproduce by creating and deleting task in psram, seems like it happened in some specific scenario, 
could you show me your commit ID and provide a code to reproduce this issue? Thanks.

jack0c, 2020-11-02:
@YouDONG-ESP PTAL

Alvin1Zhang, 2020-12-08:
@filzek Thanks for reporting, will close due to short of feedback, feel free to reopen with more details/updates, thanks.

closed 2020-12-08 by Alvin1Zhang