espressif/esp-idf #6057: I2S interrupt fires too early

hpwit, 2020-11-02:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (DevKitC): v2
- Module or chip used:  ESP32-WROOM-32
- IDF version v4.3-dev-1521-g0289d1cc8
- Build System:      idf.py
- Compiler version (xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [macOS]
- Using an IDE?: VSCODE with Esp-idf plugin
- Power Supply:        USB

## Problem Description

I2S DMA bof interrupt fires before the buffer is actually sent.
I am using I2S in  16 bit output in lcd mode with DMA.
I am using a DMA interrupt to stop the the I2S transmission
What ever the size of the DMA buffer. using either
eof, end or total interrupts trigger, the trigger seems to happen before the DMA buffer is actually fully sent.
I can see it on the oscilloscope. the clock stops before the full DMA buffer is sent.
If I add a delay after the trigger and before the (&I2S0)->conf.tx_start = 0; then I can have my entire buffer sent.

### Expected Behavior
The eof, end, total, DMA interrupts should happen at the moment the DMA buffer is done transmitting not before.

### Actual Behavior
The eof, end, total, DMA interrupts are triggered before the end of the buffer

### Steps to reproduce

1 the signal clock is on pin 17
look at pin 14 for the output signal

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue
https://github.com/hpwit/testI2Sespidf

Alvin1Zhang, 2020-11-03:
Thanks for reporting.

suda-morris, 2022-04-11:
Yes, this is a known issue or mote like a common design in the early ESP chips. That is, for high-speed communication peripheral, we need to start the DMA first, and wait for a while (e.g. 1us), then start the peripheral machine. This can ensure the DMA has sent the data into the peripheral's FIFO before we start the peripheral. See also https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/src/esp_lcd_panel_io_i2s.c#L499

Sorry for the soooo late response.

closed 2022-04-11 by suda-morris

hpwit, 2022-04-11:
Thank you for the feedback