espressif/esp-idf #6059: Mutual TLS for ESP Websocket Client library

wootsies, 2020-11-02:
I was trying to perform mutual authentication using the ESP IDF Websocket Client Library when I discovered that there was no way to set the client certificate and in the esp_websocket_client_config_t struct. This made me realise that the the web socket library does not actually support client authentication or mutual TLS.

Solution:
To add support for mutual TLS on websockets, similar to what has already been done in the ESP_TLS library where you simply need to set the client cert and key in order to support mutual TLS in the code.

Alvin1Zhang, 2020-11-03:
Thanks for raising this feature request.

wootsies, 2020-11-03:
This is linked to the questions raised by someone else here https://esp32.com/viewtopic.php?f=2&t=16194&p=67149#p67149

closed 2021-01-15 by espressif-bot