espressif/esp-idf #6060: Failing to install esp-idf-tools-setup-2.3.exe in win10

MicroGrit, 2020-11-02:
## Environment
- Development Kit:      [esp-idf-tools-setup-2.3.exe]
- Module or chip used:  [ESP32]
- IDF version: v4.1
- Build System:         [idf.py]
- Operating System:     [Windows10]
- Python  Environment: Anaconda3(python 3.7)

## Problem Description
Two computers installed esp32 development environment error by esp-idf-tools-setup-2.3.exe , and all prompted problems about virtualenv module. I am not familiar with this module and cannot  solve.
Here attach the setup log report, please help, thank you!

## Logs
```
2020-11-01 10:37:52.646   --------
2020-11-01 10:37:52.649   Installing Python environment:D:\ProgramData\Anaconda3\python.exe "E:\esp-idf\tools\idf_tools.py" --idf-path E:\esp-idf install-python-env
2020-11-01 10:39:40.876   Done, exit code=1
2020-11-01 10:39:40.876   --------
2020-11-01 10:39:40.876   Running command: D:\ProgramData\Anaconda3\python.exe "E:\esp-idf\tools\idf_tools.py" --idf-path E:\esp-idf install-python-env
                          Creating a new Python environment in C:\Users\KD\.espressif\python_env\idf4.1_py3.7_env
                          Installing virtualenv
                          Collecting virtualenv
                            Downloading https://files.pythonhosted.org/packages/97/f3/c064343ac58d1a54c393a3f66483a29500f644a5918deeb935d28673edd9/virtualenv-20.1.0-py2.py3-none-any.whl (4.9MB)
                          Requirement already satisfied: filelock<4,>=3.0.0 in d:\programdata\anaconda3\lib\site-packages (from virtualenv) (3.0.10)
                          Collecting importlib-metadata<3,>=0.12; python_version < "3.8" (from virtualenv)
                            Downloading https://files.pythonhosted.org/packages/6d/6d/f4bb28424bc677bce1210bc19f69a43efe823e294325606ead595211f93e/importlib_metadata-2.0.0-py2.py3-none-any.whl
                          Requirement already satisfied: six<2,>=1.9.0 in d:\programdata\anaconda3\lib\site-packages (from virtualenv) (1.12.0)
                          Collecting distlib<1,>=0.3.1 (from virtualenv)
                            Downloading https://files.pythonhosted.org/packages/f5/0a/490fa011d699bb5a5f3a0cf57de82237f52a6db9d40f33c53b2736c9a1f9/distlib-0.3.1-py2.py3-none-any.whl (335kB)
                          Collecting appdirs<2,>=1.4.3 (from virtualenv)
                            Downloading https://files.pythonhosted.org/packages/3b/00/2344469e2084fb287c2e0b57b72910309874c3245463acd6cf5e3db69324/appdirs-1.4.4-py2.py3-none-any.whl
                          Collecting zipp>=0.5 (from importlib-metadata<3,>=0.12; python_version < "3.8"->virtualenv)
                            Downloading https://files.pythonhosted.org/packages/41/ad/6a4f1a124b325618a7fb758b885b68ff7b058eec47d9220a12ab38d90b1f/zipp-3.4.0-py3-none-any.whl
                          Installing collected packages: zipp, importlib-metadata, distlib, appdirs, virtualenv
                            The script virtualenv.exe is installed in 'C:\Users\KD\AppData\Roaming\Python\Python37\Scripts' which is not on PATH.
                            Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
                          Successfully installed appdirs-1.4.4 distlib-0.3.1 importlib-metadata-2.0.0 virtualenv-20.1.0 zipp-3.4.0
                          FileNotFoundError: [Errno 2] No such file or directory: 'D:\\ProgramData\\Anaconda3\\Lib\\venv\\scripts\\nt\\python.exe'
                          Traceback (most recent call last):
                            File "E:\esp-idf\tools\idf_tools.py", line 1387, in <module>
                              main(sys.argv[1:])
                            File "E:\esp-idf\tools\idf_tools.py", line 1383, in main
                              action_func(args)
                            File "E:\esp-idf\tools\idf_tools.py", line 1189, in action_install_python_env
                              stdout=sys.stdout, stderr=sys.stderr)
                            File "D:\ProgramData\Anaconda3\lib\subprocess.py", line 341, in check_call
                              raise CalledProcessError(retcode, cmd)
                          subprocess.CalledProcessError: Command '['D:\\ProgramData\\Anaconda3\\python.exe', '-m', 'virtualenv', 'C:\\Users\\KD\\.espressif\\python_env\\idf4.1_py3.7_env']' returned non-zero exit status 1.
```

## Details
[Setup Log 2020-11-01 #001.txt](https://github.com/espressif/esp-idf/files/5475582/Setup.Log.2020-11-01.001.txt)

igrr, 2020-11-02:
Could you please check if virtualenv works in your Python installation, e.g. by running the following from the command line?

```
D:\ProgramData\Anaconda3\python.exe -m virtualenv tmp_dir
```
and see if that finishes successfully?

I suspect that this part of the error message points at the issue:
```
The script virtualenv.exe is installed in 'C:\Users\KD\AppData\Roaming\Python\Python37\Scripts' which is not on PATH.
```

I'm not familiar with Anaconda but it looks like the "user" location is not added to the path by the interpreter, so venv module is getting confused and looks in the wrong place. Generally Anaconda docs used to not recommend installing packages with `pip`, and suggested using `conda` as the package manager instead. However ESP-IDF relies on pip for package installation.

Probably the easiest option is to run the installer again, choosing the option to "Install Python 3.7" when you are prompted to choose the python version to use. This will install a copy of Python from python.org, for which both pip and virtualenv are known to work.

MicroGrit, 2020-11-03:
Thanks a lot.

When my PC ran the command you recommended,
```
python.exe -m virtualenv tmp_dir
```
it reported
```
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\**username**\\anaconda3\\Lib\\venv\\scripts\\nt\\python.exe'
```

As you said, choosing this option to "Install Python 3.7" solves my problem very well. We hope `Espressif` can increase its support for Anaconda environment.

Alvin1Zhang, 2021-08-12:
Thanks for reporting, fix is available https://github.com/espressif/idf-installer/commit/9e023cbb6fd0fda3a546a280f12fc90621312522, feel free to reopen.

closed 2021-08-12 by Alvin1Zhang