espressif/esp-idf #6061: strange link error in Windows: xtensa-esp32-elf-g++.exe: error: CreateProcess: No such file or directory

versaloon, 2020-11-02:
##  Environment

- Development Kit:      [none, using qemu, no hardware]
- Kit version (for WroverKit/PicoKit/DevKitC): [none]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it):
    v4.3-dev-1561-g357a27760
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt&Plain Command Prompt].
- Using an IDE?: [No]
- Power Supply:         [None]

## Problem Description

Compiles OK under Ubuntu(wsl2), but failed on windows command line.
below is the error log:
[1284/1288] Linking CXX executable hello-world.elf
FAILED: hello-world.elf
cmd.exe /C "cd . && C:\Users\SimonQian\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\hello-world.elf.rsp  -o hello-world.elf  && cd ."
xtensa-esp32-elf-g++.exe: error: CreateProcess: No such file or directory

if I remove some unused .o files in CMakeFiles\hello-world.elf.rsp, link will pass.

### Expected Behavior

Link OK.

### Actual Behavior

Link failed

### Steps to reproduce

Preparing...
will append when ready

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

versaloon, 2020-11-02:
[esp_vsf_demo.zip](https://github.com/espressif/esp-idf/files/5477195/esp_vsf_demo.zip)
This is the esp_vsf_demo I mentioned.

versaloon, 2020-11-04:
I switched to wsl2 Ubuntu, everything runs OK(and much more faster) now.

georgik, 2021-05-07:
Hi @versaloon . Thank you for reporting the issue and we're sorry to responding so late to the issue with ESP-IDF 4.4 (master).

I was trying to replicate the behavior on Windows and Ubuntu with provided file, but it failed with error like:
```
CMake Error at C:/Users/georgik/Desktop/esp-idf/tools/cmake/component.cmake:305 (message):
  Include directory 'C:/projects/tmp/vsf/source/vsf' is not a directory.
Call Stack (most recent call first):
  C:/Users/georgik/Desktop/esp-idf/tools/cmake/component.cmake:485 (__component_add_include_dirs)
  C:/projects/tmp/vsf/CMakeLists.txt:1 (idf_component_register)
```

vsf source code in latest commit does not contain `source/vsf`.

It seems that project https://github.com/vsfteam/vsf moved quite a lot. Is the esp_vsf_demo source code still relevant?

versaloon, 2021-05-18:
yes, there are changes in vsf, and esp32 project is not updated.
esp32 support in vsf is not maintained now, so just ignore the issue.

closed 2021-05-18 by versaloon