espressif/esp-idf #6062: section placement error while use __attribute__((section("xxxx"))) to decorate functions

versaloon, 2020-11-02:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it):
    v4.3-dev-1561-g357a27760
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [No]
- Power Supply:         [none]

## Problem Description

while using __attribute__((section("xxxx"))) to decorate functions, section placement will fail with log like:
A fatal error occurred: Segment loaded at 0x400e379c lands in same 64KB flash mapping as segment loaded at 0x400e3788. Can't generate binary. Suggest changing linker script or ELF to merge sections.
ninja: build stopped: subcommand failed.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

### Code to reproduce this issue

```c
#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"

__attribute__((section("text.test_print_info")))
void test_print_info(void)
{
    esp_chip_info_t chip_info;
    esp_chip_info(&chip_info);
    printf("This is %s chip with %d CPU cores, WiFi%s%s, ",
            CONFIG_IDF_TARGET,
            chip_info.cores,
            (chip_info.features & CHIP_FEATURE_BT) ? "/BT" : "",
            (chip_info.features & CHIP_FEATURE_BLE) ? "/BLE" : "");
}

__attribute__((section("text.test_restart")))
void test_restart(void)
{
    fflush(stdout);
    esp_restart();
}

void app_main(void)
{
    test_print_info();
    test_restart();
}
```

## Debug Logs

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

igrr, 2020-11-02:
@versaloon The default project linker script only catches `.text` and `.text.*`, so the section `text.test_print_info` is probably not getting placed into the .flash_text section as it should be. Can you try changing the section name to `.text.test_print_info`?

versaloon, 2020-11-03:
> @versaloon The default project linker script only catches `.text` and `.text.*`, so the section `text.test_print_info` is probably not getting placed into the .flash_text section as it should be. Can you try changing the section name to `.text.test_print_info`?

The section names are `.text.test_print_info` and "text.test_restart", and the according to the section list, the address is 0x400exxxx, is the address in flash? I'm new to esp32.

versaloon, 2020-11-03:
> @versaloon The default project linker script only catches `.text` and `.text.*`, so the section `text.test_print_info` is probably not getting placed into the .flash_text section as it should be. Can you try changing the section name to `.text.test_print_info`?

Thanks, it works.

igrr, 2020-11-03:
There is a "Close issue" button just right below the comment field. I'll close this one for you.

closed 2020-11-03 by igrr