espressif/esp-idf #6064: Guru Meditation Error: Core 0 panic'ed (Interrupt wdt timeout on CPU0) in timer_overflow_happened()

maqian-cn, 2020-11-03:
## Environment

- Development Kit:      [none]
- Kit version : [none]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version :
    // v3.3.1
- Build System:         [Make]
- Compiler version :
    // xtensa-esp32-elf-gcc.exe (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System:     [Windows]
- (Windows only) environment type: [MSYS2 mingw32].
- Using an IDE?: [No]
- Power Supply:         [external 5V]

## Problem Description

The device reboots with error message.

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40081d71  PS      : 0x00060434  A0      : 0x80081ecb  A1      : 0x3ffaf9d0
A2      : 0x00000000  A3      : 0xf261ca0a  A4      : 0xf261c96a  A5      : 0x40111034
A6      : 0x3ffc7fd8  A7      : 0x025a873e  A8      : 0x00000081  A9      : 0x00000001
A10     : 0x00000001  A11     : 0x00060423  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x0000cdcd  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

ELF file SHA256: 941b393f9b7f704b0367cbaff485911901865f3e0ecb36a2ee0c530baa3680ed

Backtrace: 0x40081d71:0x3ffaf9d0 0x40081ec8:0x3ffaf9f0 0x400d458d:0x3ffafa10

Core 1 register dump:
PC      : 0x401961de  PS      : 0x00060334  A0      : 0x800d4e3e  A1      : 0x3ffc4400
A2      : 0x00000000  A3      : 0x3ffb3ab8  A4      : 0x8008e6c0  A5      : 0x3ffcfb60
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d40fa  A9      : 0x3ffc43d0
A10     : 0x00000000  A11     : 0x3ffb003c  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x0000abab  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

ELF file SHA256: 941b393f9b7f704b0367cbaff485911901865f3e0ecb36a2ee0c530baa3680ed

Backtrace: 0x401961de:0x3ffc4400 0x400d4e3b:0x3ffc4420 0x4008db9d:0x3ffc4440

Rebooting...

$ xtensa-esp32-elf-addr2line -pfiaC -e build/M8321-ESP32.elf 0x40081d71:0x3ffaf9d0 0x40081ec8:0x3ffaf9f0 0x400d458d:0x3ffafa10
0x40081d71: timer_overflow_happened at D:/esp-idf/esp-idf/components/esp32/esp_timer_esp32.c:324
0x40081ec8: esp_timer_impl_set_alarm at D:/esp-idf/esp-idf/components/esp32/esp_timer_esp32.c:324
0x400d458d: timer_process_alarm at D:/esp-idf/esp-idf/components/esp32/esp_timer.c:316
 (inlined by) timer_task at D:/esp-idf/esp-idf/components/esp32/esp_timer.c:326

$ xtensa-esp32-elf-addr2line -pfiaC -e build/M8321-ESP32.elf 0x401961de:0x3ffc4400 0x400d4e3b:0x3ffc4420 0x4008db9d:0x3ffc4440
0x401961de: esp_task_wdt_isr_user_handler at D:/esp-idf/esp-idf/components/esp32/task_wdt.c:128
0x400d4e3b: esp_vApplicationIdleHook at D:/esp-idf/esp-idf/components/esp32/freertos_hooks.c:86
0x4008db9d: prvIdleTask at D:/esp-idf/esp-idf/components/freertos/tasks.c:4409

### Expected Behavior
The device should work normally and should not be reset itself.

### Actual Behavior
it is observed that "Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)" happened twice time.

> 1.Run continuously for 8 days and reset

> 2.Run continuously for 11 hours and reset

### Steps to reproduce

1. power off
2. power on
3.do nothing and leave it alone

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/5480092/sdkconfig.txt)

[M8321-ESP32.zip](https://github.com/espressif/esp-idf/files/5480095/M8321-ESP32.zip)
### Code to reproduce this issue

## Debug Logs

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

Alvin1Zhang, 2020-11-04:
Thanks for reporting.

igrr, 2020-11-04:
@maqian-cn this issue has been originally reported in https://github.com/espressif/esp-idf/issues/2954, and fixed in ESP-IDF release v3.3.2 (https://github.com/espressif/esp-idf/releases/tag/v3.3.2).

Latest v3.3.x bugfix release at the moment is v3.3.4: https://github.com/espressif/esp-idf/releases/tag/v3.3.4.

maqian-cn, 2020-11-04:
@igrr @Alvin1Zhang Thanks for your reply and supporting. I'll check it.

closed 2021-08-12 by Alvin1Zhang