espressif/esp-idf #6065: Crash on esp_http_client_read() If Network Connection Lost

gecko242, 2020-11-03:
## Environment

ESP32-WROOM-32U, Custom PCB.
v4.3-dev-472-gcf056a7d0
idf.py build
Compiler Version 8.2.0
Windows, CMD

[image]

```cpp

esp_http_client_set_method(client, HTTP_METHOD_GET); // get blacklist.txt
esp_http_client_set_url(client, OTA_BLACKLIST_URL);
err = esp_http_client_perform(client);
size_t blacklistLen = esp_http_client_get_content_length(client);

if(blacklistLen > BLACKLIST_MAX_LEN)
{
    esp_http_client_cleanup(client);
    vTaskSuspend(xotaTaskHandle);
 }
 blacklist = malloc(blacklistLen + 1);
 if(blacklist != NULL)
 {
     int16_t readData = 0;
     size_t downloadedBytes = 0;
     do
     {
         readData = esp_http_client_read(client, (blacklist + downloadedBytes), CHUNK_SIZE);
         downloadedBytes += readData;
         statusCode = esp_http_client_get_status_code(client);
     }while((readData > 0) && (statusCode == 200) && (downloadedBytes < blacklistLen)); 

```

Alvin1Zhang, 2020-11-04:
Thanks for reporting.

shubhamkulkarni97, 2020-11-09:
Hi @gecko242,

From the provided logs and code snippet, I see the connection was not opened after esp_http_client_perform() was called. After esp_http_client_perform, `esp_http_client_read()` is called without any error check, which can lead to issues.

However, I would like to point out that this is not a correct way to perform http_request with esp_http_client. You should either use `esp_http_client_perform()` or `esp_http_client_read()` to perform a request.

* When using `esp_http_client_perform()`, you have to register event_handler and you will get the response of the request in event_handler itself. Please check http_rest_with_url example: https://github.com/espressif/esp-idf/blob/494a124d961c4c755685b22fe986c31826d6f503/examples/protocols/esp_http_client/main/esp_http_client_example.c#L111

* When using `esp_http_client_read()` approach, `esp_http_client_open()`should be used to open a new connection. In this approach `esp_http_client_perform()` **should not be used**. Please check http_native_request example: https://github.com/espressif/esp-idf/blob/494a124d961c4c755685b22fe986c31826d6f503/examples/protocols/esp_http_client/main/esp_http_client_example.c#L573

Can you try changing the URL in one of the esp_http_client example with `OTA_BLACKLIST_URL` and check if you can reproduce the issue with it?

Thanks,
Shubham

gecko242, 2020-11-09:
Hi @shubhamkulkarni97 

Thanks for pointing that out, clearly I had misunderstood the HTTP client component somewhere along the way! 

I have updated the code to follow the examples more closely, and I can see why what I was doing previously might cause an error.

That being said, I only ever experienced this due to another bug (TFS Timer Issue, #5915), which has now been fixed, so I am unable to run another test on the esp_http_client example. 

I will close, assuming this was down to dodgy code. 

Thanks again :)

closed 2020-11-09 by gecko242