espressif/esp-idf #6067: ESP-netif crashing all the time (broken), when creating an instance in the esp-mesh "ip_internal_network " example

ramidomalys, 2020-11-03:
## Environment  

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): |v4]
- Module or chip used:  [ESP32-WROVER-E]
- IDF version  v4.2-beta1-227-gf0e87c933
- Build System:         [idf.py]
- Compiler version ( xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0) 

- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt|].
- Using an IDE?: [No|]
- Power Supply:         [USB|external 5V]

## Problem Description

by building and flashing the  esp-mesh  "ip_internal_network " example ,
right after running menuconfig, setting my router settings, i invoke build and flash, it crashes after connecting to parent node ( router,) exactly after obtaining ip adress and trying to create the Mesh netif  at line 
mesh_netif.c :248 :esp_netif_action_start(netif_ap, NULL, 0, NULL);

### Expected Behavior
continueing esp_netif_action_start , without crashing ...

### Actual Behavior
crashing at line 
```
mesh_netif.c :248 :esp_netif_action_start(netif_ap, NULL, 0, NULL);
```
last log before Guru Meditation Error : 
```
D (12657) esp_netif_lwip: esp_netif_start_api Setting the lwip netif0x3ffbac80 UP
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
```
if we activate the example option ( CONFIG_MESH_USE_GLOBAL_DNS_IP ) the crash occur before the ip obtaining event , otherwise, the crash occur after obtaining ip event , seems unrelated but anyhow. 

### Steps to reproduce
1. step1 clean install esp-idf, release/v4.2 , or master branch (both same behavior)

2. configure ip_internal_network  example using menuconfig 
3. build and flash the example 

### Code to reproduce this issue

```
ip_internal_network example code provided by idf
```

## Debug Logs
```
I (2977) mesh_main: <MESH_EVENT_MESH_STARTED>ID:77:77:77:77:77:76
I (2987) mesh_main: mesh starts successfully, heap:153396, root not fixed

D (6867) event: running post WIFI_EVENT:1 with handler 0x40119fa0 and context 0x3ffcf7d4 on loop 0x3ffc016c
0x40119fa0: mesh_wifi_event_cb at ??:?

I (6867) mesh: [S3]SFR_3DD0_2GEXT, 10:da:43:c1:9f:b2, channel:3, rssi:-66
I (6877) mesh: find router:[ssid_len:14]SFR_3DD0_2GEXT, rssi:-66, 10:da:43:c1:9f:b2(encrypted), new channel:3, old channel:0
I (6887) mesh: [FIND][ch:0]AP:28, otherID:0, MAP:0, idle:0, candidate:0, root:0[10:da:43:c1:9f:b2]router found
I (6897) mesh: [FIND:1]find a network, channel:3, cfg<channel:0, router:SFR_3DD0_2GEXT, 00:00:00:00:00:00>

D (6907) event: running post MESH_EVENT:22 with handler 0x400d5000 and context 0x3ffd028c on loop 0x3ffc016c
0x400d5000: mesh_event_handler at c:\users\rami_domalys\desktop\ip_internal_network\build/../main/mesh_main.c:204

V (6907) esp_adapter: thread sem get: sem=0x3ffd03a0
I (6917) mesh_main: <MESH_EVENT_FIND_NETWORK>new channel:3, router BSSID:00:00:00:00:00:00
I (6917) wifi:mode : sta (24:0a:c4:02:02:e8) + softAP (24:0a:c4:02:02:e9)
D (6937) nvs: nvs_set opmode 1 3
D (6937) nvs: nvs_set ap.chan 1 1
D (6947) nvs: nvs_set ap.sndchan 1 1
W (6947) wifi:<MESH AP>adjust channel:1, secondary channel offset:1(40U)
[... 156 lines cut ...]
Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060a30  A0      : 0x800f7106  A1      : 0x3ffbfd00
A2      : 0x3ffbaa04  A3      : 0x3ffbbb76  A4      : 0x0000002a  A5      : 0x3ffbbb5c
A6      : 0x3ffbf0ec  A7      : 0x00000000  A8      : 0x801776b0  A9      : 0x0000002a
A10     : 0x3ffbadd0  A11     : 0x3ffbbb76  A12     : 0x0000002a  A13     : 0x3ffbbb5c
A14     : 0x3f407c34  A15     : 0x3f408a34  SAR     : 0x00000010  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace:0xfffffffd:0x3ffbfd00 |<-CORRUPTED

ELF file SHA256: 0469695f3ab1c02d

Rebooting...
ets Jun  8 2016 00:22:57

```

## Additional info 
- [x ] sdkconfig file

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/5481967/sdkconfig.zip)

- [full log ]

[full log.txt](https://github.com/espressif/esp-idf/files/5482273/full.log.txt)

Alvin1Zhang, 2020-11-04:
Thanks for reporting.

david-cermak, 2020-11-04:
Hi @ramidomalys 

Could you please check if after applying the patch below the issue persists? We've introduced a Tx optimization in WiFi/esp-netif layer recently, transmiting buffers by reference, but some examples haven't been updated, I'm afraid.
[0001-examples-esp_mesh-Fix-internal-ip-mesh-example-to-us.patch.txt](https://github.com/espressif/esp-idf/files/5486186/0001-examples-esp_mesh-Fix-internal-ip-mesh-example-to-us.patch.txt)

ramidomalys, 2020-11-04:
Hello, thank you so much for this fast reply , indeed the patch corrected the bug i can confirm it run smoothly.
would this patch be pushed to the release/v4.2 branch ? 
thank you.

david-cermak, 2020-11-05:
> would this patch be pushed to the release/v4.2 branch ?

Yes, we have to merge it to `master` first and then backport to the `release/v4.2`. Please keep this open, the issue will close automatically when fixed on master.

closed 2020-11-19 by projectgus