espressif/esp-idf #6069: Issue: ESP32-S2 ULP RISC-V no evidence of running

timpur, 2020-11-04:
## Environment

Running Fresh esp_idf repo on master branch install as off today (2020/11/04)

- Development Kit: FeatherS2
- Module or chip used: ESP32-S2
- IDF version: v4.3-dev-1561-g357a27760
- Build System: idf.py
- Compiler version: 8.4.0
- Operating System: Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: Yes (vscode + idf plugin)
- Power Supply: USB

## Problem Description

Ive trued using the ulp risc example provided, i count get the expected output similar to (https://www.esp32.com/viewtopic.php?f=13&t=16274&p=66493&hilit=risc#p66518)

Ive also tried some basic c code to add 2+2 and store into a exported variable that the main cpu can print out ... i cant get this to work either.

### Code to reproduce this issue

#### Main CPU

```cpp
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp32s2/ulp_riscv.h"
#include "ulp_main.h"

extern const uint8_t bin_start[] asm("_binary_ulp_main_bin_start");
extern const uint8_t bin_end[] asm("_binary_ulp_main_bin_end");

void app_main(void)
{
    printf("Running ULP Program\n");
    ESP_ERROR_CHECK(ulp_riscv_load_binary(bin_start, (bin_end - bin_start)));
    ESP_ERROR_CHECK(ulp_riscv_run());
    vTaskDelay(1000 / portTICK_PERIOD_MS);
    printf("out1: %d\n", ulp_out1);
}
```

#### ULP CPU

```cpp
#include <stdio.h>
#include <stdint.h>
#include <stdbool.h>

volatile uint32_t out1 = 1;

int main(void)
{
    out1 = 2 + 2;
    return 0;
}
```

## Debug Logs

```
I (135) boot: Loaded app from partition at offset 0x10000
I (136) boot: Disabling RNG early entropy source...
I (204) spi_flash: detected chip: generic
I (205) spi_flash: flash io: dio
I (205) cpu_start: Starting scheduler on PRO CPU.
Running ULP Program
out1: 1
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

felmue, 2020-11-07:
Hi guys

I get the same results on a ESP32-S2-Kaluga-1 board. It seems like the ULP doesn't start at all or doesn't have a clock. Even the simplest code, e.g. immediately waking up the main CPU doesn't do anything.

Like @timpur I don't think the ULP is running at all. What are we missing?

Thanks
Felix

timpur, 2020-11-08:
Update: I managed to get my sample working, only on a fresh reboot, seems any software restart doesnt restart the ulp correctly ... still digging though the code and registers ....

felmue, 2020-11-09:
Hi guys

For a test I've modified the main CPU code to init and start ULP code (as before) and then to stay in a loop periodically checking `ulp_gpio_level_previous` variable and not going into deep sleep.

In the ULP code I've added `volatile` in front of the `gpio_level_previous` variable declaration and commented out the `break` statement in the while loop to keep the ULP running all the time.

With those modifications I get the GPIO0 changes, read by ULP and checked by the main code. This seems to indicate that the ULP actually is running.

However as soon as I put the main CPU into deep sleep (or light sleep for that matter) it is never woken up by the ULP. If that's because the ULP stops running as soon as the main CPU is in sleep mode or if the ULP fails to wake up the main CPU I don't know.

Thanks
Felix

boarchuz, 2020-11-09:
@felmue 
Perhaps have the ULP toggle a pin so you have a heartbeat in deep sleep?
This snippet might be useful for debugging ULP state (untested on S2 but should be about the same):
https://github.com/boarchuz/HULP/blob/581a9f84b3e18d675b3aa8788f48935c7919c6a2/src/hulp.cpp#L350-L374

@timpur 
A software restart will not restart the RTC domain, so yes ULP can still be running / in an undefined state. Consider a RTC_WDT if you want to ensure a 'ULP-safe' restart. eg.
https://github.com/espressif/esp-idf/blob/0b71a0a46d67cce681ec55973b020d950d8596bd/components/bootloader_support/src/bootloader_init.c#L76-L83
For a more complex solution, on start you could check the ULP state using the above snippet -> if not in an idle state, disable ULP wakeup triggers -> wait a moment and/or poll ULP state -> if still not idle then RTC WDT reset.
I find it useful to reserve the first word of RTC slow memory as a flag for ULP to check every loop if it ought to be stopped.

felmue, 2020-11-09:
Hello @boarchuz 

thank you very much for that helpful information and the tip to toggle a pin as ULP heartbeat indicator. Here is what I found so far:
- ULP continues to run during deep sleep as the heartbeat pin is toggling
- main CPU wake up by timer works for me
- main CPU wake up by ULP (e.g. `ulp_riscv_wakeup_main_processor()`) doesn't work for me - main CPU is not woken up

Thanks
Felix

Here is the setup for timer wake up every 5 seconds (which works for me):
```
esp_sleep_enable_timer_wakeup(5000000);
esp_deep_sleep_start();
```

Here is the setup for ULP wake up (which doesn't work for me):
```
esp_sleep_enable_ulp_wakeup();
esp_deep_sleep_start();
```

boarchuz, 2020-11-10:
@felmue 
Is your ULP program an infinite loop, or are you setting wakeup period and running periodically?

I ask because I had a play yesterday and wasn't even able to get the ULP wakeup timer to start the RISCV (oddly, it would sometimes work *once*). By monitoring registers I could see that the ULP timer was enabled, expiring, and should be starting the ULP but nothing would happen. Registers for RISCV selection and timer wakeup target seemed correct. Odd.

Also confirmed that timer wakeup works but not ulp_riscv_wakeup_main_processor (looping) here also. There is a test case for this but I haven't tried yet.

felmue, 2020-11-10:
Hello @boarchuz 

yes, for now the ULP program is an infinite loop which toggles the output (about every second) and calls `ulp_riscv_wakeup_main_processor()` every 20 toggles.

If I don't use an infinite loop, the ULP program runs only once after loading and running it, but never gets restarted independent of the value given when setting the wakeup period. I don't think the wakeup period is taken into account at all - for me the RISCV immediately starts to run the first time independent of the actual wakeup period value set.

Re ULP wakeup: I found this lines in `esp-idf/components/esp_system/sleep_modes.c`:
```
#ifdef CONFIG_IDF_TARGET_ESP32
    // Enable ULP wakeup
    if (s_config.wakeup_triggers & RTC_ULP_TRIG_EN) {
        rtc_hal_ulp_wakeup_enable();
    }
#endif
```
but changing that to be executed for the ESP32S2 variant as well did not help. I suspect more stuff is missing to make ULP work correctly for the S2 variant.

Thanks
Felix

uLipe, 2020-11-10:
Hi folks, thanks for reporting that, we are aware of the problem, from the point of support to the master to the last recent commits, the ULP-RISC-V suffered from a regression. The fix is already in internal review and should be merged soon.

felmue, 2020-11-10:
Hello @uLipe 

thank you very much for letting us know. Much appreciated. I am looking forward to trying the fix.

Cheers
Felix

timpur, 2020-11-11:
Seems everyone else did my work :P (which is good). Keen to try the fix asap.

felmue, 2020-11-19:
Hi guys

some improvement with the latest IDF (ESP-IDF v4.3-dev-1901-g178b122c1-dirty) from yesterday:

- running the `ulp_riskcv` sample it now reacts upon GPIO0 changes, but only once after a restart
- if I remove the `break` in the ulp code so that it stays in the while(1) loop forever, GPIO0 changes are detected multiple times

So it looks to me like the ULP code doesn't restart after being shutdown.

Thanks
Felix

projectgus, 2020-12-22:
Hi all,

Sorry for the long delay between updates. This fix has been merged in bd2d70ca0b3e7f98965da19a3dfa616316354c7a so the ULP restart should be working now. Please let us know if you're still seeing any issues.

boarchuz, 2020-12-22:
@projectgus 
The solution posted on the forums here by user scottsh is still required:
https://esp32.com/viewtopic.php?f=13&t=18547#p68956

ie. COCPU_DONE_FORCE bit must be set

He said he'd submit a PR but I don't see it yet. Also I think it would be best to do this in ulp_riscv_run as it only needs to be set once.

projectgus, 2020-12-22:
@boarchuz Thanks, I had missed this forum post

andymule, 2021-01-25:
As per the forum post, the workaround is to add these lines to your app's main() c init:
```
REG_SET_FIELD(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_SHUT_2_CLK_DIS, 0x3F);
SET_PERI_REG_MASK(RTC_CNTL_COCPU_CTRL_REG, RTC_CNTL_COCPU_DONE_FORCE);
```

closed 2021-02-01 by igrr

timpur, 2021-02-26:
@igrr Sorry to do this, but ive looked everwhere and cant find anything explaing ULP RISCV registers
`#define RISCV_REG_CONV(addr)        (((addr&0xffff)<<3 & 0xe000) | (addr & 0x1fff) | 0x8000)`

This seems to be the result of the translation....
```
RTCCNTL = 0x8000
RTCIO = 0xa400
SENS = 0xc800
RTC_I2C = 0xec00
```

For RTCCNTL
PeriBus1 base = 0x3F400000
DR_REG_RTCCNTL_BASE = 0x3f408000
thus ulp_addr = 0x8000, which makes sense to me but RTCIO doesnt

For RTCIO 
PeriBus1 base = 0x3F400000
DR_REG_RTCIO_BASE= 0x3f408400
thus should ulp_addr = 0x8400 but it doenst its 0xa400 ???

I cant find anything explaing why in the Tech Docs .... Please some one help me

Im just trying to fine add the addresses of these:

timpur, 2021-03-02:
@projectgus is there a better place for my question sorry.

projectgus, 2021-03-03:
Hi @timpur,

Probably the best place for this would be the https://esp32.com forums .  I apologise that there's no better documentation on the memory map for the RISCV ULP, and also that examples for the other RISCV ULP peripherals are still not available. Will ask the docs update team to update the peripherals section in a future ESP32-S2 TRM update.

The mapping you have should work for the peripherals accessible from RISCV ULP, even though the addresses may not immediately make sense and they don't follow in the same order as on the main bus.

Peripheral(s) | Base register  | Main bus address | RISCV base |
| ------- | ----------- | ------------------------- | ------------------  |
| RTC Control | DR_REG_RTCCNTL_BASE |  0x3f408000 | 0x8000 |
| RTC GPIO | DR_REG_RTC_IO_BASE | 0x3f408400 | 0xa400 |
| RTC I2C  | DR_REG_RTC_I2C_BASE | 0x3f408c00 | 0xec00 |
| ADC, DAC, Touch, Temp  | DR_REG_SENS_BASE | 0x3f408800 | 0xc800 |

It should be possible to operate the SAR ADC entirely via the "SENS" registers (sens_reg.h, sens_struct.h) however I'm sorry we don't have an example or explanation of the register operations needed for this yet.

The intention behind the RISCV_REG_CONV() macro is that you shouldn't have to use these addresses directly, instead you can use REG_READ/REG_WRITE/etc with the same soc.h headers that you would include in an application to run on the main CPU, and the compiler will convert the base addresses for you. However if you're writing your own code then obviously you can write it however you prefer!

If something specifically isn't working, please post on the forum with some more details (or open a new issue here if it looks like a bug.)

timpur, 2021-03-03:
@projectgus Thank you for answering my question. I understand that it take time for detailed docs for a complex chit to be documented in a user friendly way. No where in docs does it mention this kind of mapping of RISCV ULP bus to main bus. 

Eg from my reading and understanding I think ULP bus, 0x0 -> 0x2000 (is ram/data) (RTC_SLOW_MEM) and we know 0x8000 ->0xF000 is for RTC REGS, but there is a lot left in the ULP bus... are there things it can do that just are not documented ? (or i have this all wrong (im not a low level HW person ...)

Last thing im working on writing C API's for the ULP to interact with PERFs that as 1:1 as possible with the Main CPU API's. This this best discussed in a new issue and is that something that could be contributed back into IDF ? (I have more questions about ULP components etc if this is something IDF wants?)