espressif/esp-idf #6070: Bug in UTC -> localtime conversion after 2038

udeboer, 2020-11-04:
This is a bug in newlib. With 64bit time_t support compiled into the toolchain (v4.2). 

UTC time to local timezone conversion does not work for years after 2038. Daylight saving in the timezone was not recognized. This is in the functions tzset and localtime of the newlib time functions. 

The struct tm is filled for years after 2038. So year, day etc are all filled with the correct values. But the tm.tm_isdst is 0 and no daylight saving is applied. Unix timestamps before 2038 work. 

It was caused by the following calculation in time/tzcalc_limits.c at line 68

       /* store the change-over time in GMT form by adding offset */
       tz->__tzrule[i].change = days * SECSPERDAY +
       tz->__tzrule[i].s + tz->__tzrule[i].offset;

Here tz->__tzrule[i].change is a time_t (64bit). The compiler does the calculation in 32 bit. All variables used in the calculation are 32bit.

I solved this locally by changing the lines to.

       /* store the change-over time in GMT form by adding offset */
       tz->__tzrule[i].change = (time_t)days * SECSPERDAY +
       tz->__tzrule[i].s + tz->__tzrule[i].offset;

Adding the cast time_t should not cause a problem when time_t is still 32bit.

I don't know if this is correct solution. I also only needed this function. So more bugs like this could be hidden away in the time code.

Even when compiling the toolchain yourself this is hard to solve. Because during the ct-ng build fase all source code for newlib is overwritten.

igrr, 2020-11-05:
Hi @udeboer, as mentioned in the reply in newline mailing list, this is the correct solution. We will fix this issue in the next toolchain release. Thank you for reporting.

AxelLin, 2022-05-18:
> Hi @udeboer, as mentioned in the reply in newlib mailing list, this is the correct solution. We will fix this issue in the next toolchain release. Thank you for reporting.

Just wondering if the is already fixed?

antmak, 2025-03-17:
Fixed since IDF v5.0. `time_t` - 64-bit signed int without any special options. For details see [System time](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/system_time.html)

closed 2025-03-17 by antmak