espressif/esp-idf #6071: LED Control LEDC_INTR_FADE_END interrupt, no way of knowing which channel/pin fired it

howroyd, 2020-11-04:
**Is your feature request related to a problem? Please describe.**

Using the interrupt facility of the [LED Control API](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html) there is no way of knowing which channel has fired the interrupt.

The `ledc_isr_register` function can take an argument which is passed as a `void*` to the ISR but there is no way to register an ISR to a particular pin, channel, timer, etc.  I am combining this with the event loop library to fire a callback when a fade is complete.  The main purpose of this is to create a heartbeat type of effect; so once the fade to high is complete, begin the fade to low, and vice versa.

**Describe the solution you'd like**

To be able to know, within the ISR, which pin fired the interrupt due to the `LEDC_INTR_FADE_END` event.

Alvin1Zhang, 2020-11-05:
Thanks for raising this feature request.

costaud, 2020-11-05:
Each channel would trigger different interrupt status bits in LEDC.<br>
If you would make your own ISR implementation, you could get the interrupt status by calling the API in LL level.<br>
static inline void ledc_ll_get_fade_end_intr_status(ledc_dev_t *hw, ledc_mode_t speed_mode, uint32_t *intr_status)

howroyd, 2020-11-05:
> Each channel would trigger different interrupt status bits in LEDC.
> 
> If you would make your own ISR implementation, you could get the interrupt status by calling the API in LL level.
> 
> static inline void ledc_ll_get_fade_end_intr_status(ledc_dev_t *hw, ledc_mode_t speed_mode, uint32_t *intr_status)

Thanks @costaud ,

I added a requirement of `soc` into my `CMakeLists.txt` and included the header `hal/ledc_hal.h` to get access to the `ledc_ll_get_fade_end_intr_status` function and the `LEDC_LL_GET_HW()` macro.

My interrupt handler looks like this:
```
void IRAM_ATTR Led::interrupt_handler(void* arg)
{
    uint32_t channel{};
    ledc_ll_get_fade_end_intr_status(LEDC_LL_GET_HW(), LEDC_HIGH_SPEED_MODE, &channel);

    if (!channel)
    {
        channel |= interrupt_low_speed_flag;
        ledc_ll_get_fade_end_intr_status(LEDC_LL_GET_HW(), LEDC_LOW_SPEED_MODE, &channel);
    }

    if (loop_handle)
        istatus = esp_event_isr_post_to(loop_handle, LED_EVENTS, LEDC_INTR_FADE_END, &channel, sizeof(channel), nullptr);
}
```

Where `interrupt_low_speed_flag` is `1U << 8` so I can flag to my event loop whether this was the high speed or low speed bank.

I can confirm this now gives me the functionality I was after to find the channel and seems to be working.

Notes for @Alvin1Zhang :
I assume because I am compiling CPP (?), when I included the header `hal/ledc_hal.h` I got errors for the following functions and had to add a cast to the return type to get it to compile:
- `ledc_ll_get_slow_clk_sel`
- `ledc_ll_get_slow_clk_sel`
- `ledc_ll_get_channel_timer`

I think my request for a more convenient mechanism is still valid and would be great to see incorporated into the IDF, particularly in my case since an edit to API functions were needed.

Many thanks