espressif/esp-idf #6073: esp32 vroom32 ELx200 traces

fariouche, 2020-11-04:
## Environment

- Module or chip used:  [ESP32-WROOM-32]
- IDF version (run ``git describe --tags`` to find it):
    v4.3-dev-1720-g494a124d9
- Build System:        Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         [USB]

## Problem Description

After some random time, I get a lot of traces ELx200 and ELx40.... shortly after that all my BLE devices (2 of them) are disconnected. and BLE seems to not work anymore. But no crash, not hang, no reboot...
My application is using Wifi sta, and 2 BLE devices connected. + a BLE scan, all at the same time.
I cannot find the origin of this trace (grep "ELx" didn't return anything).
After some time, the traces are stopping and my BLE devices are disconnected. Scan is stopped.
I didn't checked if Wifi is still working, but my task that is handling the button is still working.

So I suspect that this trace is coming from the Bluetooth firmware, but why?
My application was doing nothing, just connected to the BLE devices and polling the button.

### Expected Behavior

### Actual Behavior

### Steps to reproduce
I do not know. In my case, just start everything, connect to a wifi AP and connect 2 BLE devices. Do a BLE scan in the background....  and wait.
Sometimes, one night no problems. Other times, the error happens after 1h...

If you know what this trace means, this may help me track down the issue...
Since I find no reference to this trace, it's difficult to understand what happened, where to look at or catch it to reboot or panic at least...

Thank you for your help!

Alvin1Zhang, 2020-11-05:
Thanks for reporting.

fariouche, 2020-11-10:
Hello,

Did you get a chance to look at this error? At least where this trace is coming from? I cannot help you because I have no clue from where or what this trace means
Thank you

riccardosala, 2020-11-17:
We have the same problem in our design based on ESP32-WROVER-B module.
IDF Version: v4.2-beta1-227-gf0e87c933
After this error code the Bluetooth stops to advertise and is no longer connectable.
Wifi, camera capture and GPIOs continues to work.

The problem occurs after a few hours of Bluetooth activity.

alutov, 2020-11-28:
The same problem with v4.3-dev-1720-g494a124d9-dirty. Error in precompiled library libbtdm_app.a. Ble stack 8192, 3 ble devices.

alutov, 2020-11-29:
The same problem on esp32 wroom 4M. IDF version: v4.3-dev-1901-g178b122c1 and  v4.3-dev-1720-g494a124d9. All works except BLE.

fariouche, 2020-11-29:
It is very rare but last time I got this error while the app was only doing a BLE scan! (my esp32 was running for 48h non stop in that case)

xiewenxiang, 2020-11-30:
@fariouche @alutov @riccardo-springtech 
Thank you for the information you provided. We will fix it as soon as possible and give you feedback.

alutov, 2020-11-30:
I tried to decrease and increase the memory stack. Does not help. Suspect that the problem appears if there are several esp32 clients are nearby. Now I turned off CONFIG_BT_GATTS_ENABLE in bluetooth settings sdkconfig as I use only BLE client. Watching...

ROYMURLI, 2020-12-11:
Hi, Any updates on this. I am also facing this issue. IDFv- v4.3-dev-1561-g357a27760
I am also attaching the screenshot and sdconfig file. 

[image]
[sdkconfig (copy).txt](https://github.com/espressif/esp-idf/files/5677645/sdkconfig.copy.txt)

bittailor, 2020-12-16:
I also see this issue on v4.2-beta1-227-gf0e87c933 on a ESP32-DevKitC-VB.

I use the NimBLE stack. The eps32 used a GATT client connected to ~8 GATT servers. The eps32 is permanently connected to those GATT servers for around 2-4 days before it happens.

When it happens I see a lot of 
```
ELx200
``` 
log entries and then a first BLE connection disconnects (BLE_GAP_EVENT_DISCONNECT). My code then tries to reconnect after one second.
```
GAP procedure initiated: connect; peer_addr_type=0 peer_addr=...
```
With this the log entries change to 
```
ELx40
```
Then all other connections also disconnect. And the reconnect of the first connection receives a BLE_GAP_EVENT_CONNECT 0 with an error status 0x0d (BLE_HS_ETIMEOUT=Operation timed out). 
Then for the next reconnect already `ble_gap_connect` fails with 0x13 (BLE_HS_ETIMEOUT_HCI) 
1 ms later the `ble_hs_cfg.reset_cb` callback is executed with reason 0x13 (BLE_HS_ETIMEOUT_HCI) 
~ 50ms later the `ble_hs_cfg.sync_cb` callback is executed.

My code does not yet fully handle `reset_cb` and `sync_cb` so I can not yet say if reconnects would work again after the `sync_cb`.

saloid, 2021-01-19:
Same issue. Release 4.2. I am using Bluedroid ble server and wifi STA+AP.

AmrutaCh, 2021-01-29:
Noticed same issue when device was kept running overnight.

- ESP-IDF v4.2-dirty 2nd stage bootloader
- Using Bluedroid BLE server and wifi STA 
- No active BLE connection
- After around 26K `ELx200`, watchdog timeouted and device rebooted

```
ELx200
ELx200
ELx200
ELx200
ELx200
ASSERT_PARAM(4096 0), in rwble.c at line 349
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x40082e48  PS      : 0x00060034  A0      : 0x80088622  A1      : 0x3ffbe544  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x60008048  A5      : 0x00000000  
A6      : 0x00000004  A7      : 0x3ffbdbdc  A8      : 0x80082e48  A9      : 0x3ffbe524  
A10     : 0x00000000  A11     : 0x0000002d  A12     : 0x00000014  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40082d80  LEND    : 0x40082d87  LCOUNT  : 0x00000000  
Core  0 was running in ISR context:
EPC1    : 0x4000bff0  EPC2    : 0x00000000  EPC3    : 0x40082e4b  EPC4    : 0x40082e48

Backtrace:0x40082e45:0x3ffbe544 |<-CORRUPTED
```

chhajedji, 2021-02-10:
Hi @fariouche and others,

I tried to recreate the issue but wasn't able to get it. I have a few queries

1) Is [`386a8e37f19fecc9ef62e72441e6e1272fa985b9`](https://github.com/espressif/esp-idf/commit/386a8e37f19fecc9ef62e72441e6e1272fa985b9) this commit present in your working tree? It is supposed to solve this issue. Please let me know if this is present and still you are facing the issue.

2) As I understood, you have 2 devices, one running gatt client and other running gatt server, connected to each other. One deivce (say gatt server) doing BLE scan and other (gatt client) acting as AP, connected to a station and doing nothing over wifi. Is this correct? If not then please share the procedure to recreate this issue.

3) What is the approximate time after which you see the issue for first time.

alutov, 2021-02-10:
As I understand it, the compiled ble library was replaced in December 2020? I didn't know about it, I will try, thanks.

fariouche, 2021-02-10:
@chhajedji , you. Are correct.
In fact I have many more ble devices around in the building that are not mine.
Lately I did see the but only once.

I'm starting to think that this bug is triggered by some BLE devices, or the number of BLE devices around,. I don't know....

There is a "real time " bug in the BLE that makes bugs really difficult to fix or reproduce.

For example, yesterday I tried to connect to a BLE device and it was not working anymore, while it was perfectly working the morning. The code didn't change at all. I had to comment my heart code that enable security MITM to have it work again. If not, I'm get no response to set mtu request function and then I'm disconnected by the device.... Strange

I will check if I have  the commit

chhajedji, 2021-02-17:
Hi,

Is this issue still present after the fixing commit?

If yes, please try again after calling this function after `esp_bt_controller_enable()`

```c
extern int esp_bt_controller_set_log_level(int level);
esp_bt_controller_set_log_level(0);
```

fariouche, 2021-02-17:
Hello @chhajedji 

It is long to reproduce. I've updated to latest master head branch and I do not see the error elx200... but unfortunately, the error is different this time but with same effects (note that I do not set a scan timeout, I want infinite scan time) :
scan_evt timeout
E (45010378) BT_HCI: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x200c
E (45910378) BT_HCI: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x200c

chhajedji, 2021-02-23:
Hi @fariouche 

I have tested for almost one week continuous and still didn't find any error. Neither any `ELx` error nor the latest `command_timed_out` issue you reported. Can you give some more pointers regarding the issue which can help me recreate or debug in the issue. You can also see my setup and recommend if I should change something.

My setup:
I have 2 devices: 1 running gatt server (1 task), other running gatt client, a wifi connection to AP and a BLE scan (3 task).

fariouche, 2021-02-23:
Hello @chhajedji 
Yes, same for me.

But I didn't see any error since then... still looking and very hard to reproduce.

I see an other strange behavior that may be not related, when my scan reports sometimes an adv data from an other ble device (I still need to check with the sample application to be sure it's not on my side. I believe I will open a new ticket if this is confirmed)

Good news is I didn't see the ELx200 error... bad news is that BLE is still not production ready for my use case but it's close!

fariouche, 2021-02-25:
Just as a note: I've just have seen an Elx40 trace, got my device disconnected.... but this is not a fatal error. The esp32 was able to reconnect to it right behind.
No more Elx200 so far

chhajedji, 2021-02-26:
Hi @fariouche 

Good to know that `ELx200` are no more seen. `ELx40` are also handled so hopefully seeing them is just an indication of a state. Should not be an issue.

> I see an other strange behavior that may be not related, when my scan reports sometimes an adv data from an other ble device

Can you open a new issue explaining more details about this. This would be easier for us to track the progress. Also if "ELx" errors are resolved, please close this issue.

fariouche, 2021-02-26:
yes, I've opened an other case.

I think that we ca close this ticket... we'll reopen it if it hapens again

closed 2021-02-26 by fariouche

romocitto88, 2021-03-03:
Hi, same issue: I'm working with BLE and WiFi together, and I was trying to test the reconnection to the router (router on with ethernet cable sconnected) when this log came out.
After that, the BLE stopped working.
I'm trying to reproduce it sistematically.

chhajedji, 2021-03-03:
@romocitto88 If you are getting ELx errors, see if `386a8e37f19fecc9ef62e72441e6e1272fa985b9` this commit is present in your working tree. It is supposed to solve the "ELx" issues.

romocitto88, 2021-03-03:
Sorry, I'm using platformIO how can I verify the commit?

AngelVersa, 2021-03-31:
I am facing this issue with "ESP-IDF v4.2-beta1-227-gf0e87c933" version. There is some way to fix this issue without update to the latest version?

chhajedji, 2021-03-31:
@AngelVersa please replace the lib at `$IDF_PATH/components/bt/controller/lib/libbtdm_app.a` with the lib attached. It contains the fix for this issue on your IDF version.

[lib.zip](https://github.com/espressif/esp-idf/files/6235609/lib.zip)

AngelVersa, 2021-03-31:
@chhajedji thanks for your quick reply. I will try this fix on some devices, let them work for a couple days and will give feedback.