espressif/esp-idf #6075: idf.py Bootloader Build Pauses Unexpectedly on Windows in ESP-IDF 3.3.2

andycarle, 2020-11-05:
Greetings!

I have been working to help a contributor with [this open issue](https://github.com/Moddable-OpenSource/moddable/issues/372) on the [Moddable SDK repository](https://github.com/Moddable-OpenSource/moddable). 

They are reporting an unexpectedly long pause during the bootloader build phase of the idf.py build for ESP32. Since this pause is in the ESP-IDF build, I was hoping a contributor over here might be able to help point me in the right direction to further diagnose the problem.

## Environment
- Development Kit:  M5StickC
- Module or chip used:  ESP32-PICO-D4
- IDF version: v3.3.2
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
- Operating System: Windows
- (Windows only) environment type: Developer Command Prompt for VS 2017
- Using an IDE?: No

## Problem Description
The Moddable SDK tools are invoking an ESP32 idf.py build with this command:

`python C:\Users\rhys\esp32\esp-idf\tools\idf.py -n -p COM3 -b 1500000 -B C:\Users\rhys\Projects\moddable\build\tmp\esp32\m5stick_c\debug\idf app-flash -D mxDebug=1 SDKCONFIG_H="C:\Users\rhys\Projects\moddable\build\tmp\esp32\m5stick_c\debug\idf\config\sdkconfig.h" CMAKE_MESSAGE_LOG_LEVEL=ERROR DEBUGGER_SPEED=1500000`

The build produces correct output, but unexpectedly pauses for about 18 seconds between the step that reports:

```
Executing "ninja -v all"...
[1/5] cmd.exe /C "cd /D C:\Users\rhys\Projects\moddable\build\tmp\esp32\m5stick_c\debug\idf\bootloader && "C:\Program Files\CMake\bin\cmake.exe" --build ."
ninja: no work to do.
```
and the step
```
[2/3] cmd.exe /C "cd . && C:\PROGRA~1\ESPRES~1\ESP-ID~1\tools\bin\XT1FE9~1.EXE  -mlongcalls  -nostdlib @CMakeFiles/xs_esp32.elf.rsp  -o xs_esp32.elf  && cd ."
```

You can see the issue in this screen capture:
[image]

This long pause does not happen on other systems we have tested with. We would like to sort out what is happening during that pause so that we can advise our users on how to fix it.

Can anyone here recommend steps we can take to determine what part of the build process is causing this delay? Thank you in advance!

### Expected Behavior

Bootloader build completes with no major pauses.

### Actual Behavior

Bootloader build pauses for about 18 seconds between 
```
Executing "ninja -v all"...
[1/5] cmd.exe /C "cd /D C:\Users\rhys\Projects\moddable\build\tmp\esp32\m5stick_c\debug\idf\bootloader && "C:\Program Files\CMake\bin\cmake.exe" --build ."
ninja: no work to do.
```
and
```
[2/3] cmd.exe /C "cd . && C:\PROGRA~1\ESPRES~1\ESP-ID~1\tools\bin\XT1FE9~1.EXE  -mlongcalls  -nostdlib @CMakeFiles/xs_esp32.elf.rsp  -o xs_esp32.elf  && cd ."
```

projectgus, 2020-11-06:
Hi @andycarle ,

Thanks for such a detailed bug report, even a video! That is indeed a long pause.

Do I understand correctly that you see this pause on all the Windows systems you've tried this on, but usually it's more like 2 seconds (still relatively long!) rather than 18 seconds which is on one particular computer?

This stage of the build is happening inside ninja (in the ninja build script generated by CMake).

I can't say for sure what is happening, but one thing that jumps out at me is that this command:

> [2/3] cmd.exe /C "cd . && C:\PROGRA~1\ESPRES~1\ESP-ID~1\tools\bin\XT1FE9~1.EXE  -mlongcalls  -nostdlib @CMakeFiles/xs_esp32.elf.rsp  -o xs_esp32.elf  && cd ."

Is actually running the gcc linker to link the app ELF file, not the bootloader ELF file. The bootloader step is the previous one and it's come back with "no work to do" already.

Linking can take a few seconds... my Windows test computer is relatively low end (i3 6th gen) and it's definitely a couple of seconds to link a simple app (not 18, but a couple). The more complex the app is, the longer the linking step can take. Linking can also use a lot of RAM, if the 18 second system is tight on RAM then maybe it's paging to make a slow-ish process into a very slow one.

My guess is that the output is being buffered in a way that you're not seeing the linker command being printed until after the linker has finished executing, and this makes it look like a long pause before it starts.

To confirm this, if you have access to the 18 second computer then please look in Windows task manager during the pause. If you see a single gcc.exe process running and using CPU the whole time, it's probably the app linker step.

The last thing I suggest checking is to show "More Details" in Task Manager and look for any service host process hogging the CPU during this time. We find that Windows Defender service can slow down the IDF build by more than 50% as it scans every file access. In ESP-IDF v4.1 the ESP-IDF Windows Tools Installer added an option to automatically add Defender exclusions for the ESP-IDF toolchain processes, as a trade-off between build time and security. There is no such option in earlier versions, but running an experiment by temporarily disabling Defender live scanning is probably worthwhile.

andycarle, 2020-11-06:
Hi @projectgus, 

Thank you for all the information; that is very helpful!

> Do I understand correctly that you see this pause on all the Windows systems you've tried this on, but usually it's more like 2 seconds (still relatively long!) rather than 18 seconds which is on one particular computer?

That is correct. As you suggest, it varies a bit from computer to computer: ~2.5 seconds on my laptop, maybe 1 second on my Ryzen 9 3900x, and 18 seconds on the reported machine.

> Is actually running the gcc linker to link the app ELF file, not the bootloader ELF file. The bootloader step is the previous one and it's come back with "no work to do" already.

Ah, interesting. We had been narrowing in on the bootloader as the issue because the contributor [had reported earlier](https://github.com/Moddable-OpenSource/moddable/issues/372#issuecomment-642531307) that the long pause can be resolved by using the `app-flash` idf.py command rather than our normal `build flash` command. But given your comment, it seems likely that there is some nuance to that change that we have missed?

> To confirm this, if you have access to the 18 second computer then please look in Windows task manager during the pause. If you see a single gcc.exe process running and using CPU the whole time, it's probably the app linker step.

> The last thing I suggest checking is to show "More Details" in Task Manager and look for any service host process hogging the CPU during this time. We find that Windows Defender service can slow [...]

Got it; those are good suggestions. I'll pass those along to our contributor and see what we hear back.

Thank you again!
&nbsp;- Andy

wilberforce, 2020-11-07:
Thanks for you help.
[image]

It does appear to be the linking step as suggested.

I have excluded the folder the project is in, and also `C:\Program Files\Espressif\ESP-IDF Tools\tools\xtensa-esp32-elf\bin`
and added an exclusion for:
`C:\Program Files\Espressif\ESP-IDF Tools\tools\xtensa-esp32-elf\bin\ld.exe`

Turning off here did not seem to affect the build time:
[image]

projectgus, 2020-11-08:
Hi @wilberforce ,

Ah yes, this looks like it indeed!

Is this the "More Details" view of the Processes tab? If it is then I don't think Windows Defender is causing any slowdown, when this happens you see a Service Host process for Windows Defender using more CPU than even ld.exe (it's very unsubtle when it takes over!) In this case I think you can safely turn these options back on.

There are only two other options I can think of:

* Some other virus scanner type process is intercepting file access and slowing the linker pass down to 18 seconds.
* The linker pass is actually taking 18 seconds (the causes being some combination of: system I/O throughput, single thread CPU performance, available free RAM / paging activity, complexity of the app you're linking)

> the long pause can be resolved by using the app-flash idf.py command rather than our normal build flash command

@andycarle, this sounds a bit odd. The ESP-IDF build system should only be triggering a re-link if one of the source files or other input files of the ELF file changed. The `idf.py app` target (or `idf.py app-flash`) shouldn't need to rebuild the ELF otherwise, it should be exiting or moving directly to the flashing phase.

The CMake build system in ESP-IDF v3.3 is a "preview" version and a lot of things were cleaned up in ESP-IDF v4.0 when we made it the default, so it's possible there is a bug here. If you are able to find a reliable way to narrow down the trigger of the re-link then please let us know, we will see if we can fix it.

projectgus, 2020-11-26:
@wilberforce @andycarle Is there anything else you think we should chase up here, from the ESP-IDF side?

wilberforce, 2020-11-27:
Thanks for following up.
I think it's the linking step of an relatively large bin, ~1.8Mb that is talking the time.
I believe this can be closed - it will be interesting to review when the project goes to version 4 of the idf.

closed 2020-11-29 by projectgus

GerryBriggs, 2024-01-26:
I am at the linker step and the build just hangs forever. At least over ten minutes. I am on Windows VS Code Espressif latest extension. Come to think of it, there was a new IDF update 9 hours ago and the last time I compiled fully without the problem may have been before that.