espressif/esp-idf #6076: example_connect(): Never receiving global IPv6 address

nx518, 2020-11-05:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version:   v4.2-beta1-227-gf0e87c933
- Build System:   idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

Can't obtain global IPv6 address with any example using example_connect eg. sockets/udp_server.
Routers tested:  Fritzbox 7430 stock FW and TP-Link Archer C50v4 with openwrt 19.07.4

All other PCs and phones on the network get linklocal, unique and global IPv6 addresses.
ESP32 only gets linklocal.

### Expected Behavior
program receives global ipv6 address.

### Actual Behavior
program waits for global ipv6 address forever.

### Code to reproduce this issue

examples/protocols/sockets/upd_server
with CONFIG_EXAMPLE_CONNECT_IPV6_PREF_GLOBAL

## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7072
ho 0 tail 12 room 4
load:0x40078000,len:13732
load:0x40080400,len:3988
0x40080400: _init at ??:?

entry 0x40080688
I (30) boot: ESP-IDF v4.2-beta1-227-gf0e87c933 2nd stage bootloader
I (31) boot: compile time 17:42:43
I (31) boot: chip revision: 1
I (35) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 4MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (79) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00100000
I (94) boot: End of partition table
I (99) boot_comm: chip revision: 1, min. application chip revision: 0
I (106) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x16f5c ( 94044) map
I (151) esp_image: segment 1: paddr=0x00026f84 vaddr=0x3ffb0000 size=0x037f4 ( 14324) load
I (157) esp_image: segment 2: paddr=0x0002a780 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/mx/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (158) esp_image: segment 3: paddr=0x0002ab8c vaddr=0x40080404 size=0x0548c ( 21644) load
I (176) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x724fc (468220) map
0x400d0020: _stext at ??:?

I (354) esp_image: segment 5: paddr=0x000a2524 vaddr=0x40085890 size=0x102e0 ( 66272) load
0x40085890: ram_write_gain_mem at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7.c:1089

I (395) boot: Loaded app from partition at offset 0x10000
I (396) boot: Disabling RNG early entropy source...
I (396) cpu_start: Pro cpu up.
I (400) cpu_start: Application information:
I (404) cpu_start: Project name:     udp_server
I (410) cpu_start: App version:      1
I (414) cpu_start: Compile time:     Nov  4 2020 20:25:58
I (420) cpu_start: ELF file SHA256:  ead5ebfc0cbce563...
I (426) cpu_start: ESP-IDF:          v4.2-beta1-227-gf0e87c933
I (433) cpu_start: Starting app cpu, entry point is 0x400818a8
0x400818a8: call_start_cpu1 at /home/mx/esp/esp-idf/components/esp32/cpu_start.c:287

I (0) cpu_start: App cpu up.
I (443) heap_init: Initializing. RAM available for dynamic allocation:
I (450) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (456) heap_init: At 3FFB7D70 len 00028290 (160 KiB): DRAM
I (462) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (469) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (475) heap_init: At 40095B70 len 0000A490 (41 KiB): IRAM
I (481) cpu_start: Pro cpu start user code
I (500) spi_flash: detected chip: generic
I (500) spi_flash: flash io: dio
I (500) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (603) wifi:wifi driver task: 3ffc01b0, prio:23, stack:6656, core=0
I (603) system_api: Base MAC address is not set
I (603) system_api: read default base MAC address from EFUSE
I (623) wifi:wifi firmware version: f8b2044
I (623) wifi:wifi certification version: v7.0
I (623) wifi:config NVS flash: enabled
I (623) wifi:config nano formating: disabled
I (633) wifi:Init data frame dynamic rx buffer num: 32
I (633) wifi:Init management frame dynamic rx buffer num: 32
I (643) wifi:Init management short buffer num: 32
I (643) wifi:Init dynamic tx buffer num: 32
I (653) wifi:Init static rx buffer size: 1600
I (653) wifi:Init static rx buffer num: 10
I (653) wifi:Init dynamic rx buffer num: 32
I (663) wifi_init: rx ba win: 6
I (663) wifi_init: tcpip mbox: 32
I (673) wifi_init: udp mbox: 6
I (673) wifi_init: tcp mbox: 6
I (673) wifi_init: tcp tx win: 5744
I (683) wifi_init: tcp rx win: 5744
I (683) wifi_init: tcp mss: 1440
I (693) wifi_init: WiFi IRAM OP enabled
I (693) wifi_init: WiFi RX IRAM OP enabled
I (703) example_connect: Connecting to OpenWrt...
I (793) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (793) wifi:mode : sta (3c:71:bf:ab:33:10)
I (803) example_connect: Waiting for IP(s)
I (2133) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2133) wifi:state: init -> auth (b0)
I (2133) wifi:state: auth -> assoc (0)
I (2143) wifi:state: assoc -> run (10)
I (2143) wifi:connected with OpenWrt, aid = 1, channel 11, BW20, bssid = 98:da:c4:5a:0e:f3
I (2143) wifi:security: Open Auth, phy: bgn, rssi: -32
I (2143) wifi:pm start, type: 1

I (2213) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (3093) esp_netif_handlers: example_connect: sta ip: 192.168.1.226, mask: 255.255.255.0, gw: 192.168.1.1
I (3093) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.1.226
I (3593) example_connect: Got IPv6 event: Interface "example_connect: sta" address: fe80:0000:0000:0000:3e71:bfff:feab:3310, type: ESP_IP6_ADDR_IS_LINK_LOCAL
```

```
#
# Example Connection Configuration
#
CONFIG_EXAMPLE_CONNECT_WIFI=y
CONFIG_EXAMPLE_WIFI_SSID="OpenWrt"
CONFIG_EXAMPLE_WIFI_PASSWORD=""
# CONFIG_EXAMPLE_CONNECT_ETHERNET is not set
CONFIG_EXAMPLE_CONNECT_IPV6=y
# CONFIG_EXAMPLE_CONNECT_IPV6_PREF_LOCAL_LINK is not set
CONFIG_EXAMPLE_CONNECT_IPV6_PREF_GLOBAL=y
# CONFIG_EXAMPLE_CONNECT_IPV6_PREF_SITE_LOCAL is not set
# CONFIG_EXAMPLE_CONNECT_IPV6_PREF_UNIQUE_LOCAL is not set
# end of Example Connection Configuration
```

Alvin1Zhang, 2020-11-06:
Thanks for reporting and letting us know, we will look into.

david-cermak, 2020-11-06:
Hi @nx518 

Is the SLAAC enabled in lwIP in your project? You can find this option under *component config* -> *lwIP* -> **Enable IPV6 stateless address autoconfiguration**, i.e. `CONFIG_LWIP_IPV6_AUTOCONFIG=y` (this is off by default, and the options you are mentioning above are only for the example connection code to prefer (wait for) a specific address type.

nx518, 2020-11-06:
Thank you, enabling CONFIG_LWIP_IPV6_AUTOCONFIG solved it. Now I have all needed IPv6 addresses.

Would you consider enabling SLAAC by default and add the term SLAAC to the description in `components/lwip/Kconfig` like:
```
    config LWIP_IPV6_AUTOCONFIG
        bool "Enable IPV6 stateless address autoconfiguration (SLAAC)"
        default y
        help
            Enabling this option allows the devices to IPV6 stateless address autoconfiguration (SLAAC).

            See RFC 4862.
```

david-cermak, 2020-11-06:
@nx518 I don't think we'd like to enable the *SLAAC* by default. There might be a slight security concern (users might end up with devices, globally accessible on the internet, without even knowing about it).
Good point with the **SLAAC** description, it would be useful to have this in the description/name field of the config option, I think.

closed 2020-11-19 by projectgus