espressif/esp-idf #6077: Failed to execute export.bat for esp-idf v4.1: AttributeError: 'list' object has no attribute 'read'

VicSpectator, 2020-11-05:
## Environment

- IDF version: v4.1
- Operating System: Windows
- (Windows only) environment type: Plain Command Prompt.
- Python 3.7.9

## Problem Description

Trying to prepare development environment. ESP-IDF Tools installer failed. That's why I decided to install everything manually. See log. The mistake is strange. Is it an issue in my environment but not esp-idf issue?

### Steps to reproduce

1. Install required software for Windows
2. Download esp-idf from github by following command:
`git clone -b v4.1 --recursive https://github.com/espressif/esp-idf.git esp-idf-v4.1`
3. Create new IDF_TOOLS_PATH environment variable
4. Launch install.bat
5. Launch export.bat
I don't have ESP-IDF Command Prompt. That's why I need to execute export.bat if I'm not mistaken.

## Logs

D:\Viktor\software\esp-idf-v4.1>export.bat
Setting IDF_PATH: D:\Viktor\software\esp-idf-v4.1

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool ninja found in PATH: 1.10.0.
Traceback (most recent call last):
  File "D:\Viktor\software\esp-idf-v4.1\tools\idf_tools.py", line 1387, in <module>
    main(sys.argv[1:])
  File "D:\Viktor\software\esp-idf-v4.1\tools\idf_tools.py", line 1383, in main
    action_func(args)
  File "D:\Viktor\software\esp-idf-v4.1\tools\idf_tools.py", line 1008, in action_export
    idf_python_env_path, idf_python_export_path, virtualenv_python = get_python_env_path()
  File "D:\Viktor\software\esp-idf-v4.1\tools\idf_tools.py", line 847, in get_python_env_path
    cwd=global_idf_path, env=os.environ).decode()
  File "C:\Python\Python37\lib\subprocess.py", line 411, in check_output
    **kwargs).stdout
  File "C:\Python\Python37\lib\site-packages\run\__init__.py", line 145, in __new__
    process = cls.create_process(command, stdin, cwd=cwd, env=env, shell=shell)
  File "C:\Python\Python37\lib\site-packages\run\__init__.py", line 121, in create_process
    shlex.split(command),
  File "C:\Python\Python37\lib\shlex.py", line 310, in split
    return list(lex)
  File "C:\Python\Python37\lib\shlex.py", line 299, in __next__
    token = self.get_token()
  File "C:\Python\Python37\lib\shlex.py", line 109, in get_token
    raw = self.read_token()
  File "C:\Python\Python37\lib\shlex.py", line 140, in read_token
    nextchar = self.instream.read(1)
AttributeError: 'list' object has no attribute 'read'

Alvin1Zhang, 2020-11-06:
Thanks for reporting.

VicSpectator, 2020-11-06:
The same problem with **esp-idf-tools-setup-2.3.exe** when I use existing esp-idf on my hard drive:

> Running command: C:\Python\Python37\python.exe "D:\Viktor\software\esp-idf-v4.1\tools\idf_tools.py" --idf-path D:\Viktor\software\esp-idf-v4.1 install-python-env
>                           Traceback (most recent call last):
>                             File "D:\Viktor\software\esp-idf-v4.1\tools\idf_tools.py", line 1387, in <module>
>                               main(sys.argv[1:])
>                             File "D:\Viktor\software\esp-idf-v4.1\tools\idf_tools.py", line 1383, in main
>                               action_func(args)
>                             File "D:\Viktor\software\esp-idf-v4.1\tools\idf_tools.py", line 1172, in action_install_python_env
>                               idf_python_env_path, _, virtualenv_python = get_python_env_path()
>                             File "D:\Viktor\software\esp-idf-v4.1\tools\idf_tools.py", line 847, in get_python_env_path
>                               cwd=global_idf_path, env=os.environ).decode()
>                             File "C:\Python\Python37\lib\subprocess.py", line 411, in check_output
>                               **kwargs).stdout
>                             File "C:\Python\Python37\lib\site-packages\run\__init__.py", line 145, in __new__
>                               process = cls.create_process(command, stdin, cwd=cwd, env=env, shell=shell)
>                             File "C:\Python\Python37\lib\site-packages\run\__init__.py", line 121, in create_process
>                               shlex.split(command),
>                             File "C:\Python\Python37\lib\shlex.py", line 310, in split
>                               return list(lex)
>                             File "C:\Python\Python37\lib\shlex.py", line 299, in __next__
>                               token = self.get_token()
>                             File "C:\Python\Python37\lib\shlex.py", line 109, in get_token
>                               raw = self.read_token()
>                             File "C:\Python\Python37\lib\shlex.py", line 140, in read_token
>                               nextchar = self.instream.read(1)
>                           AttributeError: 'list' object has no attribute 'read'

igrr, 2020-11-06:
@VicSpectator The path `C:\Python\Python37\lib\site-packages\run\__init__.py` hints that you might have [`subprocess.run`](https://pypi.org/project/subprocess.run/) package installed, which is interfering with the standard `subprocess` module.

Could you please try to install that package (`C:\Python\Python37\pip.exe uninstall subprocess.run`), and then run the installer again?

VicSpectator, 2020-11-06:
> @VicSpectator The path `C:\Python\Python37\lib\site-packages\run\__init__.py` hints that you might have [`subprocess.run`](https://pypi.org/project/subprocess.run/) package installed, which is interfering with the standard `subprocess` module ([issue link](https://github.com/xando/subprocess.run/issues/5)).
> 
> Could you please try to install that package (`C:\Python\Python37\pip.exe uninstall subprocess.run`), and then run the installer again?

You are right. Thank's a lot! Uninstallation solved my problem. Sorry for creating the issue.
I hope somebody will help me to close the issue :) I don't see any button for doing this.

closed 2020-11-06 by igrr