espressif/esp-idf #6079: BLE-MESH Provisioner

MrCurio, 2020-11-06:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): v4.3-dev-905-g9e45dca65
- Build System:         [Make]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     [Linux]
- Using an IDE?: [Yes (VsCode)]
- Power Supply:         [USB]

## Problem Description
Hello, 
I am developing an Bluetooth Mesh application with the ESP32 and i think i have encountered a relevant issue, but i am not sure.
In my project, i am using a **MODIFIED** version of the fast provisioning, so that the first device starting (let's say the "top provisioner"), which has an "standard" provisioner (CONFIG_BLE_MESH_PROVISIONER = 1), has 1 element with: 

- Config Client
- Config Server
- Fast Prov Client
-  Fast Prov Server

The rest of the devices have the same element structure, but they use the feature of Fast Prov for provisioning (CONFIG_BLE_MESH_FAST_PROV = 1), instead of the "Standard Provisioner".

The idea of having a Fast Prov Server in the first device is to get the addresses via SET ACK messages from every node in the network so, everytime a nodes is provisioned, a SET message is sent to the Fast Prov Server of the 1st device and lately acked by the Server.

i have modified the code to have the previous performance, but SET messages seem to not be received from the Server. This behaviour makes me wonder the following:

Can a "standard" provisioner have a server in the same element? If not, should it work if they are in separate elements?  

If have realized that a "standard" provisioner cannot be at the same time node and provisioner (CONFIG_BLE_MESH_NODE && CONFIG_BLE_MESH_PROVISIONER), how does this affect my issue?

Thanks!

### Expected Behavior
Messages received by the server
### Actual Behavior
Messages are not being received, but they are succesfully sent.
 
### Steps to reproduce

1. Any sample code trying to use a "standard" provisioner with a server in the same element should work. For trying, the onoff example can be modified for testing purposes.

### Code to reproduce this issue

1st device messages:
```
/* Messages That are received */
static esp_ble_mesh_model_op_t fast_prov_srv_op[] = {
    ESP_BLE_MESH_MODEL_OP(ESP_BLE_MESH_VND_MODEL_OP_FAST_PROV_NODE_ADDR,         2),
    ESP_BLE_MESH_MODEL_OP_END,
};

/* Messages that are Received */
static esp_ble_mesh_model_op_t fast_prov_cli_op[] = {
    ESP_BLE_MESH_MODEL_OP(ESP_BLE_MESH_VND_MODEL_OP_FAST_PROV_INFO_STATUS,      1),
    ESP_BLE_MESH_MODEL_OP_END,
};

static esp_ble_mesh_model_t root_models[] = {
    ESP_BLE_MESH_MODEL_CFG_SRV(&config_server),
    ESP_BLE_MESH_MODEL_CFG_CLI(&config_client),
};

static esp_ble_mesh_model_t vnd_models[] = {
    ESP_BLE_MESH_VENDOR_MODEL(CID_ESP, ESP_BLE_MESH_VND_MODEL_ID_FAST_PROV_SRV,
    fast_prov_srv_op, NULL, &prov_info),
    ESP_BLE_MESH_VENDOR_MODEL(CID_ESP, ESP_BLE_MESH_VND_MODEL_ID_FAST_PROV_CLI,
    fast_prov_cli_op, NULL, &fast_prov_client),
};
```

Ohter devices
```
static const esp_ble_mesh_client_op_pair_t fast_prov_cli_op_pair[] = {
    { ESP_BLE_MESH_VND_MODEL_OP_FAST_PROV_INFO_SET,      ESP_BLE_MESH_VND_MODEL_OP_FAST_PROV_INFO_STATUS      },
    { ESP_BLE_MESH_VND_MODEL_OP_FAST_PROV_NODE_ADDR,     ESP_BLE_MESH_VND_MODEL_OP_FAST_PROV_NODE_ADDR_ACK    },
};
static esp_ble_mesh_model_op_t fast_prov_srv_op[] = {
    ESP_BLE_MESH_MODEL_OP(ESP_BLE_MESH_VND_MODEL_OP_FAST_PROV_INFO_SET,          3),
    ESP_BLE_MESH_MODEL_OP(ESP_BLE_MESH_VND_MODEL_OP_FAST_PROV_NODE_ADDR,         2),
    ESP_BLE_MESH_MODEL_OP_END,
};

static esp_ble_mesh_model_op_t fast_prov_cli_op[] = {
    ESP_BLE_MESH_MODEL_OP(ESP_BLE_MESH_VND_MODEL_OP_FAST_PROV_INFO_STATUS,    1),
    ESP_BLE_MESH_MODEL_OP(ESP_BLE_MESH_VND_MODEL_OP_FAST_PROV_NODE_ADDR_ACK,  0),
    ESP_BLE_MESH_MODEL_OP_END,
};

static esp_ble_mesh_model_t root_models[] = {
    ESP_BLE_MESH_MODEL_CFG_SRV(&config_server),
    ESP_BLE_MESH_MODEL_CFG_CLI(&config_client),
    ESP_BLE_MESH_MODEL_GEN_ONOFF_SRV(&onoff_pub, &onoff_server),
};

static esp_ble_mesh_model_t vnd_models[] = {
    ESP_BLE_MESH_VENDOR_MODEL(CID_ESP, ESP_BLE_MESH_VND_MODEL_ID_FAST_PROV_SRV,
    fast_prov_srv_op, NULL, &fast_prov_server),
    ESP_BLE_MESH_VENDOR_MODEL(CID_ESP, ESP_BLE_MESH_VND_MODEL_ID_FAST_PROV_CLI,
    fast_prov_cli_op, NULL, &fast_prov_client),
};
```

## Debug Logs
There are barely no significant logs

Thanks! Hope my doubt/bug can me resolved

Alvin1Zhang, 2020-11-09:
Thanks for reporting, we will look into.

Campou, 2020-11-09:
Hi @MrCurio 

Currently a standard Provisioner doesn't support having server models, except for Configuration Server model. Because having a server model means that the corresponding states could be changed by other nodes, but for a Provisioner, currently we haven't meet any requirements/applications that its states needs to be updated by other nodes.

If a Provisioner is going to support a server model, it needs some changes in the mesh stack.

Thanks.

MrCurio, 2020-11-09:
Hi @Campou , thank you for your reply!

Is there any future intention of developing a server model for the provisioner role in the stack? 

It would be interesting for certain applications were the information is set in the provisioner's server such as addresses, MACs..etc, instead of being requested by the provisioner. By doing this, some developments may reduce complexity.

Is there any workflow alternative, besides the fast provisioning demo example, that may be worth to mention for obtaining addresses without a primary provisioner role (Just a top provisioner and temporary provisioners)?
 
Anyway, i will modify my application for avoiding servers in the provisioner.

When i receive a final answer, i will proceed to close this issue.

Thank you!