espressif/esp-idf #6081: Error ESP_ERR_HTTP_FETCH_HEADER in HTTP POST

Gustbel, 2020-11-06:
Hello community, I am using the basic example to POST over HTTP in ESP32.

The problem is that when I do the **POST Perform**, I never get ```ESP_OK```
And instead I always get: ```ESP_ERR_HTTP_FETCH_HEADER```
The json stream is sent anyway, but I would like to know why I am getting this unexpected error.

I do not know if I have detected a bad behavior or I am failing in some step.

Here I leave the code that I am using:
```
    char* u = "http://192.168.1.115:8080/sensors";

    esp_http_client_config_t config = {
           .url = u,
           .event_handler = _http_event_handler,
    };
    client = esp_http_client_init(&config); 
    err = esp_http_client_perform(client);	/* Check connection */
    if (err == 0)
	{
		esp_http_client_set_url(client, u);
		esp_http_client_set_method(client, HTTP_METHOD_POST);
		esp_http_client_set_header(client, "Content-Type", "application/json");
		esp_http_client_set_post_field(client, j, strlen(j));
		err = esp_http_client_perform(client);
		printf ("error: %d\n", err);
        }
    esp_http_client_close(client);
    esp_http_client_cleanup(client);
```

As I said before the Json is sent correctly, but tbh I would like to receive `ESP_OK`

Thank you!!

Alvin1Zhang, 2020-11-09:
Thanks for reporting.

shubhamkulkarni97, 2020-11-09:
Hi @Gustbel,

`esp_http_client_perform()` should be called only after setting all the configurations of esp_http_client. Please check the implementation of POST request in http_rest_with_url example: https://github.com/espressif/esp-idf/blob/494a124d961c4c755685b22fe986c31826d6f503/examples/protocols/esp_http_client/main/esp_http_client_example.c#L142

If this doesn't resolve the issue, please share version of ESP-IDF you used to run the code (`git describe --tags`)

Thanks,
Shubham

Gustbel, 2020-11-09:
> Hi @Gustbel,
>
> `esp_http_client_perform()` should be called only after setting all the configurations of esp_http_client. Please check the implementation of POST request in http_rest_with_url [...]

Hi @shubhamkulkarni97 . 
I don't understand why you write that. In my code you can see that I do all the configuration before doing `esp_http_client_perform()`
my problem is not solved.

chegewara, 2020-11-09:
Duplicate:
https://github.com/espressif/esp-idf/issues/2684

shubhamkulkarni97, 2020-11-11:
@Gustbel, Can you share which server did you use for the test?

Please try using `http://httpbin.org/post` as test URL and check if you can perform http_request.

Can you share debug logs when issue occurs? (To enable debug logs, run `idf.py menuconfig`, goto `Component config > Log Output` and set Default log verbosity to debug)

Also enable socket debugging by setting [SOCKETS_DEBUG](https://github.com/espressif/esp-idf/blob/5d258af93c668d748a5d190137c1bf8ae58489b4/components/lwip/port/esp32/include/lwipopts.h#L755) to `LWIP_DBG_ON`.

Gustbel, 2020-11-19:
Thanks for the replies. Sorry, apparently the problem is the server what was i using, then i tried with others HTTP servers and in those cases it works fine.
Thank you very much for your answers

closed 2020-11-19 by Gustbel

kzoltaan, 2024-03-20:
I have same problem when the target webserver is Apache. On Nginx looks okay. I any new idea?