espressif/esp-idf #6082: ESP32 cannot connect to a smartphone

enimzuk, 2020-11-06:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): v4.3-dev-1521-g0289d1cc8
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [macOS]
- Using an IDE?: [Yes (VS Code)]
- Power Supply:         [USB]

## Problem Description

Hello,

We're developing MITM project which is supposed to stream audio from a smartphone to another headset both connected to one ESP32. It is based on BTStack develop branch: https://github.com/bluekitchen/btstack and its MITM sample:
https://bluekitchen-gmbh.com/files/aug-global/a2dp_mitm_demo.c

The headset is Philips SHB 5250. We're not sure that this behavior will reproduce with all types of headsets.
The smartphone is iphone 7 plus or Samsung galaxy S7. It doesn't look like it depends on a phone model.

First we initiate a connection to the headset from ESP32. Then we initiate a connection to the smartphone from ESP32.

We use the following write link policy fix to make sure that ESP32 stays in Master role when connected to the headset.
Only in this role configuration it has enough throughput to stream audio with this headset.

When we connect the headset we use the hci command of btstack to disable role switches after receiving HCI_EVENT_CONNECTION_COMPLETE event for the headset connection, e.g.

hci_send_cmd(&hci_write_link_policy_settings, mitm_context.a2dp_source_con_handle, LM_LINK_POLICY_DISABLE_ALL_LM_MODES) where LM_LINK_POLICY_DISABLE_ALL_LM_MODES is 0.

The headset connects successfully and ESP32 is in master role.

After that we try to connect the smartphone to start streaming audio from the smartphone to the headset.
However, the smartphone cannot connect.

HCI log doesn't provide much information to understand the issue. BTStack developers say that the actual issue is with ESP32.

The following is also notable.
We can connect both the headset and the smartphone without such issues if for example:
1. First we connect the smartphone and then we connect the headset with the fix mentioned above. The connections are established and the streaming is working fine. ESP32 is in Master role.
2. First we connect the headset and then the smartphone without the fix mentioned above. The connections are established but the headset is in Slave role in this case. Therefore, it cannot steam audio fast enough.

### Expected Behavior

ESP32 connects to the smartphone after the headset is connected with the write link policy fix.

### Actual Behavior

ESP32 cannot connect to the smartphone after the headset is connected with the write link policy fix.

### Steps to reproduce

1. Get latest btstack from develop branch
2. Compile and run mitm sample https://bluekitchen-gmbh.com/files/aug-global/a2dp_mitm_demo.c
2. Connect headset using console command 'h'
3. Connect smartphone using console command 'p'

### Code to reproduce this issue

BTStack from develop branch: https://github.com/bluekitchen/btstack
Sample application: https://bluekitchen-gmbh.com/files/aug-global/a2dp_mitm_demo.c

## Debug Logs

[logs.zip](https://github.com/espressif/esp-idf/files/5502408/logs.zip)

https://drive.google.com/drive/folders/1CvIkjiJ7Vm39Pt4G_zFrzuJe27FIgKxL?usp=sharing
OR
https://drive.google.com/drive/folders/1gd4sZkBdaDjndkYJ6Qta42DEmzMBjncU?usp=sharing

Thanks!

Alvin1Zhang, 2020-11-09:
Thanks for reporting.

mringwal, 2020-11-11:
Hi there

A few details:
- when the connection to the smartphone is created, it fails with connection timeout/link layer error
- it only happens with some headsets, not with all. e.g. it didn't reproduce for me.

Quick summary from HCI view:
- allow role switch as default link policy
- create connection to headset with allow role switch flag set to false
- then immediately disable role switches for this connection with hci_write_link_policy_settings
- create connection to smartphone with allow role switch flag set to true
-> connection fails to establish

Asides from trying to reproduce this, could you review the BTDM sources if there is anything suspicious about this use of link policies? the goal was to enforce the connection to headset to stay in master role.

Thanks 
 Matthias

munho-espressif, 2020-12-24:
Hi @mringwal 
1. Why did the developer of BTStack say  that the actual issue is with ESP32？
2. I can't open your log because I don't have a corresponding application.
Thanks

mringwal, 2020-12-24:
Hi @munho-espressif 

The connection to the smartphone fails with a link layer error. However, it only happens if there's already a connection to the specific Philips headset. If there's no second connection, or, the second connection is to some other speaker, the connection to the smartphone is established as expected. So, the connection seems to fail only if there other connection of the ESP32 is to this headphone. (that's why it seems that the ESP32 Link Layer is causing this issue. The headset should not have any influence on other connections od the ESP32)

You can open the log files with Wireshark.

Best
 Matthias

munho-espressif, 2020-12-25:
Hi @mringwal 

Ok, I will try to reproduce your issue.

Thanks

mringwal, 2020-12-25:
@munho-espressif That would be great. Asides from getting the same headphone, I don't know this could be reproduced - as mentioned, it doesn't happen with other headsets.

Wth-Esp, 2021-01-20:
@mringwal Thanks for your patience. We have bought the Philips SHB 5250 and will look into the problem tomorrow.

xiongweichao, 2021-01-25:
Hi, @mringwal 

First, we start the connection to the headset from ESP32. Then we started connecting to the smartphone from the ESP32, and there might be a connection failure. So I think it might have something to do with phone compatibility.

Thanks

Wth-Esp, 2021-01-25:
@xiongweichao  

Please provide the device list which cannot be connected after successfully connecting to Philips SHB 5250. 
Thanks.

mringwal, 2021-01-25:
Hi. The issue was reported by @enimzuk 
He originally reported issues with an iPhone 7 Plus and a Samsung Galaxy S7. 

I would assume it's somehow related to the Philips headset that tries to become master, but this is rejected in the example application. If that's true, it should also happen with other smartphones.

My reason is like this: if the second connection fails if the fist connection is with this Philips headset, but not with another headset, then the ESP32 is responsible for this. The connection to the headset should have no impact on the new one to the smartphone.

Have you been able to reproduce this? Do you also have a Bluetooth sniffer and are able to understand what's actually going on?

Thanks

mringwal, 2021-01-25:
@enimzuk Hi. Could you try with current esp-idf again? I've just rested a similar, but different issue from Jan 2020, and that one has been fixed by Espressif.

xiongweichao, 2021-01-27:
Hi, @mringwal 

The secure connection is not supported by ESP32.  The ESP32 could not establish a connection with HUAWEI P30 when tested with A2DP_MITM_DEMO. The reason why the connection failed is shown in the figure. ESP32 is in master role.

Thanks.
[image]

mringwal, 2024-07-10:
@enimzuk You still around? I wasn't able to reproduce this at the time.

closed 2025-03-14 by Alvin1Zhang