espressif/esp-idf #6084: Bluetooth Classic sending packet loss at "high" throughput

francis2tm, 2020-11-07:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

I'm using the esp_spp_write() function to send data over bluetooth classic. My application requires esp to send at most 8bytes every 1ms -> throughput of 64kbps (less than what esp32 is capable of). I noticed that when I send data every 1ms, I get packets lost in the receiver side, a laptop or a smartphone (tried with both of them), I know this because I put a sequence number in every frame I send. I think it's a problem with the bt API, because in the ESP side I check if the sequence number is correct in every frame that has been sent and it never accuses a sequence number error, only at the receiver side.
I've tried reducing the amount of bytes per frame to 3bytes, the incorrect behavior remains. If I send data every 10ms or 100ms, I don't have frames lost.
The way I send data is this: I accumulate frames to be sent in a buffer and every time ESP_SPP_WRITE_EVT happens, it sends the data existing in the buffer (in order to prevent congestion). The buffers never overflow of data, the bt sends can keep up with the data generation.

I'm using the latest esp-idf.

### Expected Behavior

### Actual Behavior

### Steps to reproduce
// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

## Other items if possible
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/5504858/sdkconfig.zip)

Alvin1Zhang, 2020-11-09:
Thanks for reporting.

Wth-Esp, 2020-11-09:
Hi @francis2tm,

We have optimized the spp flow control, but I'm not sure it can solve your issue. And here are my suggestions to help us with solving this issue:

1. Could you please provide your simplified demo? This would help us with reproducing and debugging the issue.
2. Please try the SPP VFS mode in IDF first to see whether the issue is still there. This is very helpful, will help us to diagnose the flow control side codes of SPP common mode.
3. Please pay attention to the  `ESP_SPP_CONG_EVT` event add some log to help us to diagnose the issue.

Thanks
Weitianhua

Alvin1Zhang, 2020-11-23:
@francis2tm Thanks for reporting, would you please help share if any updates for this issue? Thanks.

francis2tm, 2020-11-25:
Yes, I'm having esp_spp_cb: ESP_SPP_CONG_EVT when the packets are lost

francis2tm, 2020-12-02:
@Alvin1Zhang  @Wth-Esp Any news?

francis2tm, 2020-12-03:
Okay I have updates.
Now I prevented congestion by only making esp_spp_write when the previous write has been finished (ESP_SPP_WRITE_EVT) and when param->cong.cong == 0 (in ESP_SPP_CONG_EVT). Now, it works well with 8kBytes/s throughput for some 200s. Then, my production buffers overflow-> meaning, the bluetooth sends can't keep up with the data production (8kBytes/s).

`case ESP_SPP_CONG_EVT:                          //connection congestion status changed
        if(param->cong.cong == 0 && cong_flag == 2){
            cong_flag = 0;
        }
        break;
    case ESP_SPP_WRITE_EVT:                         //write operation completed (quando um envio foi completado)
        if(param->write.cong == 0){
            cong_flag = 0;
        }else{
            cong_flag = 2;
        }
        break;`

And every 1ms, I check: 
`if(!cong_flag){
                sendData();
            }`

Wth-Esp, 2020-12-04:
Hi @francis2tm,

4Mbps is the maximum baud rate via UART port to BT controller not the baud rate of BREDR. The BREDR transmission bandwidth is generally 1Mbps. The theoretical maximum peak value is 3 Mbps, and this value cannot be reached continuously and stably. From our test between two ESP32-WROVER-B DevKit, the throughput of  SPP is about 1900 kbps.

Could you provide your demo for us to retest the real throughput?

Thanks
Weitianhua

francis2tm, 2020-12-08:
[Embedded.zip](https://github.com/espressif/esp-idf/files/5662092/Embedded.zip)
Here is the demo!

boblane1, 2020-12-09:
Hi @francis2tm, 
1. I suggest you to reduce the priority of task 'initBt', because the max priority of ESP32 is 25. And also, you can raise the task 'acqAdc1Task' properly.

2. I recommend you to use SPP vfs write in ‘acqAdc1Task’, and it will make your application code much easier. You can refer the demo in https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/bt_spp_vfs_initiator. However, if you don't want to use this mode, I offer you some suggestion.

francis2tm, 2020-12-09:
Hey @boblane1 ,
Thanks for your reply.
a) Isn't VFS less efficient? Since you have an extra layer for the virtual file system which increases overhead?
b) Isn't lists/queues also less efficient than the 2 fixed sized arrays? Because that way I always need to alloc/free memory in every iteration, which also seems to increase the overheads.

boblane1, 2020-12-09:
Hi @francis2tm,
a) Yes, VFS will be less efficient, but you don't need to care about the flow control.
b) Lists/queues will increase the overheads, but it is convenient to resend the failed written packet.

If it is unnecessary to send ADC data every 1ms, you can accumulate the data as much as ESP_SPP_MAX_MTU to send. It is not recommended sending a small amount of data at high frequency. The flow control is not affected by the amount of data but also the number of packets.

francis2tm, 2020-12-09:
> Hi @francis2tm,
> a) Yes, VFS will be less efficient, but you don't need to care about the flow control.
> b) Lists/queues will increase the overheads, but it is convenient to resend the failed written [...]

Okay I see, But I think by adding your suggestions my flow control will be correct right?
Because I really want to achieve the most bluetooth throughput possible in esp-wroom-32. That being said, is my approach the best one to achieve the best throughput possible in esp-wroom-32?

Thanks for your help

boblane1, 2020-12-10:
Hi @francis2tm,

If you want to achieve the most Bluetooth throughput possible in esp-wroom-32, I think you can send much more data(but not exceed ESP_SPP_MAX_MTU) when the spp is not congest. For example, you can send 80bytes every 10ms rather than 8 bytes every 1ms.

francis2tm, 2020-12-11:
Thanks @boblane1 , your suggestions worked!

closed 2020-12-11 by francis2tm