espressif/esp-idf #6085: ESP32 Hostname is not working

rmshub, 2020-11-08:
## Environment

- Development Kit:  ESP32-Ethernet-Kit
- Kit version : v1.1
- Module or chip used: ESP32-WROVER-B
- IDF version: v4.2-beta1
- Build System: idf.py
- Compiler version: esp-2020r2-8.2.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes (Visual Studio Code Version: 1.51.0)
- Power Supply: USB

## Problem Description

Unable to access or ping the webserver using hostname. Tried default hostname ( espressif ) and customized after start event.
The customized hostname is set after ETHERNET_EVENT_START for ethernet and WIFI_EVENT_STA_START for WiFi STA. Ethernet is connected to router and wifi station is connected to mobile hotspot.  The laptop is connected to router wifi for ethernet ping/webserver verification and mobile hotspot for wifi sta ping/webserver verification.

### Expected Behavior
The webserver should be access using hostname

### Actual Behavior
The webserver is not accessible using hostname

### Code to set hostname

```cpp
ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ETHERNET_EVENT_START, &set_hostname, netif));
ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_STA_START, &set_hostname, netif));

static void set_hostname(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data)
{
    esp_netif_t *netif = (esp_netif_t *)arg;
    nvs_handle_t hNVS;
    char hostname[32];
    size_t len = sizeof(hostname); 
    /* Open NVS partition */
    ESP_ERROR_CHECK(nvs_open(TAG, NVS_READONLY, &hNVS));
    if (event_id == ETHERNET_EVENT_START) {
        /* Get the ethernet interface hostname from NVS */
        ESP_ERROR_CHECK(nvs_get_str(hNVS, "eth_hostname", hostname, &len));
        /* Set the ethernet interface hostname */
        ESP_ERROR_CHECK(esp_netif_set_hostname(netif,hostname));
    } else {
        /* Get the WiFi interface hostname from NVS */
        ESP_ERROR_CHECK(nvs_get_str(hNVS, "wifi_hostname", hostname, &len));
        /* Set the WiFi interface hostname */
        ESP_ERROR_CHECK(esp_netif_set_hostname(netif,hostname));
    }
    /* Close NVS partition */
    nvs_close(hNVS);
}

```

## Other items
[image]
[image]

chegewara, 2020-11-08:
I have this code before `esp_wifi_init()` and it works:
```
esp_netif_t *esp_netif = NULL;
esp_netif = esp_netif_next(esp_netif);
esp_netif_set_hostname(esp_netif, "hostname");
```
my commit is bd7fecd396e4985c6db0be4ad3de79cf83b29dae from master branch.

rmshub, 2020-11-08:
@chegewara I am able to see customized hostname in DHCP Discover and DHCP Request packets. However, the DHCP ACK packet is missed from router. Due to this miss, may be hostname is not working. Is anything missed in DHCP Discover and DHCP Request packets from esp32.

rmshub, 2020-11-08:
Code to initialize WiFi Station:
```cpp

    char *desc;
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    esp_netif_inherent_config_t esp_netif_config = ESP_NETIF_INHERENT_DEFAULT_WIFI_STA();
    /* Prepare the WiFi Station driver description with module name */
    asprintf(&desc, "%s: %s", TAG, esp_netif_config.if_desc);
    esp_netif_config.if_desc = desc;
    esp_netif_config.route_prio = 128;
    esp_netif_t *netif = esp_netif_create_wifi(ESP_IF_WIFI_STA, &esp_netif_config);
    free(desc);
    /* Set the WiFi interface hostname */
    ESP_ERROR_CHECK(esp_netif_set_hostname(netif,"ipnet-wifi"));
    esp_wifi_set_default_wifi_sta_handlers();
```
[image]

rmshub, 2020-11-08:
The hostname is updated initially in router and gone immediately within few seconds. Is anything related to lease time in ESP32 DHCP server?

KaeLL, 2020-11-13:
> The customized hostname is set after ETHERNET_EVENT_START for ethernet

Mine's [called way before that](https://github.com/KaeLL/w5100_esp32/blob/master/eth_main.c#L133), which, to my knowledge, it's the soonest possible time (right after the esp_netif object is initialized), and it still doesn't work
[image]
despite ...
```
D (368) eth_example: entered eth_main
V (368) esp_netif_objects: esp_netif_next_unsafe 0x0
V (368) esp_netif_objects: esp_netif_next_unsafe 0x0
D (370) esp_netif_objects: esp_netif_add_to_list 0x3ffbb968
D (370) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 1)
D (370) esp_netif_lwip: check: remote, if=0x3ffbb968 fn=0x400ed16c
0x400ed16c: esp_netif_set_hostname_api at /home/dev07/Documents/VSCode/esp_idf/prod/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1000

D (372) esp_netif_lwip: esp_netif_set_hostname_api esp_netif:0x3ffbb968 hostname w5100_esp32
D (372) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (374) w5100_eth_mac: entered esp_eth_mac_new_w5100
```

OP's at least looks like at least half of it works. Mine doesn't work at all.

KaeLL, 2020-11-17:
@david-cermak any input on this?

rmshub, 2020-11-17:
Still no solutions for me also

david-cermak, 2020-11-18:
Hi @rmshub 

just for me to better understand the issue, by `the hosname not working` you mean:
1) the hostname doesn't shown in the leases of your dhcp server?
2) cannot ping the hostname (or access, i.e. `getaddrinfo()` fails) as you indicate in the issue description?

Please note, that the (2) is **not** supposed to work, the hostname option in dhcp packet doesn't automatically mean the device to be accessible by its hostname in your network.

In case of (1), we would need more info about your dhcpd, I've just checked on linux and macos and seems to work okay, the hostname is present and correct (in `dhcpd.leases` on linux and `/var/db/dhcpd_leases` on mac)

rmshub, 2020-11-19:
Hi @david-cermak 
Thank you for your information.
Currently I am looking for option (2). Is there any method to identify the dynamic IP address (mDNS is not consistent in windows).

david-cermak, 2020-11-19:
I would suggest using mDNS, I think you can find stable bonjour implementation on windows, too.

KaeLL, 2020-11-21:
@david-cermak Any thoughts on the first DHCP not using the changed hostname, or should I open a new issue?

david-cermak, 2020-11-23:
@KaeLL Does it show the default hostname only for on first DHCP packet or all of them?
Yes, setting the hostname should work any time between creating the netif object `esp_netif_new()` and setting the netif up, i.e. starting the interface and thus starting the DHCP server.

Do you use a custom ethernet port layer in lwip? Isn't it possible your hostname gets overwritten when ethernet started? Such as here: https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/netif/ethernetif.c#L205 ?

KaeLL, 2020-11-23:
@david-cermak Nevermind, [you already fixed it](https://github.com/espressif/esp-idf/commit/7c8cbe02e48326ce4a590150aad71e3b90750c08#diff-aec48c7e5964007ae6f571d7a74f00e46b41946ec411ff7e39dd82a33c733e76). I'm using v4.1, which is why I'm still having this issue, but good to know it's been fixed for v4.2.

david-cermak, 2020-11-24:
@KaeLL Thanks for the reference! it looks like we've forgotten to backport this commit to `v4.1`. will fix!

david-cermak, 2020-11-27:
@rmshub Could you please close the issue, as setting the hostname in DHCP apparently works. (for the hostname resolution, some other means of "advertising" of the name is needed)

closed 2020-11-29 by rmshub