espressif/esp-idf #6087: Release V4.1 - TCP-IP losing part of the received packet on HTTP example in http_request_example_main.c

filzek, 2020-11-08:
After a lof of development we found out that there is something causing a problem while reading the server answer in the example  http_request_example_main.c

            r = read(s, recv_buf, sizeof(recv_buf)-1);

sometimes the read of the server answer is cutted so only parcial is read.

The problem is caused when the answer from the server is larger than 1440 chars, it doesnt happen sequential, only happen random so we keep sending and receiving and doing operations until it happen.

But once the problem happen the read is only sized 1440 chars, so its broken, not sure if there is a fix to it or if it is already know this bug?

shubhamkulkarni97, 2020-11-12:
@filzek I have a couple of questions regarding this issue.
1. Which server did you use to perform the test?
2. You have mentioned that issue occurs randomly, on an average how much time does it take to reproduce the issue?

Would you mind sharing code you used to reproduce this issue?
Also, please share the debug logs when issue occurs (To enable debug logs, run `idf.py menuconfig`, goto `Component config > Log Output` and set Default log verbosity to debug)

Thanks,
Shubham

shubhamkulkarni97, 2020-11-30:
@filzek Any updates on the issue?

filzek, 2020-11-30:
yes, it keep happen, we will prepare 3 answers for the issues open, so we can share log and exact what happen on server side and in the client side.

Alvin1Zhang, 2021-01-04:
@filzek Thanks for reporting, would you please help share if any updates for this issue? Thanks.

filzek, 2021-01-05:
We are debuging right now, its similar problem as seen in [https://github.com/espressif/esp-idf/issues/5846#issuecomment-754348620]Here on this issue

closed 2021-06-07 by Alvin1Zhang