espressif/esp-idf #6088: BLE service name doe snot change in provisioning manager

rhr407, 2020-11-08:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version (run ``git describe --tags`` to find it): v4.1

- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0

- Operating System:     [Linux]

- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

I have been working on a battery charger device based on GATT server in which I needed to set the BLE service name as BAT_CHARGER. I am now trying to use provisioning manager example but the service name is the same and it never goes away. I tried to use another esp32 devkit and after doing provisioning for several times now the service name is fixed to its MAC service name as is done with PROV_ prefix. I have tried another devkit which is now changing the service names but the earlier two have sort of bricked with the particular service names as mentioned above.  I have tried to do erase_flash many a times with no success. So I am still confused whether it's a H/W problem or a S/W issue. Any help will be highly appreciated.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. Just use the provisioning/manager example in esp-idf v4.1

chegewara, 2020-11-08:
Try to connect with nRF connect app installed on the same device you have provisioning manager app and refresh GATT attributes. Android does not do it automatically to decrease data exchange and battery usage.

rhr407, 2020-11-08:
Alright. I am using the espressif provisioning app on iphone. I have tried again on it but the service name is the same. I have tested it on nRF app and it's showing me the correct service name with PROV_  prefix.

Thanks a lot. Could you please let me know why is this issue happening in iOS as well?

chegewara, 2020-11-08:
I dont know iOS, but i think you can assume its the same mechanism. GATT attributes are stored/buffered and you have exclusively refresh from app level. It is a bit annoying during development, but in normal usage services and characteristics UUIDs very rare are changed.

rhr407, 2020-11-08:
Alright thanks a lot. I will close this issue.

closed 2020-11-08 by rhr407