espressif/esp-idf #6090: Add support for using specific interface to ESP HTTP client (and OTA)

jeremyherbert, 2020-11-09:
ESP32 devices can have multiple network interfaces (Ethernet, WiFi, PPP, etc). It would be useful if the various ESP APIs (HTTP client, OTA, etc) could be bound to a specific interface/transport/IP address.

Currently I am looking at implementing OTA, but I want to force it to only occur on the PPP interface, even though WiFi may be connected as well. It doesn’t seem like this is possible.

david-cermak, 2020-11-20:
Hi @jeremyherbert 

Thanks for raising this. Since most of espressif network libraries use the `tcp_transport` component, would it help to make this [`get_socket()`](https://github.com/espressif/esp-idf/blob/master/components/tcp_transport/private_include/esp_transport_internal.h#L54) API public? App code can simply bind this socket to a specific interface.

Not sure if it helps here, but you can also configure a `routing_prio` for each interface, so that PPP interface takes precedence over Wifi, but that would work for networking in general, not OTA only. Another possible workaround might be to set the WiFi down before OTA.

jeremyherbert, 2020-11-20:
I think that binding a socket sounds like a good idea. It’s currently what I do for all of my other code.

closed 2021-03-12 by espressif-bot