espressif/esp-idf #6092: TCP client is showing wrong error code when successfully connected to server (ENOTCONN)

shivam5594, 2020-11-09:
## Environment

- Development Kit:      ESP32-Wrover-Kit (esp32-cam to be precise)
- IDF version :  v4.3-dev-1561-g357a27760
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux 
- Using an IDE?: Yes Eclipse
- Power Supply:  USB

## Problem Description
I am trying to revive the TCP connection when it goes off in the middle of the active connection. For that I used the TCP_Client example from ESP_IDF git and surrounded the task while loop by another while loop to redo the thing and it works fine with necessary changes highlighted below. 

**ISSUE** is when it reconnects to the server, error code is 128 which is ENOTCONN (socket not connected) but I see data being successfully transmitted and received. 

To receive I am using netcat command in linux like
```
nc -l PORT_NUM
```

### Expected Behavior
I am not an expert of networking but either of 2 should be the case"
1- if connection established fine then errno should be 0 instead of 128 (ENOTCONN)
2- if it is actually 128 (ENOTCONN) then data sending should fail.

### Actual Behavior
Data is received without any problem at server. 

### Steps to reproduce

1. Use TCP_Client example from ESP_IDF
2. Add lines highlighted below using <---------------------

### Code to reproduce this issue

```cpp
static void tcp_client_task(void *pvParameters)
{
    char rx_buffer[128];
    char host_ip[] = HOST_IP_ADDR;
    int addr_family = 0;
    int ip_protocol = 0;

    while (1) <---------------------
    { <---------------------
		while (1)
		{
			struct sockaddr_in dest_addr;
			dest_addr.sin_addr.s_addr = inet_addr(host_ip);
			dest_addr.sin_family = AF_INET;
			dest_addr.sin_port = htons(PORT);
			addr_family = AF_INET;
			ip_protocol = IPPROTO_IP;
			int sock =  socket(addr_family, SOCK_STREAM, ip_protocol);
			if (sock < 0)
			{
				ESP_LOGE(TAG, "Unable to create socket: errno %d", errno);
				shutdown(sock, 0); <---------------------
				close(sock); <---------------------
				break;
			}
			ESP_LOGI(TAG, "Socket created, connecting to %s:%d", host_ip, PORT);

			int err = connect(sock, (struct sockaddr *)&dest_addr, sizeof(struct sockaddr_in6));
			if (err != 0)
			{
				ESP_LOGE(TAG, "Socket unable to connect: errno %d", errno);
				shutdown(sock, 0); <---------------------
				close(sock); <---------------------
				break;
			}
			ESP_LOGI(TAG, "Successfully connected");

			while (1)
			{
				int err = send(sock, payload, strlen(payload), 0);
				if (err < 0)
				{
					ESP_LOGE(TAG, "Error occurred during sending: errno %d", errno);
					break;
				}

				int len = recv(sock, rx_buffer, sizeof(rx_buffer) - 1, 0);
				// Error occurred during receiving
				if (len < 0)
				{
					ESP_LOGE(TAG, "recv failed: errno %d", errno);
					break;
				}
				// Data received
				else
				{
					rx_buffer[len] = 0; // Null-terminate whatever we received and treat like a string
					ESP_LOGI(TAG, "Received %d bytes from %s:", len, host_ip);
					ESP_LOGI(TAG, "%s", rx_buffer);
				}

				vTaskDelay(2000 / portTICK_PERIOD_MS);
			}

			if (sock != -1)
			{
				ESP_LOGE(TAG, "Shutting down socket and restarting...");
				shutdown(sock, 0);
				close(sock);
			}
		}
    }
    vTaskDelete(NULL);
}
```

Alvin1Zhang, 2020-11-09:
Thanks for reporting.

shivam5594, 2020-11-12:
@Alvin1Zhang is this going to be added in next release or any updates on this?

david-cermak, 2020-11-27:
@shivam5594 Just to be sure that I understand the issue. The `errno=128` is reported here:
https://github.com/espressif/esp-idf/blob/master/examples/protocols/sockets/tcp_client/main/tcp_client.c#L88
below the `recv()` from the socket, when the client re-connects to the server after broken connection?
In that case it seems like the `errno` just holds the previous error, or do I miss something?

shivam5594, 2020-12-03:
@david-cermak that's true, Sorry for delayed reply.

david-cermak, 2020-12-16:
@shivam5594 Could you please close the issue, then?

This is just  in line with the POSIX behaviour, i.e.  `errno` holding the last error, applications should check it only if `recv()` or any similar API returns fails. It is also possible possible to check internal socket error using `getsockopt(fd, SOL_SOCKET, SO_ERROR,...);` which gets cleared on read. Please check online resource for more details, e.g.:
* https://pubs.opengroup.org/onlinepubs/007908799/xnsix.html
* https://en.wikipedia.org/wiki/Berkeley_sockets

closed 2021-01-06 by Alvin1Zhang