espressif/esp-idf #6093: [ESP32-PICO-D4] SW_RESET bootloop before application invocation with ESP-IDF > v4.1

PaulFreund, 2020-11-09:
## Environment

- Module or chip used:  ESP32-PICO-D4
- IDF versions: v4.2-beta1-227-gf0e87c933, release/4.3, master
- Build System:         Make
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0 (depends on IDF version)
- Operating System:     WSL Debian
- Using an IDE?: VSCode
- Power Supply:         USB

## Problem Description

Application and boards works fine with release/v4.1 and before. 

### Expected Behavior

Application runs

### Actual Behavior

Reboot with SW_RESET before application invocation

### Steps to reproduce

1. Build firmware with any application code and the used sdkconfig/esp-idf version
2. Flash firmware
3. Bootloop

* Application code is not invoked. 
* Happens also with external pSRAM disabled.
* Works with release/v4.1
* Similar issue with release/v4.1 and minimum revision set to > 0

## Debug Logs

Boot loop with the following output:

```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:1228
load:0x40078000,len:12300
ho 0 tail 12 room 4
load:0x40080400,len:3268
entry 0x400805e4
```

## Other items if possible

- [x] sdkconfig file (attach the sdkconfig file from your project folder)

Alvin1Zhang, 2020-11-10:
Thanks for reporting.

projectgus, 2020-11-10:
Hi @PaulFreund,

Thanks for the clear issue report.

I'm not able to reproduce immediately, using the ESP-IDF commit you mention and the attached sdkconfig then I can build a firmware that runs on ESP32-PICO-D4 (silicon revision 1) without an issue.

I have a few questions to help us narrow this down:

* Are you able to post the output of `espefuse.py -p PORT summary` for the ESP32-PICO-D4 where the problem occurs, please?
* Are you able to set the [bootloader log level](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-bootloader-log-level) to "Error" or "Warning" and then reflash and capture the boot log, please? There is probably a bootloader error that is not being logged because bootloader logging is set to "No Output"
* Do you see this issue on one ESP32-PICO-D4, or on multiple units?

PaulFreund, 2020-11-10:
Hi @projectgus 

thank you so much for seeing this! I totally forgot I disabled the bootloader output. the error in my case was this in the end:

    E (930) esp_image: Image length 2057152 doesn't fit in partition length 2031616

Looks like with the SDK update the size grew and now my image won't fit. I'm going to be optimizing now :)

Thanks again!

closed 2020-11-10 by PaulFreund