espressif/esp-idf #6094: SIM800L PPPoS Client stops working after some hours with pppos_input_tcpip failed with -1

berkutta, 2020-11-09:
Using ESP-IDF 357a2776032299b8bc4044900a8f1d6950d7ce89 with modified example https://github.com/espressif/esp-idf/tree/master/examples/protocols/pppos_client on a LILYGO TTGO T-Call Publishing MQTT Messages every 10 Second.

```
Publishing to topic /topic/esp-860389056310861/status
test0
Received data from Modem: ~!E, 37 Bytes
7E 21 45 0 0 28 40 9A 40 0 2B 6 97 56 89 87 53 D9 A C1 8F BE 7 5B DD D4 F3 E3 5B 10 0 4 99 7C 50 10 FF
Received data from Modem: �jP, 8 Bytes
FF 6A 50 0 0 E2 4B 7E
Publishing to topic /topic/esp-860389056310861/status
test0
Received data from Modem: ~!E, 37 Bytes
7E 21 45 0 0 28 40 9B 40 0 2B 6 97 55 89 87 53 D9 A C1 8F BE 7 5B DD D4 F3 E3 5B 10 0 4 9A 21 50 10 FF
Received data from Modem: �i�, 8 Bytes
FF 69 AB 0 0 8A A5 7E
^[[0;31mE (17968901) esp-netif_lwip-ppp: pppos_input_tcpip failed with -1^[[0m
Publishing to topic /topic/esp-860389056310861/status
test0
Publishing to topic /topic/esp-860389056310861/status
test0
Publishing to topic /topic/esp-860389056310861/status
test0
```

MrCurio, 2020-12-08:
Hello @berkutta ,
Could you please share your code to check if i find something strange?

Have you checked other issues relates to pppos, there might be something related.

I'm not an expert but i worked with the Sim800L for hours and no problems.

berkutta, 2020-12-09:
Hi @MrCurio 

This is my code, it's basically the PPPoS Example with some modifications to send messages every 10 seconds and also turn on the modem via GPIO. It works fine for ~12 hours and then fails with pppos_input_tcpip returning -1 at some point.

```
/* PPPoS Client Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "esp_netif.h"
#include "esp_netif_ppp.h"
#include "mqtt_client.h"
#include "esp_modem.h"
#include "esp_modem_netif.h"
#include "esp_log.h"
#include "sim800.h"
#include "bg96.h"
#include "sim7600.h"

#include "cJSON.h"
#include "driver/gpio.h"

#define BROKER_URL "mqtt://mqtt.eclipse.org"

static const char *TAG = "pppos_example";
static EventGroupHandle_t event_group = NULL;
static const int CONNECT_BIT = BIT0;
static const int STOP_BIT = BIT1;
static const int GOT_DATA_BIT = BIT2;

static void modem_event_handler(void *event_handler_arg, esp_event_base_t event_base, int32_t event_id, void *event_data)
{
    switch (event_id) {
    case ESP_MODEM_EVENT_PPP_START:
        ESP_LOGI(TAG, "Modem PPP Started");
        break;
    case ESP_MODEM_EVENT_PPP_STOP:
        ESP_LOGI(TAG, "Modem PPP Stopped");
        xEventGroupSetBits(event_group, STOP_BIT);
        break;
    case ESP_MODEM_EVENT_UNKNOWN:
        ESP_LOGW(TAG, "Unknow line received: %s", (char *)event_data);
        break;
    default:
        break;
    }
}

static esp_err_t mqtt_event_handler(esp_mqtt_event_handle_t event)
{
    esp_mqtt_client_handle_t client = event->client;
    int msg_id;
    switch (event->event_id) {
    case MQTT_EVENT_CONNECTED:
        ESP_LOGI(TAG, "MQTT_EVENT_CONNECTED");
        //msg_id = esp_mqtt_client_subscribe(client, "/topic/esp-pppos", 0);
        //ESP_LOGI(TAG, "sent subscribe successful, msg_id=%d", msg_id);
        break;
    case MQTT_EVENT_DISCONNECTED:
        ESP_LOGI(TAG, "MQTT_EVENT_DISCONNECTED");
        break;
    case MQTT_EVENT_SUBSCRIBED:
        ESP_LOGI(TAG, "MQTT_EVENT_SUBSCRIBED, msg_id=%d", event->msg_id);
[... 268 lines cut ...]
             esp_dte_handle_line(esp_dte);
         } else {
@@ -167,6 +175,13 @@ static void esp_handle_uart_data(esp_modem_dte_t *esp_dte)
     length = uart_read_bytes(esp_dte->uart_port, esp_dte->buffer, length, portMAX_DELAY);
     /* pass the input data to configured callback */
     if (length) {
+        printf("Received data from Modem: %.*s, %d Bytes \n", length, esp_dte->buffer, length);
+
+        for(int i = 0; i < length; i++) {
+            printf("%X ", esp_dte->buffer[i]);
+        }
+        printf("\n");
+
         esp_dte->receive_cb(esp_dte->buffer, length, esp_dte->receive_cb_ctx);
     }
 }
@@ -494,7 +509,7 @@ esp_err_t esp_modem_start_ppp(modem_dte_t *dte)
     MODEM_CHECK(dce, "DTE has not yet bind with DCE", err);
     esp_modem_dte_t *esp_dte = __containerof(dte, esp_modem_dte_t, parent);
     /* Set PDP Context */
-    MODEM_CHECK(dce->define_pdp_context(dce, 1, "IP", CONFIG_EXAMPLE_COMPONENT_MODEM_APN) == ESP_OK, "set MODEM APN failed", err);
+    MODEM_CHECK(dce->define_pdp_context(dce, 1, "IP", "TM") == ESP_OK, "set MODEM APN failed", err);
     /* Enter PPP mode */
     MODEM_CHECK(dte->change_mode(dte, MODEM_PPP_MODE) == ESP_OK, "enter ppp mode failed", err);
```

MrCurio, 2020-12-09:
Hello @berkutta ,
From the error you are getting (-1), it seems you are running out of memory.

Take a look at the line i am pointing at:
````
err_t
pppos_input_tcpip(ppp_pcb *ppp, u8_t *s, int l)
{
  struct pbuf *p;
  err_t err;

  p = pbuf_alloc(PBUF_RAW, l, PBUF_POOL);
  if (!p) {
    return ERR_MEM; ---------------------------------> ERR_MEM error corresponds to -1 in err.h
  }
  pbuf_take(p, s, l);

  err = tcpip_inpkt(p, ppp_netif(ppp), pppos_input_sys);
  if (err != ERR_OK) {
     pbuf_free(p);
  }
  return err;
}
````

Since your pppos_input_tcpip function is returning -1, it means you ran out of memory. 

Be careful with initializing multiple instances and not freeing them. I can see that in every loop you create a new instance by calling xTimerCreate. This function does not override the previous timer, instead it allocates space for another one and returns the handle. 

You should only instance one timer before the while. Please give it a try and let me know. If this is not the issue, check other memory leaks that may be present in your code.

When you confirm, please close this issue!
Thanks!

berkutta, 2020-12-09:
Hi @MrCurio

Thanks for the code review. I think you are on a very hot track.

xTimerCreate just gets called once, it's within a loop but it never loops as it waits forever for `GOT_DATA_BIT` which doesn't get set any more. 

But my big flaw is after sending the MQTT Message. I correctly delete the cJSON object with `cJSON_Delete(root)`, but afterwards I try to free the cJSON object with `free(root)` which is wrong. Instead I should return `cJSON_PrintUnformatted(root)` to a pointer which I can then free after usage. Otherwise, cJSON allocates a string on every run without ever freeing it..

TL;DR: RTFM on cJSON, the updated code is now running with some additional code to monitor the heap usage. Will report back in a few days if it now runs stable.

Updated function:
```
void vTimerCallback( TimerHandle_t xTimer )
{
    static uint32_t loop = 0;

    cJSON *root = cJSON_CreateObject();
    cJSON *device = cJSON_CreateObject();

    cJSON_AddNumberToObject(device, "loop", loop);
    cJSON_AddNumberToObject(device, "heap", esp_get_free_heap_size());

    cJSON_AddItemToObject(root, "device", device);

    char publish_topic[60];

    sprintf(publish_topic, "/topic/esp-%d/status", 1234);

    char *payload = cJSON_PrintUnformatted(root);
    esp_mqtt_client_publish(mqtt_client, publish_topic, payload, 0, 0, 0);

    printf("Publishing to topic %s \n", publish_topic);
    
    cJSON_Delete(root);
    free(payload);

    loop++;
}
```

lhespress, 2020-12-10:
@berkutta 

I suggest that the code as follows:

```
static char *pppos_publish_payload(uint32_t loop) 
{
    cJSON *root = cJSON_CreateObject();
    cJSON *device = cJSON_CreateObject();

    cJSON_AddNumberToObject(device, "loop", loop);

    cJSON_AddItemToObject(root, "device", device);

    char *payload = cJSON_PrintUnformatted(root);
    cJSON_Delete(root);

    return payload;
} 

void vTimerCallback( TimerHandle_t xTimer )
{
    static uint32_t loop = 0;

    char *paylaod = pppos_publish_payload(loop);
    char publish_topic[60];
    sprintf(publish_topic, "/topic/esp-%d/status", 1234);
    
    esp_mqtt_client_publish(mqtt_client, publish_topic, paylaod, strlen(paylaod), 0, 0);

    printf("Publishing to topic %s \n", publish_topic);
    free(payload);
    loop++;
}
```
You‘ll remember to free the heap.

berkutta, 2020-12-11:
Works fine now, thanks @MrCurio!

yeah, thats a nice way of handling it, will remember that @lhespress

closed 2020-12-11 by berkutta